coolsnowwolf/lede #11611: 编译报错，貌似依赖冲突

guochan2019, 2023-10-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今天正常更新编译，结果报错，貌似依赖冲突

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
Collected errors:
 * resolve_conffiles: Existing conffile /home/hellomax/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/hellomax/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat-opkg.
 * check_data_file_clashes: Package libnetwork wants to install file /home/hellomax/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/docker-proxy
        But that file is already provided by package  * dockerd
 * opkg_install_cmd: Cannot install package libnetwork.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/hellomax/lede'
make[1]: *** [package/Makefile:111: /home/hellomax/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/hellomax/lede'
make: *** [/home/hellomax/lede/include/toplevel.mk:231: world] Error 2

Zane-E, 2023-10-21:
CONFIG_PACKAGE_libnetwork=y 改 # CONFIG_PACKAGE_libnetwork is not set

guochan2019, 2023-10-21:
> CONFIG_PACKAGE_libnetwork=y 改 # CONFIG_PACKAGE_libnetwork is not set

谢谢，已经通过

closed 2023-10-21 by coolsnowwolf

wuqingxin1987, 2023-10-24:
在那个位置改具体路径谢谢