coolsnowwolf/lede #11616: 请教lede默认编译是内核6.1，如何指定内核5.15？

luii2020, 2023-10-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：lede默认编译是内核6.1，如何指定内核5.15,最近编译的固件默认是内核6.1，有bug，例如lan口过段时间要手动连接才能上网，例如ssr+或者passwall 有两个ss节点，再添加x-ray节点后切换xray节点可以上网，切换回ss节点就上不了，原来只有两个ss节点来回切换是没问题，在ssr+和passwall里测试是通的，可能和lan有关或者固件不能保存有关，所以想用回内核5.15

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志
openwrt是旁路由，ip192.168.50.236，开启了ipv6
A：Mon Oct 23 09:22:17 2023 daemon.err nmbd[3779]: [2023/10/23 09:22:17.998565,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 09:22:17 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 09:22:17 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 09:27:17 2023 daemon.err nmbd[3779]: [2023/10/23 09:27:17.248273,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 09:27:17 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 09:27:17 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 09:32:18 2023 daemon.err nmbd[3779]: [2023/10/23 09:32:18.949901,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 09:32:18 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 09:32:18 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 09:37:19 2023 daemon.err nmbd[3779]: [2023/10/23 09:37:19.198055,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 09:37:19 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 09:37:19 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 09:38:07 2023 daemon.info avahi-daemon[2197]: Registering new address record for fd7e:61f3:8d3f:4f6c:fe7e:c774 on br-lan.*.
Mon Oct 23 09:39:20 2023 daemon.info avahi-daemon[2197]: Withdrawing address record for fd7e:61f3:8d3f:4f6c:fe7e:c774 on br-lan.
Mon Oct 23 09:42:21 2023 daemon.err nmbd[3779]: [2023/10/23 09:42:21.358001,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 09:42:21 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 09:42:21 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 09:47:21 2023 daemon.err nmbd[3779]: [2023/10/23 09:47:21.608184,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 09:47:21 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 09:47:21 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 09:52:21 2023 daemon.err nmbd[3779]: [2023/10/23 09:52:21.898566,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 09:52:21 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 09:52:21 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 09:57:27 2023 daemon.err nmbd[3779]: [2023/10/23 09:57:27.268302,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 09:57:27 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 09:57:27 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:02:27 2023 daemon.err nmbd[3779]: [2023/10/23 10:02:27.539104,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:02:27 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:02:27 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:05:04 2023 daemon.info avahi-daemon[2197]: Registering new address record for fd7e:61f3:8d3f:4f6c:fe7e:c774 on br-lan.*.
Mon Oct 23 10:06:17 2023 daemon.info avahi-daemon[2197]: Withdrawing address record for fd7e:61f3:8d3f:4f6c:fe7e:c774 on br-lan.
Mon Oct 23 10:07:31 2023 daemon.err nmbd[3779]: [2023/10/23 10:07:31.878151,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:07:31 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:07:31 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:12:37 2023 daemon.err nmbd[3779]: [2023/10/23 10:12:37.290062,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:12:37 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:12:37 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:17:37 2023 daemon.err nmbd[3779]: [2023/10/23 10:17:37.548859,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:17:37 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:17:37 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:22:44 2023 daemon.err nmbd[3779]: [2023/10/23 10:22:44.958488,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:22:44 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:22:44 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:27:47 2023 daemon.err nmbd[3779]: [2023/10/23 10:27:47.279031,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:27:47 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:27:47 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:32:55 2023 daemon.err nmbd[3779]: [2023/10/23 10:32:55.488438,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:32:55 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:32:55 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:36:16 2023 daemon.notice ttyd[9139]: [2023/10/23 10:36:16:5191] N: rops_handle_POLLIN_netlink: DELADDR
Mon Oct 23 10:38:00 2023 daemon.err nmbd[3779]: [2023/10/23 10:38:00.288806,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:38:00 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:38:00 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:43:02 2023 daemon.err nmbd[3779]: [2023/10/23 10:43:02.718095,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:43:02 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
[... cut ...]
Mon Oct 23 10:48:02 2023 daemon.err nmbd[3779]: [2023/10/23 10:48:02.989501,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mon Oct 23 10:48:02 2023 daemon.err nmbd[3779]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.50.236 for name WORKGROUP<1d>.
Mon Oct 23 10:48:02 2023 daemon.err nmbd[3779]:   This response was from IP 192.168.50.235, reporting an IP address of 192.168.50.235.
Mon Oct 23 10:48:29 2023 cron.err crond[20742]: crond (busybox 1.36.0) started, log level 5
Mon Oct 23 10:48:30 2023 cron.err crond[22074]: crond (busybox 1.36.0) started, log level 5

Travel2Here, 2023-10-23:
朋友打不开google？常规基础编译操作啊~
*做一次温柔好心人 在/target/linux 目录下~

Wilson-coder, 2023-10-23:
按照现在的配置，make menuconfig->Global Build Setting->use the testing kernel version，这个选项对应的就是5.15版内核

closed 2023-10-24 by luii2020

hztbhtbh, 2024-02-27:
指定编译内核版本的openwrt软路由系统的视频教程
[https://youtu.be/-jiSyMSGPOI](url)
网址需要复制粘贴游览器打开收看