coolsnowwolf/lede #11621: 最新6.1.60 内核 transmission添加任务后假死，web页面打不开

nb12nb34, 2023-10-27:
Fri Oct 27 08:11:13 2023 authpriv.info dropbear[24491]: Child connection from 192.168.0.103:64647
Fri Oct 27 08:11:13 2023 authpriv.notice dropbear[24491]: Password auth succeeded for 'root' from 192.168.0.103:64647
Fri Oct 27 08:11:53 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerd: line 6: extra_command: not found
Fri Oct 27 08:11:53 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerd: line 7: extra_command: not found
Fri Oct 27 08:11:53 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 6: config_load: not found
Fri Oct 27 08:11:53 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 130: extra_command: not found
Fri Oct 27 08:11:53 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 131: extra_command: not found
Fri Oct 27 08:11:59 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerd: line 6: extra_command: not found
Fri Oct 27 08:11:59 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerd: line 7: extra_command: not found
Fri Oct 27 08:11:59 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 6: config_load: not found
Fri Oct 27 08:11:59 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 130: extra_command: not found
Fri Oct 27 08:11:59 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 131: extra_command: not found
Fri Oct 27 08:12:00 2023 cron.err crond[16241]: USER root pid 27997 cmd /usr/sbin/tencentddns >> /var/log/tencentddns.log 2>&1
Fri Oct 27 08:12:00 2023 user.notice transmission: Starting with 3996498000 virt mem
Fri Oct 27 08:12:00 2023 daemon.err uhttpd[2120]: /etc/rc.common: eval: line 139: [-d: not found
Fri Oct 27 08:13:00 2023 cron.err crond[16241]: USER root pid 31859 cmd /usr/sbin/tencentddns >> /var/log/tencentddns.log 2>&1
Fri Oct 27 08:13:07 2023 kern.info kernel: [ 2456.586390] transmission-da[28459]: segfault at 0 ip 00007f0547293c00 sp 00007f0546f87010 error 4 in libcrypto.so.3[7f0547166000+270000] likely on CPU 3 (core 3, socket 0)
Fri Oct 27 08:13:07 2023 kern.info kernel: [ 2456.601407] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
Fri Oct 27 08:13:15 2023 authpriv.info dropbear[24491]: Exit (root) from <192.168.0.103:64647>: Exited normally
Fri Oct 27 08:13:23 2023 kern.info kernel: [ 2472.117142] transmission-da[31955]: segfault at 0 ip 00007fb44f0e7c00 sp 00007fb44eddb010 error 4 in libcrypto.so.3[7fb44efba000+270000] likely on CPU 3 (core 3, socket 0)
Fri Oct 27 08:13:23 2023 kern.info kernel: [ 2472.131964] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
Fri Oct 27 08:13:40 2023 kern.info kernel: [ 2489.649642] transmission-da[867]: segfault at 0 ip 00007fc094db8c00 sp 00007fc094aac010 error 4 in libcrypto.so.3[7fc094c8b000+270000] likely on CPU 3 (core 3, socket 0)
Fri Oct 27 08:13:40 2023 kern.info kernel: [ 2489.664289] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
Fri Oct 27 08:13:56 2023 kern.info kernel: [ 2504.925789] transmission-da[1990]: segfault at 0 ip 00007f442f75ec00 sp 00007f442f452010 error 4 in libcrypto.so.3[7f442f631000+270000] likely on CPU 2 (core 2, socket 0)
Fri Oct 27 08:13:56 2023 kern.info kernel: [ 2504.940499] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
Fri Oct 27 08:14:00 2023 cron.err crond[16241]: USER root pid 2066 cmd /usr/sbin/tencentddns >> /var/log/tencentddns.log 2>&1
Fri Oct 27 08:14:16 2023 kern.info kernel: [ 2525.808380] transmission-da[2097]: segfault at 0 ip 00007f0766252c00 sp 00007f0765f46010 error 4 in libcrypto.so.3[7f0766125000+270000] likely on CPU 3 (core 3, socket 0)
Fri Oct 27 08:14:16 2023 kern.info kernel: [ 2525.823092] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
Fri Oct 27 08:14:40 2023 kern.info kernel: [ 2548.997984] transmission-da[3765]: segfault at 0 ip 00007f3c5387ac00 sp 00007f3c5356e010 error 4 in libcrypto.so.3[7f3c5374d000+270000] likely on CPU 1 (core 1, socket 0)
Fri Oct 27 08:14:40 2023 kern.info kernel: [ 2549.012666] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
Fri Oct 27 08:14:40 2023 daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 18 seconds since last crash
Fri Oct 27 08:14:52 2023 authpriv.info dropbear[5373]: Child connection from 192.168.0.103:64792
Fri Oct 27 08:14:52 2023 authpriv.notice dropbear[5373]: Password auth succeeded for 'root' from 192.168.0.103:64792
Fri Oct 27 08:15:00 2023 cron.err crond[16241]: USER root pid 5475 cmd /usr/sbin/tencentddns >> /var/log/tencentddns.log 2>&1
Fri Oct 27 08:15:09 2023 authpriv.info dropbear[5373]: Exit (root) from <192.168.0.103:64792>: Exited normally
Fri Oct 27 08:15:41 2023 user.notice transmission: Starting with 3996498000 virt mem
Fri Oct 27 08:15:47 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerd: line 6: extra_command: not found
Fri Oct 27 08:15:47 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerd: line 7: extra_command: not found
Fri Oct 27 08:15:47 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 6: config_load: not found
Fri Oct 27 08:15:47 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 130: extra_command: not found
Fri Oct 27 08:15:47 2023 daemon.err uhttpd[2120]: sh: /etc/init.d/dockerman: line 131: extra_command: not found
Fri Oct 27 08:15:53 2023 kern.info kernel: [ 2622.189197] transmission-da[7494]: segfault at 0 ip 00007f45e1c8bc00 sp 00007f45e197f010 error 4 in libcrypto.so.3[7f45e1b5e000+270000] likely on CPU 3 (core 3, socket 0)
Fri Oct 27 08:15:53 2023 kern.info kernel: [ 2622.203902] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
Fri Oct 27 08:15:53 2023 daemon.info procd: Instance transmission::instance1 s in a crash loop 7 crashes, 12 seconds since last crash
Fri Oct 27 08:16:00 2023 cron.err crond[16241]: USER root pid 8577 cmd /usr/sbin/tencentddns >> /var/log/tencentddns.log 2>&1

eearphon, 2023-11-15:
遇到同样问题。刚回滚到
“
commit a2a0e5b9f6ce420398487ea29a85b0b0c49c9087 (HEAD)
Author: coolsnowwolf <coolsnowwolf@gmail.com>
Date:   Wed Sep 6 23:54:37 2023 +0800

    ipq-wifi: bump to latest git HEAD
”
编译了则正常，不过我用的内核 5.15。
结合log，所以感觉是“[openssl: bump to 3.0.10]“导致的

Kisoul, 2023-12-07:
我也遇到同样的问题，内核时6.1，就transmission每次启动的前10秒能进去，后面就进不去了，日志和大家的一样，今天准备试试5.15内核

Kisoul, 2023-12-10:
5.15.138的内核还是不行
内核日志如下

[ 1747.766056] transmission-da[10519]: segfault at 0 ip 00007f291a989c00 sp 00007f291a67d010 error 4 in libcrypto.so.3[7f291a85c000+270000]
[ 1747.766514] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
[ 1781.375146] transmission-da[11518]: segfault at 0 ip 00007f0927caec00 sp 00007f09279a2010 error 4 in libcrypto.so.3[7f0927b81000+270000]
[ 1781.375620] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
[ 2508.872580] transmission-da[25146]: segfault at 0 ip 00007f166fe1bc00 sp 00007f166fb0f010 error 4 in libcrypto.so.3[7f166fcee000+270000]
[ 2508.873024] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
系统日志
Sun Dec 10 22:08:44 2023 kern.info kernel: [ 2508.872580] transmission-da[25146]: segfault at 0 ip 00007f166fe1bc00 sp 00007f166fb0f010 error 4 in libcrypto.so.3[7f166fcee000+270000]
Sun Dec 10 22:08:44 2023 kern.info kernel: [ 2508.873024] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 41 54 41 89 f4 55 48 89 fd 48 81 ec 98 00 00 00 <48> 8b 07 48 83 78 78 00 0f 84 c2 00 00 00 66 0f ef c0 48 63 c6 48
Sun Dec 10 22:08:44 2023 daemon.info procd: Instance transmission::instance1 s in a crash loop 9 crashes, 23 seconds since last crash

dudulongg, 2023-12-11:
我也有相同的问题，请大佬们提供帮助

eearphon, 2023-12-12:
[image]
想用transmission可暂时回滚到这之前

Kisoul, 2023-12-12:
> ![微信截图_20231212143114](https://private-user-images.githubusercontent.com/46431124/289765822-8e50a457-f356-447d-bc68-0fb3548b5236.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIzNjU0NDAsIm5iZiI6MTcwMjM2NTE0MCwicGF0aCI6Ii80NjQzMTEyNC8yODk3NjU4MjItOGU1MGE0NTctZjM1Ni00NDdkLWJjNjgtMGZiMzU0OGI1MjM2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjEyVDA3MTIyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMyM2YyNjdhNTlhZTgyYWZkOGM2OTQ1ZTVhOTc4OGU5YzQwOTJjM2FhMzJjZDc1OTgzNDY5ZDA1NWY1ZGU1ZjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AOUJnKKMGMGxFcqzCuPefBhnPBrUBNwbZp1JEQhNIsQ) 想用transmission可暂时回滚到这之前

怎么回滚？求教，我现在把内核降到5.10了准备试试

dudulongg, 2023-12-12:
> 怎么回滚？求教，我现在把内核降到5.10了准备试试

git checkout a2a0e5b9f6ce420398487ea29a85b0b0c49c9087
我正在回滚编译

Kisoul, 2023-12-12:
> > > ![微信截图_20231212143114](https://private-user-images.githubusercontent.com/46431124/289765822-8e50a457-f356-447d-bc68-0fb3548b5236.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIzNjU0NDAsIm5iZiI6MTcwMjM2NTE0MCwicGF0aCI6Ii80NjQzMTEyNC8yODk3NjU4MjItOGU1MGE0NTctZjM1Ni00NDdkLWJjNjgtMGZiMzU0OGI1MjM2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjEyVDA3MTIyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMyM2YyNjdhNTlhZTgyYWZkOGM2OTQ1ZTVhOTc4OGU5YzQwOTJjM2FhMzJjZDc1OTgzNDY5ZDA1NWY1ZGU1ZjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AOUJnKKMGMGxFcqzCuPefBhnPBrUBNwbZp1JEQhNIsQ) 想用transmission可暂时回滚到这之前
> > 
> > 
> > 怎么回滚？求教，我现在把内核降到5.10了准备试试
> 
> git checkout [a2a0e5b](https://github.com/coolsnowwolf/lede/commit/a2a0e5b9f6ce420398487ea29a85b0b0c49c9087) 我正在回滚编译

谢谢，我也试试

nb12nb34, 2023-12-13:
在线编译的。如何回滚编译。。求教

Kisoul, 2023-12-14:
> git checkout [a2a0e5b](https://github.com/coolsnowwolf/lede/commit/a2a0e5b9f6ce420398487ea29a85b0b0c49c9087) 我正在回滚编译

有没有什么办法可以只openssl回滚，但其他的不回滚？

Kisoul, 2023-12-14:
> 在线编译的。如何回滚编译。。求教

[image]
如果是借用p3terx的代码，那么就在diy-part1.sh中加入如图内容

nb12nb34, 2023-12-16:
> ![image](https://private-user-images.githubusercontent.com/32985553/290386963-a1621e7b-4800-4aa1-b4f9-151f56e8d623.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

谢谢。。测试看看

eearphon, 2024-03-05:
可以不直接集成编译，而采取后安装的方式使用最新版：
一、编译最新版本 lede
二、opkg 安装 transmission 4.0.4
`root@OpenWrt:~# sed -i '/^option check_signature/s/^/#/' /etc/opkg.conf
root@OpenWrt:~# opkg update
root@OpenWrt:~# opkg install luci-i18n-transmission-zh-cn transmission-web-control
root@OpenWrt:~# opkg list-installed | grep transmission
luci-app-transmission - git-20.356.64372-1259bb1-1
luci-i18n-transmission-zh-cn - git-20.356.64372-1259bb1-1
transmission-daemon - 4.0.4-1
transmission-web-control - 2021-09-25-4b2e1858-3
root@OpenWrt:~# sed -i '/seccomp_path="\/etc\/seccomp\/transmission-daemon.json"/s/^/#/' /etc/init.d/transmission
`

eearphon, 2024-03-06:
[image]
命令中反斜杠被过滤了，已图片为准

dudulongg, 2024-03-28:
> 可以不直接集成编译，而采取后安装的方式使用最新版： 一、编译最新版本 lede 二、opkg 安装 transmission 4.0.4 root@OpenWrt:~# sed -i '/^option check_signature/s/^/#/' /etc/opkg.conf
>
> root@OpenWrt:~# opkg update
>
> root@OpenWrt:~# opkg [...]

在没编译TR的固件里手动安装好以后，怎么设置也没有进入web界面的按钮，输入IP地址加端口也没法进入tr，请问大佬应该怎么做才能正常运行TR

eearphon, 2024-04-09:
> 在没编译TR的固件里手动安装好以后，怎么设置也没有进入web界面的按钮，输入IP地址加端口也没法进入tr，请问大佬应该怎么做才能正常运行TR

最后一个命令运行了就会正常的

dudulongg, 2024-04-11:
> 最后一个命令运行了就会正常的

补全了反斜杠，和截图里的一样运行完成也是不行，很奇怪

eearphon, 2024-04-13:
> 补全了反斜杠，和截图里的一样运行完成也是不行，很奇怪

是通过 opkg 从默认源安装的吗？
opkg list-installed | grep transmission 看下

dudulongg, 2024-04-14:
> opkg list-installed | grep transmission

root@OpenWrt:~# opkg list-installed | grep transmission
luci-app-transmission - git-20.356.64372-1259bb1-1
luci-i18n-transmission-zh-cn - git-20.356.64372-1259bb1-1
transmission-daemon - 4.0.4-r1
transmission-web-control - 2021.09.25~4b2e1858-r3

eearphon, 2024-04-15:
> > opkg list-installed | grep transmission
> 
实在不好意思，这样的交互方式很难帮你排查。
刚Fork了L大的[luci](https://github.com/coolsnowwolf/luci)和[packages](https://github.com/coolsnowwolf/packages)并把Transmission升级到了4.0.5，你有兴趣的话可以编译测试。我多测试下再择机Pull Request

dudulongg, 2024-04-15:
> > > opkg list-installed | grep transmission
> 
> 实在不好意思，这样的交互方式很难帮你排查。 刚Fork了L大的[luci](https://github.com/coolsnowwolf/luci)和[packages](https://github.com/coolsnowwolf/packages)并把Transmission升级到了4.0.5，你有兴趣的话可以编译测试。我多测试下再择机Pull Request

确实这样交流很累，如果方便请加我qq：12503783

eearphon, 2024-07-13:
如想继续使用 transmission 2.94 可把依赖库从 openssl 改为 mbedtls 即可！
附上我改的 transmission/Makefile
[Makefile.zip](https://github.com/user-attachments/files/16201527/Makefile.zip)

fourcanacsa, 2024-07-13:
> > > > opkg list-installed | grep transmission
> > 
> > 
> > 实在不好意思，这样的交互方式很难帮你排查。 刚Fork了L大的[luci](https://github.com/coolsnowwolf/luci)和[packages](https://github.com/coolsnowwolf/packages)并把Transmission升级到了4.0.5，你有兴趣的话可以编译测试。我多测试下再择机Pull Request
> 
> 确实这样交流很累，如果方便请加我qq：12503783

你设置了验证，能添加一下我吗？谢谢。

fourcanacsa, 2024-07-13:
> 可以不直接集成编译，而采取后安装的方式使用最新版： 一、编译最新版本 lede 二、opkg 安装 transmission 4.0.4 root@OpenWrt:~# sed -i '/^option check_signature/s/^/#/' /etc/opkg.conf
>
> root@OpenWrt:~# opkg update
>
> root@OpenWrt:~# opkg [...]

你好，我的Openwrt固件里自带了一个Transmission 版本是2.9.4 请问怎么升级到4.0.5 现在貌似内核6.1以上的，只要有下载任务必定卡死。

fourcanacsa, 2024-07-13:
Transmission 2.9.4

> 依赖库

请问这要怎么改阿？

dudulongg, 2024-09-03:
> Transmission 2.9.4
> 
> > 依赖库
> 
> 请问这要怎么改阿？

我早放弃了，改qbit了

coolsnowwolf, 2026-09-05:
已确认问题根因：日志中的 `transmission-da` 在 OpenSSL 3 的 `libcrypto.so.3` 内反复段错误（并非内核 6.1 的 transmission 任务逻辑）。当前 2.94 源码已提供 mbedTLS 构建变体，因此提交 `216b77971e918c666fd4a236a96d12c29772323e` 将 transmission daemon/cli/remote 的默认加密后端切换为 mbedTLS，避免该旧版与 OpenSSL 3 的不兼容。包的 prepare 阶段已验证 mbedTLS 变体正确生成；完整 target compile 被工作树缺失 Linux 6.12 配置和 pppox.ko 的既有环境问题阻断。修复已 push 到 packages master，关闭 issue。

closed 2026-09-05 by coolsnowwolf