coolsnowwolf/lede #11624: pcie_mhi模块用不了

ChaingTsung, 2023-10-29:
源码中的移远pcie_mhi用不了
```
...
[   16.000926] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
[   16.010262] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
[   16.652163] UBIFS (ubi0:2): Mounting in unauthenticated mode
[   16.657924] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 784
[   17.128576] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rootfs_data"
[   17.136402] UBIFS (ubi0:2): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   17.146297] UBIFS (ubi0:2): FS size: 27680768 bytes (26 MiB, 218 LEBs), max 228 LEBs, journal size 1396736 bytes (1 MiB, 11 LEBs)
[   17.157926] UBIFS (ubi0:2): reserved for root: 1307430 bytes (1276 KiB)
[   17.164524] UBIFS (ubi0:2): media format: w5/r0 (latest is w5/r0), UUID 52931B17-F339-4DF1-B562-E22CBC6CB8AC, small LPT model
[   17.187120] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   17.229573] block: extroot: not configured
[   17.244860] UBIFS (ubi0:2): un-mount UBI device 0
[   17.249564] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[   17.261960] mount_root: switching to f2fs overlay
[   17.271913] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.
[   17.283941] urandom-seed: Seeding with /etc/urandom.seed
[   17.880289] mtdblock: MTD device 'reserved' is NAND, please consider using UBI block devices instead.
[   18.320907] mtdblock: MTD device 'ubi' is NAND, please consider using UBI block devices instead.
[   19.044089] mtdblock: MTD device 'bl2' is NAND, please consider using UBI block devices instead.
[   19.217329] mtdblock: MTD device 'fip' is NAND, please consider using UBI block devices instead.
[   19.656724] procd: - early -
[   19.659677] procd: - watchdog -
[   20.216184] procd: - watchdog -
[   20.220479] procd: - ubus -
[   20.282239] procd: - init -
[   21.006523] urngd: v1.0.2 started.
[   21.042092] kmodloader: loading kernel modules from /etc/modules.d/*
[   21.062910] trusted_key: encrypted_key: failed to alloc_cipher (-2)
[   21.199581] NET: Registered PF_ALG protocol family
[   21.229326] FS-Cache: Loaded
[   21.235365] CacheFiles: Loaded
[   21.239841] Key type dns_resolver registered
[   21.259558] Key type cifs.idmap registered
[   21.266938] NET: Registered PF_KEY protocol family
[   21.273964] Initializing XFRM netlink socket
[   21.279290] mpls_gso: MPLS GSO support
[   21.308256] tun: Universal TUN/TAP device driver, 1.6
[   21.325536] ipip: IPv4 and MPLS over IPv4 tunneling driver
[   21.341705] l2tp_core: L2TP core driver, V2.0
[   21.347294] l2tp_netlink: L2TP netlink interface
[   21.354374] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   21.362406] IPv4 over IPsec tunneling driver
[   21.371466] nat46: module (version 4c5beee236841724219598fabb1edc93d4f08ce5) loaded.
[   21.385062] ixgbe: Intel(R) 10 Gigabit PCI Express Network Driver
[   21.391159] ixgbe: Copyright (c) 1999-2016 Intel Corporation.
[   21.399094] ixgbevf: Intel(R) 10 Gigabit PCI Express Virtual Function Network Driver
[   21.406889] ixgbevf: Copyright (c) 2009 - 2018 Intel Corporation.
[   21.414519] gre: GRE over IPv4 demultiplexor driver
[   21.420917] ip_gre: GRE over IPv4 tunneling driver
[   21.428710] ip6_gre: GRE over IPv6 tunneling driver
[   21.441741] ksmbd: The ksmbd server is experimental, use at your own risk.
[   21.470645] st: Version 20160209, fixed bufsize 32768, s/g segs 256
[   21.516734] snd-soc-dummy snd-soc-dummy: no of_node; not parsing pinctrl DT
[   21.529338] i2c-gpio i2c-gpio-0: GPIO lookup for consumer sda
[   21.535157] i2c-gpio i2c-gpio-0: using device tree for GPIO lookup
[   21.541338] of_get_named_gpiod_flags: parsed 'sda-gpios' property of node '/i2c-gpio-0[0]' - status (0)
[   21.550735] mt7986a-pinctrl 1001f000.pinctrl: request pin 16 (GPIO_11) for pinctrl_moore:427
[   21.559168] gpio gpiochip0: Persistence not supported for GPIO 16
[   21.565259] i2c-gpio i2c-gpio-0: GPIO lookup for consumer scl
[   21.570987] i2c-gpio i2c-gpio-0: using device tree for GPIO lookup
[   21.577162] of_get_named_gpiod_flags: parsed 'scl-gpios' property of node '/i2c-gpio-0[0]' - status (0)
[   21.586542] mt7986a-pinctrl 1001f000.pinctrl: request pin 17 (GPIO_12) for pinctrl_moore:428
[   21.594970] gpio gpiochip0: Persistence not supported for GPIO 17
[   21.601283] i2c-gpio i2c-gpio-0: using lines 427 (SDA) and 428 (SCL)
[   21.607913] i2c-gpio i2c-gpio-1: GPIO lookup for consumer sda
[   21.613660] i2c-gpio i2c-gpio-1: using device tree for GPIO lookup
[   21.619865] of_get_named_gpiod_flags: parsed 'sda-gpios' property of node '/i2c-gpio-1[0]' - status (0)
[... 261 lines cut ...]
```
root@ChaingTsung:~# lspci
root@ChaingTsung:~# lsmod |grep mhi
mhi_wwan_mbim          20480  0
mhi_wwan_ctrl          20480  0
wwan                   24576  2 mhi_wwan_mbim,mhi_wwan_ctrl
qrtr_mhi               20480  0
qrtr                   40960  5 qrtr_mhi
mhi_net                16384  0
mhi                    81920  4 mhi_wwan_mbim,mhi_wwan_ctrl,mhi_net,qrtr_mhi
pcie_mhi              176128  0
root@ChaingTsung:~#
```

> dmesg中又能找到
```
root@ChaingTsung:~# dmesg |grep mhi
[   21.821145] mhi_init Quectel_Linux_PCIE_MHI_Driver_V1.3.4
```

> ls /dev 没有相关dev设备
```
root@ChaingTsung:~# ls /dev |grep mhi_*
root@ChaingTsung:~#
```

lhc1993, 2024-10-27:
请问解决了吗

ChaingTsung, 2024-10-27:
> 请登录

没有

coolsnowwolf, 2026-09-05:
日志只能证明 pcie_mhi 模块完成初始化（mhi_init Quectel_Linux_PCIE_MHI_Driver），不能证明 modem 已被 PCIe 枚举；报告同时明确 lspci 无任何输出，没有 PCIe BDF、MHI controller、probe 错误或 modem 型号/设备树/链路状态。当前 LEDE 中该 pcie_mhi 是移远厂商驱动，OpenWrt 主线没有同名包可用于回移。请先确认 PCIe 链路和设备枚举（lspci -nn、dmesg | grep -Ei pcie|aer|mhi、设备树 PCIe 节点、modem 型号和固件），再按具体硬件重新提交。本 issue 关闭为硬件/设备树信息不足，而非已证实的源码缺陷。

closed 2026-09-05 by coolsnowwolf