coolsnowwolf/lede #11626: 无法为banana pi r3编译

ChaingTsung, 2023-10-31:
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found libevent: /home/yoki/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libevent.so
CMake Error at cmake/Modules/FindOpenSSL.cmake:299 (list):
  list GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:12 (find_package)

CMake Error at cmake/Modules/FindOpenSSL.cmake:300 (list):
  list GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:12 (find_package)

CMake Error at cmake/Modules/FindOpenSSL.cmake:302 (list):
  list GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:12 (find_package)

CMake Error at cmake/Modules/FindOpenSSL.cmake:304 (list):
  list GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/yoki/lede/build_dir/target-aarch64_cortex-a53_musl/xfrpc-1.04.515/CMakeFiles/CMakeOutput.log".
make[3]: *** [Makefile:59: /home/yoki/lede/build_dir/target-aarch64_cortex-a53_musl/xfrpc-1.04.515/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/yoki/lede/feeds/packages/net/xfrpc'
time: package/feeds/packages/xfrpc/compile#0.10#0.02#0.15
    ERROR: package/feeds/packages/xfrpc failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xfrpc/compile] Error 1
make[2]: Leaving directory '/home/yoki/lede'
make[1]: *** [package/Makefile:110: /home/yoki/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/lede'
make: *** [/home/yoki/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-01-05:
升级一下即可，opensll升级了 这个xfrpc版本太低了 不匹配了，怎么升级 自己参考openwrt 官方那边  有现成的

coolsnowwolf, 2026-09-05:
已同步上游 packages 的 xfrpc 修复：从旧版 3.05.661 Git 源码构建切换到 5.06.909 官方发布 tarball（包含更新后的 CMake/OpenSSL 检测逻辑），提交 356fc788e6924e3a87ac6e65423b9f3d56bd2e6d 已独立推送。当前工作树执行构建时在进入 xfrpc 前被已有的 x86_64 内核 `pppox.ko` 缺失阻断（`kmod-pppox` 打包错误），与 xfrpc 无关；包定义已与上游一致。

closed 2026-09-05 by coolsnowwolf