coolsnowwolf/lede #11634: 求教~编译ROCKCHIP和X86_64的WiFi模块到底应该选择哪些选项？

rshi0212, 2023-11-03:
网上搜的看的迷迷糊糊
对应WiFi模块型号的 kmod
对应WiFi模块型号的 firmware
kmod-cfg80211
wireless-regdb *没找到在哪里
iw
iwinfo *没找到在哪里
hostapd-common
wpad-openssl
还有什么嘛？
有没有完整list？

rshi0212, 2023-11-05:
root@OpenWrt:~# lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:10:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:11:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:21:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
root@OpenWrt:~# lsusb
Bus 007 Device 001: ID 1d6b:0002 Linux 5.15.136 xhci-hcd xHCI Host Controller
Bus 003 Device 002: ID 0bda:c822 Realtek  Bluetooth Radio 
Bus 002 Device 001: ID 1d6b:0002 Linux 5.15.136 ehci_hcd EHCI Host Controller
Bus 004 Device 001: ID 1d6b:0001 Linux 5.15.136 ohci_hcd Generic Platform OHCI controller
Bus 006 Device 001: ID 1d6b:0003 Linux 5.15.136 xhci-hcd xHCI Host Controller
Bus 008 Device 001: ID 1d6b:0003 Linux 5.15.136 xhci-hcd xHCI Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.136 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0001 Linux 5.15.136 ohci_hcd Generic Platform OHCI controller
Bus 005 Device 001: ID 1d6b:0002 Linux 5.15.136 xhci-hcd xHCI Host Controller

rshi0212, 2023-11-05:
root@OpenWrt:~# dmesg 
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.15.136 (rshi0212@DESKTOP-5950X) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6237-8a4651f11) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Sat Nov 4 07:24:57 2023
[    0.000000] Machine model: FriendlyElec NanoPi R5C
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 64 MiB at 0x00000000e5000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s35736 r8192 d29800 u73728
[    0.000000] pcpu-alloc: s35736 r8192 d29800 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3772260K/3930112K available (8896K kernel code, 2408K rwdata, 2516K rodata, 1664K init, 742K bss, 92316K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS [mem 0xfd440000-0xfd45ffff]
[    0.000000] GIC: enabling workaround for ITS: Rockchip erratum 114514
[    0.000000] ITS@0x00000000fd440000: allocated 8192 Devices @440000 (indirect, esz 8, psz 64K, shr 0)
[    0.000000] ITS@0x00000000fd440000: allocated 32768 Interrupt Collections @450000 (flat, esz 2, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] ITS ALLOCATE PROP WORKAROUND
[    0.000000] GICv3: using LPI property table @0x0000000000460000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x0000000000470000
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001444] Console: colour dummy device 80x25
[    0.001934] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002987] pid_max: default: 32768 minimum: 301
[    0.003655] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.004413] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.007606] rcu: Hierarchical SRCU implementation.
[    0.008340] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[... 436 lines cut ...]
[   16.179483] kmodloader: dependency not loaded rtw88_8822b
[   16.179978] kmodloader: - rtw88_8822be - 2
[   16.180359] kmodloader: dependency not loaded rtw88_core
[   16.180858] kmodloader: - rtw88_8822c - 1
[   16.181231] kmodloader: dependency not loaded rtw88_pci
[   16.181710] kmodloader: dependency not loaded rtw88_8822c
[   16.182203] kmodloader: - rtw88_8822ce - 2
[   16.182588] kmodloader: - rtw88_core - 0
[   16.182953] kmodloader: dependency not loaded rtw88_core
[   16.183438] kmodloader: - rtw88_pci - 1
[   17.676760] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: . Quota mode: none.
[... cut ...]
[   21.779103] eth1: 0xffff800009d90000, ce:f2:0a:5c:cd:75, IRQ 80
[   24.676314] r8125: eth0: link up
[   24.676722] br-lan: port 1(eth0) entered blocking state
[   24.677201] br-lan: port 1(eth0) entered forwarding state
[   24.678092] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   31.714155] vccio_acodec: disabling
[   56.344950] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   56.345674] NFSD: Using legacy client tracking operations.
[   56.346155] NFSD: starting 10-second grace period (net f0000000)

rshi0212, 2023-11-05:
root@OpenWrt:~# opkg list-installed
6in4 - 26
attr - 2.5.1-1
autocore-arm - 66
automount - 1-40
autosamba - 1-13
avahi-dbus-daemon - 0.8-7
base-files - 71-r6237-8a4651f11
batctl-default - 2022.3-1
blkid - 2.38.1-1
block-mount - 2023-01-22-1ea5855e-1
boost - 1.81.0-1
boost-chrono - 1.81.0-1
boost-random - 1.81.0-1
boost-system - 1.81.0-1
brook - 20230606-2
btrfs-progs - 5.16.1-1
busybox - 1.36.0-1
ca-bundle - 20211016-1
ca-certificates - 20211016-1
cgroupfs-mount - 2020-06-26-05494281-3
containerd - 1.7.1-1
coremark - 2022-07-27-eefc986e-1
coreutils - 9.1-1
coreutils-base64 - 9.1-1
coreutils-nohup - 9.1-1
coreutils-timeout - 9.1-1
curl - 8.4.0-1
dawn - 2022-07-24-9e8060ea-1
dbus - 1.13.18-1
ddns-scripts - 2.7.8-5
ddns-scripts_aliyun - 1.0.3-1
ddns-scripts_dnspod - 1.0.2-1
default-settings - 3-8
dnsmasq-full - 2.88-2
docker - 24.0.5-1
dockerd - 24.0.5-1
dropbear - 2022.82-4
e2fsprogs - 1.46.5-2
etherwake - 1.09-5
ethtool - 6.0-1
firewall - 2022-02-17-4cd7d4f3-3
fstools - 2023-01-22-1ea5855e-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-08-13-4c7b720b-14
haproxy - 2.6.6-1
haveged - 1.9.18-1
hd-idle - 1.05-2
hostapd-common - 2022-07-29-b704dc72-7
htop - 3.2.2-1
hysteria - 2.2.0-1
ip-full - 6.2.0-1
ip6tables - 1.8.7-2
ip6tables-mod-nat - 1.8.7-2
ipset - 7.17-1
iptables - 1.8.7-2
iptables-mod-conntrack-extra - 1.8.7-2
iptables-mod-extra - 1.8.7-2
iptables-mod-fullconenat - 2022-02-13-108a36cb-10
iptables-mod-iprange - 1.8.7-2
iptables-mod-socket - 1.8.7-2
iptables-mod-tproxy - 1.8.7-2
ipv6helper - 1.0-2
iw - 5.19-1
iwinfo - 2023-05-17-c9f5c3f7-1
jansson4 - 2.14-3
jshn - 2022-09-27-ea560134-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.15.136-1-174ccca68a3845602e614ce884cd2ada
kmod-asn1-decoder - 5.15.136-1
[... 391 lines cut ...]
uci - 2021-10-22-f84f49f0-6
uclient-fetch - 2021-05-14-6a6011df-1
uhttpd - 2022-10-31-23977554-1
uhttpd-mod-ubus - 2022-10-31-23977554-1
umdns - 2021-05-13-b777a0b5-7
unzip - 6.0-8
urandom-seed - 3
urngd - 2020-01-21-c7f7b6b6-1
usb-modeswitch - 2022-02-24-3c8595a4-1
usbutils - 014-1
usign - 2020-05-23-f1f65026-1
uugamebooster - v3.15.0-1
v2ray-geoip - 202302090046-1
v2ray-geosite - 20230208103804-1
v2ray-plugin - 5.11.0-1
vsftpd-alt - 3.0.5-1
wget-ssl - 1.21.3-1
wireguard-tools - 1.0.20210914-2
wireless-regdb - 2022.08.12-1
wol - 0.7.1-3
wpad-openssl - 2022-07-29-b704dc72-7
wsdd2 - 2021-10-22-9831daf2-1
xray-core - 1.8.4-1
zerotier - 1.12.2-1
zlib - 1.2.13-9

beelabcloud, 2023-12-30:
网卡推荐 mt7921(wifi6)  mt7922 (wifi6)    rtl8822*E(wifi5)
k-->wireless drivers:  mt7921e  mt7922   rtw88
n-->wirelessapd-->只选 wpad-openssl (如果只做热点）其它不选
                               只选 wpad (同时做client)  其它不选

coolsnowwolf, 2026-09-05:
这是通用配置咨询，不是可复现的软件缺陷。当前 OpenWrt/LEDE 中无线驱动、固件和用户空间组件通过依赖自动选择：在 menuconfig 选择目标平台及具体 `kmod-*` 驱动后，会自动带入 `kmod-mac80211`/`kmod-cfg80211`、对应 firmware、`wireless-regdb`、`iw`/`iwinfo` 及 hostapd/wpad 依赖；无需手工维护完整清单。不同芯片（ath10k/ath11k/mt76/rtw88 等）所需包不同，应按 PCI/USB 芯片型号选择对应驱动，并用 `make defconfig` 检查依赖。issue 未给出具体硬件型号，无法生成唯一列表，现关闭。

closed 2026-09-05 by coolsnowwolf