coolsnowwolf/lede #11635: pass hy2 服务端无问题，最近几个版本都是GOOGLE PING 通，也在恩山找大神看过了，不能解决

josbu, 2023-11-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：本人小白也查看了ISSUE 和我这个问题相关的没有发现，目前我用5.15.120 最新PASS可以连HY2（老的版本也不是哪个都行），手机也可用NEKOBOX连HY2 没 有问题，
换成最新的内核6.1不化哪个版本都不行，PASS面板显示GOOGLE GITHUB 都是通的，但就是连不上最近两周都测试了不行。可以确认的有目前我服务端没有问题，OP只有特定版本才能用，看到之前 ISSUE 有说是dnsmasq 问题我也不太知道哪 怎么用不太懂 也看不出版本
贴个日志

### (2) 路由器型号和固件版本

A：x86-64  
内核版本 | 6.1.60

OpenWrt R23.10.24  我是爱快虚拟机装的OP

### (3) 详细日志

A：
Chain INPUT (policy ACCEPT)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere             /* !fw3 */
input_rule  all  --  anywhere             anywhere             /* !fw3: Custom input rule chain */
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED /* !fw3 */
syn_flood  tcp  --  anywhere             anywhere             tcp flags:FIN,SYN,RST,ACK/SYN /* !fw3 */
zone_lan_input  all  --  anywhere             anywhere             /* !fw3 */

Chain FORWARD (policy DROP)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere
forwarding_rule  all  --  anywhere             anywhere             /* !fw3: Custom forwarding rule chain */
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED /* !fw3 */
zone_lan_forward  all  --  anywhere             anywhere             /* !fw3 */
reject     all  --  anywhere             anywhere             /* !fw3 */

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere             /* !fw3 */
output_rule  all  --  anywhere             anywhere             /* !fw3: Custom output rule chain */
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED /* !fw3 */
zone_lan_output  all  --  anywhere             anywhere             /* !fw3 */

Chain MINIUPNPD (1 references)
target     prot opt source               destination

Chain forwarding_lan_rule (1 references)
target     prot opt source               destination

Chain forwarding_rule (1 references)
target     prot opt source               destination

Chain forwarding_wan_rule (1 references)
target     prot opt source               destination

Chain input_lan_rule (1 references)
target     prot opt source               destination

Chain input_rule (1 references)
target     prot opt source               destination

Chain input_wan_rule (1 references)
target     prot opt source               destination

Chain output_lan_rule (1 references)
target     prot opt source               destination

Chain output_rule (1 references)
target     prot opt source               destination

Chain output_wan_rule (1 references)
target     prot opt source               destination

Chain reject (1 references)
target     prot opt source               destination
REJECT     tcp  --  anywhere             anywhere             /* !fw3 */ reject-with tcp-reset
REJECT     all  --  anywhere             anywhere             /* !fw3 */ reject-with icmp-port-unreachable

Chain syn_flood (1 references)
target     prot opt source               destination
RETURN     all  --  anywhere             anywhere             limit: avg 25/sec burst 50 /* !fw3 */
DROP       all  --  anywhere             anywhere             /* !fw3 */

Chain zone_lan_dest_ACCEPT (4 references)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere             /* !fw3 */

Chain zone_lan_forward (1 references)
target     prot opt source               destination
MINIUPNPD  all  --  anywhere             anywhere
forwarding_lan_rule  all  --  anywhere             anywhere             /* !fw3: Custom lan forwarding rule chain */
zone_wan_dest_ACCEPT  all  --  anywhere             anywhere             /* !fw3: Zone lan to wan forwarding policy */
ACCEPT     all  --  anywhere             anywhere             ctstate DNAT /* !fw3: Accept port forwards */
zone_lan_dest_ACCEPT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_lan_input (1 references)
target     prot opt source               destination
input_lan_rule  all  --  anywhere             anywhere             /* !fw3: Custom lan input rule chain */
ACCEPT     all  --  anywhere             anywhere             ctstate DNAT /* !fw3: Accept port redirections */
zone_lan_src_ACCEPT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_lan_output (1 references)
target     prot opt source               destination
output_lan_rule  all  --  anywhere             anywhere             /* !fw3: Custom lan output rule chain */
zone_lan_dest_ACCEPT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_lan_src_ACCEPT (1 references)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere             ctstate NEW,UNTRACKED /* !fw3 */

Chain zone_wan_dest_ACCEPT (2 references)
target     prot opt source               destination

Chain zone_wan_dest_REJECT (1 references)
target     prot opt source               destination

Chain zone_wan_forward (0 references)
target     prot opt source               destination
forwarding_wan_rule  all  --  anywhere             anywhere             /* !fw3: Custom wan forwarding rule chain */
zone_lan_dest_ACCEPT  esp  --  anywhere             anywhere             /* !fw3: Allow-IPSec-ESP */
zone_lan_dest_ACCEPT  udp  --  anywhere             anywhere             udp dpt:isakmp /* !fw3: Allow-ISAKMP */
ACCEPT     all  --  anywhere             anywhere             ctstate DNAT /* !fw3: Accept port forwards */
zone_wan_dest_REJECT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_wan_input (0 references)
target     prot opt source               destination
input_wan_rule  all  --  anywhere             anywhere             /* !fw3: Custom wan input rule chain */
ACCEPT     udp  --  anywhere             anywhere             udp dpt:bootpc /* !fw3: Allow-DHCP-Renew */
ACCEPT     icmp --  anywhere             anywhere             icmp echo-request /* !fw3: Allow-Ping */
ACCEPT     igmp --  anywhere             anywhere             /* !fw3: Allow-IGMP */
ACCEPT     tcp  --  anywhere             anywhere             tcp dpt:1688 /* !fw3: kms */
ACCEPT     all  --  anywhere             anywhere             ctstate DNAT /* !fw3: Accept port redirections */
zone_wan_src_REJECT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_wan_output (0 references)
target     prot opt source               destination
output_wan_rule  all  --  anywhere             anywhere             /* !fw3: Custom wan output rule chain */
zone_wan_dest_ACCEPT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_wan_src_REJECT (1 references)
target     prot opt source               destination

EulerBlind, 2024-01-12:
同样遇到，hy2 测试通过，但无法代理

coolsnowwolf, 2026-09-05:
该 issue 提供的日志只有防火墙规则，没有 Passwall/Hysteria2 客户端握手、DNS 解析或连接错误信息，且环境为 2023 年的 LEDE 23.10/内核 6.1.60 与特定服务端配置，无法据此定位可复现的软件缺陷。当前 Passwall/Hysteria2 代码已持续更新；请在现行版本下仅保留一套代理内核并提供完整运行日志、节点协议参数和 `logread` 输出后再重新提交。现将此缺少可复现信息的历史 issue 关闭。

closed 2026-09-05 by coolsnowwolf