coolsnowwolf/lede #11644: 编译X86出错，请问下这是什么原因

yuzhe1129, 2023-11-06:
编译X86出错，

Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency luci-lua-runtime for luci-base
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	luci-lua-runtime
 * opkg_install_cmd: Cannot install package default-settings.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/c/lede'
make[1]: *** [package/Makefile:111: /home/c/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/c/lede'
make: *** [/home/c/lede/include/toplevel.mk:231：world] 错误 2

coolsnowwolf, 2023-11-06:
rm -rf bin

yuzhe1129, 2023-11-06:
> rm-rf bin

执行了还是一样的错误

yuzhe1129, 2023-11-06:
e[1]: *** [package/Makefile:111: /home/c/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/c/lede'
make: *** [/home/c/lede/include/toplevel.mk:231：world] 错误 2
c@ubuntu:~/lede$ rm -rf ./tmp && rm -rf .config
c@ubuntu:~/lede$ make menuconfig
Collecting package info: done
Collecting target info: done
WARNING: Makefile 'package/feeds/luci/luci-base/Makefile' has a dependency on 'luci-lua-runtime', which does not exist
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

c@ubuntu:~/lede$

aa4851698, 2023-11-06:
> rm -rf bin

删除了还是编译报错

jackiezhang29, 2023-11-06:
helloworld加#

DL53, 2024-02-16:
./scripts/feeds update -a
./scripts/feeds install -a
今天碰到了同样的问题，一开始执行过上面两条命令的，再执行一遍，后面就好了

coolsnowwolf, 2026-09-05:
该错误来自 2023 年旧版 LuCI/feed 混用：当时 `luci-base` 依赖包名为 `luci-lua-runtime`，但所生成的 package 索引中没有同架构候选，通常是 luci 与 buildroot/feeds revision 不匹配或只编译了部分 LuCI 包。当前 feeds/luci 仍提供 `luci-lua-runtime`，`luci-compat` 也明确依赖它，且 default-settings 已通过 luci-compat 间接依赖；使用匹配的 feeds revision 并完整编译 LuCI 即可。无当前代码缺陷，issue 关闭。

closed 2026-09-05 by coolsnowwolf