coolsnowwolf/lede #11648: GL.iNet GL-MT6000，无线网络信道选择auto时，概率性出现无线无法启动

moweiraul, 2023-11-07:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：经过自己反复测试发现，当无线网络（不论2.4G还是5G），如果信道选择了auto，概率性出现无线无法启动，并产生如下日志。看报错像是命令被传入了无效参数

但是如果手工指定具体信道，目前没发现有问题

### (2) 路由器型号和固件版本

A：GL.iNet GL-MT6000，OpenWrt R23.11.11 / LuCI Master (git-23.305.37706-43242b7)
目前不确定该问题是一个共性问题，还是仅在该路由器型号出现

### (3) 详细日志
A：
```
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957): Usage: ubus [<options>] <command> [arguments...]
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957): Options:
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  -s <socket>:		Set the unix domain socket to connect to
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  -S:			Use simplified output (for scripts)
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  -v:			More verbose output
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  -m <type>:		(for monitor): include a specific message type
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957): 			(can be used more than once)
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957): Commands:
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  - list [<path>]			List objects
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  - call <path> <method> [<message>]	Call an object method
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  - listen [<path>...]			Listen for events
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  - send <type> [<message>]		Send an event
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):  - monitor				Monitor ubus traffic
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957):
Tue Nov  7 16:50:26 2023 daemon.notice netifd: radio0 (6957): Device setup failed: HOSTAPD_START_FAILED
```

coolsnowwolf, 2023-11-07:
mt76 目前不支持auto

moweiraul, 2024-02-29:
> mt76 目前不支持auto

请问现在是否支持了？或者有什么计划吗？

coolsnowwolf, 2026-09-05:
该问题来自 2023 年 MT7986/mt76 旧版本，日志只显示 hostapd `config_add` 收到 Invalid argument，没有完整 hostapd 配置、ACS 扫描结果或驱动日志。当前 LEDE 已使用 mac80211 6.18.39，并在 2026-09-01 更新 mt76 到最新 Git HEAD；现有 hostapd/mt76 代码未发现针对 MT6000 自动信道的同类缺陷。固定信道可规避本地 ACS/DFS 环境问题，但不构成需要回移的通用代码修复。建议用最新固件复测，若仍失败请提供 ACS 扫描和完整无线配置。

closed 2026-09-05 by coolsnowwolf