coolsnowwolf/lede #11660: 编译x86/64时出错，应该怎么解决

AllenZhang45, 2023-11-13:
###  路由器型号和固件版本

X86/64  J4125

### 详细日志

sed -i "s/Installed-Time: .*/Installed-Time: 1699868848/" /home/allenzhang/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/status
rm -rf /home/allenzhang/lede/build_dir/target-x86_64_musl/root-x86/boot /home/allenzhang/lede/build_dir/target-x86_64_musl/root-x86/tmp/* /home/allenzhang/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/info/*.postinst* /home/allenzhang/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/lists/* /home/allenzhang/lede/build_dir/target-x86_64_musl/root-x86/var/lock/*.lock
find /home/allenzhang/lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1699868848" "{}" +
find: The relative path 'Files/NVIDIA' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
make[2]: *** [package/Makefile:73: package/install] Error 1
make[2]: Leaving directory '/home/allenzhang/lede'
make[1]: *** [package/Makefile:111: /home/allenzhang/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/allenzhang/lede'
make: *** [/home/allenzhang/lede/include/toplevel.mk:231：world] 错误 2

yanxin152133, 2023-11-18:
https://github.com/coolsnowwolf/lede/issues/3822#issuecomment-600596929

LazyBoBo, 2023-12-19:
同样的问题，x86/64，N100 Pve->debian11

echo "fdisk" >> /home/daniel/lede/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/daniel/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/daniel/lede/tmp/.root-copy.flock -c 'cp -fpR /home/daniel/lede/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/mount-utils/. /home/daniel/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/daniel/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.mount-utils_installed
echo "mount-utils" >> /home/daniel/lede/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/daniel/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/daniel/lede/tmp/.root-copy.flock -c 'cp -fpR /home/daniel/lede/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/partx-utils/. /home/daniel/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/daniel/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.partx-utils_installed
echo "partx-utils" >> /home/daniel/lede/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
make[3]: Leaving directory '/home/daniel/lede/package/utils/util-linux'
time: package/utils/util-linux/compile#86.47#14.00#68.09
make[2]: Leaving directory '/home/daniel/lede'
make[1]: *** [package/Makefile:110: /home/daniel/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/daniel/lede'
make: *** [/home/daniel/lede/include/toplevel.mk:231：world] 错误 2

LazyBoBo, 2023-12-21:
> 没说编译环境。 我昨天在windows11下，使用虚拟机VBOX编译。 第一次编译的时候，给虚拟机的内存设置成2048mb，也是有出现类似的错误。 因为第一次给虚拟机的内存设置成2048mb，结果出现mark错误，后来强制关闭了debian，vbox虚拟机给debian11内存设置成8G，结果进去就能正常编译了。
> 
> 以上方法可以试试，不行换个代理节点，重新试试git下载lede后的操作。

PVE虚拟机，2核 2G，换成2核8G，还是报了一样的错

xingshu1990, 2023-12-21:
> > 没说编译环境。 我昨天在windows11下，使用虚拟机VBOX编译。 第一次编译的时候，给虚拟机的内存设置成2048mb，也是有出现类似的错误。 因为第一次给虚拟机的内存设置成2048mb，结果出现mark错误，后来强制关闭了debian，vbox虚拟机给debian11内存设置成8G，结果进去就能正常编译了。
> > 以上方法可以试试，不行换个代理节点，重新试试git下载lede后的操作。
> 
> PVE虚拟机，2核 2G，换成2核8G，还是报了一样的错

我编译的情况不多（大概就编译两天），编译的环境也极其不稳定（vbox虚拟机装ubuntu14，vbox装debian11、debian12），
我所做的尝试是：
1.添加http代理。（但有一个问题，这个好像不是全局代理，还会有部分软件不会使用代理）
2.换成国内源，当时环境是ubuntu14，但是有三个依赖装不起来，arc、python2.7，还有一个忘记了，当时里面就换了阿里的源、清华的源，但是arc还是装不起来，遂放弃。
3.后安装了debian12，碰到的问题是不能安装python2.7。
4.后调整为debian11，非root账户下第一次执行【sudo apt update -y】还是【git clone https://github.com/coolsnowwolf/lede】会出现报错：初步原因是非root账户，没有权限，这个要谷歌一下方法。
添加权限后，sudo apt install -y ack……这部分我跑的比较久。
这里到了上面说的虚拟机内存太小的问题：
当时做过的尝试：重新按照：https://github.com/coolsnowwolf/lede的编译命令，重新按顺序执行一遍。

coolsnowwolf, 2026-09-05:
日志已明确给出原因：`find -execdir` 拒绝当前 `PATH`，因为其中包含相对路径 `Files/NVIDIA`，这是宿主机环境变量污染（常见于 Windows/WSL 或手工追加 NVIDIA 路径），与 x86_64/J4125 目标代码无关。请在编译前使用仅包含 Linux 绝对路径的 PATH，例如 `PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin`，重新打开 shell 后再构建。当前无需 LEDE 源码修复，关闭 issue。

closed 2026-09-05 by coolsnowwolf