coolsnowwolf/lede #11661: 5.4内核编译报错

slippersheepig, 2023-11-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由器编译固件遇到内核版本HASH不一致问题

### (2) 路由器型号和固件版本

A：gl-inet mt1300，编译5.4.260内核固件

### (3) 详细日志

A：make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt/target/linux'
make[3]: Entering directory '/workdir/openwrt/target/linux/ramips'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.linux-5.4.260.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "linux-5.4.260.tar.xz" "9830820714c8f5985c50071cc9e1b40533ee81a4f6c704916c7148d16e54ebfeO" "" "@KERNEL/linux/kernel/v5.x"    '
Cannot find appropriate hash command, ensure the provided hash is either a MD5 or SHA256 checksum.
make[3]: *** [Makefile:24: /workdir/openwrt/dl/linux-5.4.260.tar.xz] Error 255
make[3]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[2]: *** [Makefile:11: download] Error 2
make[2]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/download#0.05#0.04#0.09
    ERROR: target/linux failed to build.
make[1]: *** [target/Makefile:30: target/linux/download] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:203: download] Error 2
Error: Process completed with exit code 2.

Hirro029, 2023-11-14:
修改一下./include/kernel-5.4这个文件
把
LINUX_KERNEL_HASH-5.4.260 = 9830820714c8f5985c50071cc9e1b40533ee81a4f6c704916c7148d16e54ebfeO
改为
LINUX_KERNEL_HASH-5.4.260 = 9830820714c8f5985c50071cc9e1b40533ee81a4f6c704916c7148d16e54ebfe
就能正常编译了

slippersheepig, 2023-11-14:
> 修改一下./include/kernel-5.4这个文件 把 LINUX_KERNEL_HASH-5.4.260 = 9830820714c8f5985c50071cc9e1b40533ee81a4f6c704916c7148d16e54ebfeO 改为 LINUX_KERNEL_HASH-5.4.260 = 9830820714c8f5985c50071cc9e1b40533ee81a4f6c704916c7148d16e54ebfe 就能正常编译了

谢谢，这个可以PR吗😂

commit e14b03b6 (2023-11-14): kernel: fix linux 5.4 hash typo

Fixes: #11661
closed 2023-11-14 by aiamadeus