coolsnowwolf/lede #11663: 麻烦大佬们升级下go版本>=1.21.4

binge8, 2023-11-15:
有的插件需要>=1.21.4版本go   写了

GreatMichaelLee, 2023-11-15:
同样的问题，xray

rm -f /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5/.built
touch /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/builder/lede_x86/tmp/go-build" GO_MOD_CACHE_DIR="/home/builder/lede_x86/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/teddysun/xray-plugin" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/teddysun/xray-plugin/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5" GO_BUILD_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5/.go_work/build" GO_BUILD_BIN_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5=xray-plugin-1.8.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5=xray-plugin-1.8.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5/.go_work/build" GOCACHE="/home/builder/lede_x86/tmp/go-build" GOMODCACHE="/home/builder/lede_x86/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash /home/builder/lede_x86/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1699963787' -linkmode external -extldflags '-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"
Finding targets
go: ../../go.mod requires go >= 1.21.4 (running go 1.21.3; GOTOOLCHAIN=local)

Building targets
go: ../../go.mod requires go >= 1.21.4 (running go 1.21.3; GOTOOLCHAIN=local)

make[3]: *** [Makefile:47: /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.5/.built] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/passwall/xray-plugin'
time: package/feeds/passwall/xray-plugin/compile#0.21#0.54#3.68
    ERROR: package/feeds/passwall/xray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/passwall/xray-plugin/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231: world] Error 2

bi4tms, 2023-11-15:
请求大佬合并PR @coolsnowwolf 
https://github.com/coolsnowwolf/packages/pull/731

closed 2023-11-15 by binge8