coolsnowwolf/lede #11664: usr/Makefile:82: recipe for target 'usr/initramfs_data.cpio' failed

AndyLin2020, 2023-11-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ACRH-17最新源码编译过不了

### (2) 路由器型号和固件版本

A：ACRH-17最新源码,目前只有5.10内核

### (3) 详细日志

`make[5]: Entering directory '/home/andy/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.10.199'
  SYNC    include/config/auto.conf.cmd
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:484:warning: multi-line strings not supported
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     usr/initramfs_data.cpio
sing-box ../init.d/sing-box 777 0 0' line 486
usr/Makefile:82: recipe for target 'usr/initramfs_data.cpio' failed
make[6]: *** [usr/initramfs_data.cpio] Error 255
make[6]: *** Deleting file 'usr/initramfs_data.cpio'
Makefile:1832: recipe for target 'usr' failed
make[5]: *** [usr] Error 2
make[5]: Leaving directory '/home/andy/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.10.199'
Makefile:24: recipe for target '/home/andy/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.10.199/.image' failed
make[4]: *** [/home/andy/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.10.199/.image] Error 2
make[4]: Leaving directory '/home/andy/lede/target/linux/ipq40xx'
Makefile:11: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/andy/lede/target/linux'
time: target/linux/install#27.52#17.14#39.60
    ERROR: target/linux failed to build.
target/Makefile:28: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 1
make[2]: Leaving directory '/home/andy/lede'
target/Makefile:24: recipe for target '/home/andy/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed
make[1]: *** [/home/andy/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/andy/lede'
/home/andy/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2`

squallxgithub, 2023-11-18:
我也遇到了这个问题。。。。

CantionGH, 2023-11-20:
把passwall选项里面的sing-box去掉就可以了

AndyLin2020, 2023-11-20:
> 把passwall选项里面的sing-box去掉就可以了

正是需要sing-box呢

coolsnowwolf, 2026-09-05:
日志中的失败项是 initramfs 生成阶段处理 `sing-box ../init.d/sing-box 777 0 0` 返回 255，说明自定义 sing-box 包向内核 initramfs 列表写入了无效条目/路径；该包和 `../init.d/sing-box` 并非当前 LEDE 主线目标树的标准文件。issue 未提供所使用 feeds、sing-box 包来源或完整上一行错误，无法形成可验证补丁。请移除该自定义包或在其仓库修正 initramfs 安装脚本后重试；关闭该历史第三方包问题。

closed 2026-09-05 by coolsnowwolf