coolsnowwolf/lede #11665: Github Action云编译磁盘不足、编译超时问题解决方案

lorchr, 2023-11-17:
# Github Action云编译磁盘不足、编译超时问题解决方案

## 1. 超时原因
根据[Lean's OpenWrt Issue区](https://github.com/coolsnowwolf/lede/issues)大佬[EnnawYang](https://github.com/coolsnowwolf/lede/issues/11038#issuecomment-1486089183)和[Gabrielxzx](https://github.com/coolsnowwolf/lede/issues/11605#issuecomment-1774006018)的描述，结合[stupidloud](https://github.com/coolsnowwolf/lede/pull/7796)大佬的PR，编译超时主要有两个原因

1. `Toolchain`编译工具链会消耗一个多小时。`Toolchain`是可以复用的
2. 部分插件需要使用`Node`环境编译，非常耗时。编译时可以去掉，编译完成后再单独安装

## 2. 超时解决方案
针对这两个问题的解决方案
1. 将`Toolchain`缓存起来，后面编译时引用前面编译好的`Toolchain`，具体脚本可以参考大佬[Haiibo](https://github.com/haiibo/OpenWrt)的脚本。（[stupidloud](https://github.com/coolsnowwolf/lede/pull/7796)大佬的PR似乎不生效，不知道是不是我的使用有问题）
2. 去掉需要`Node`环境编译的插件，常用的有`unblockmusic` 和 `alist`

把 `Alist` 和 `Unblockmusic` 注释掉，在所在行前面加 `#` 或者 把 `=` 号后的 `y`(yes) 改为 `n`(no)
```shell
# 修改前
CONFIG_PACKAGE_luci-app-alist=y
CONFIG_PACKAGE_luci-app-unblockmusic=y

# 修改后
CONFIG_PACKAGE_luci-app-alist=n
CONFIG_PACKAGE_luci-app-unblockmusic=n
```
实测一次编译从超时（超过6小时）变为3小时左右，顺便磁盘的占用也会下降很多

## 3. 磁盘空间不足
编译前删除不需要的工具包，参考 [P3TERX](https://github.com/P3TERX/Actions-OpenWrt) 和 [Haiibo](https://github.com/haiibo/OpenWrt) 大佬的脚本

```shell
- name: Initialization environment
  env:
    DEBIAN_FRONTEND: noninteractive
  run: |
    docker rmi `docker images -q`
    sudo rm -rf /etc/apt/sources.list.d/* /usr/share/dotnet /usr/local/lib/android /opt/ghc /etc/mysql /etc/php
    sudo -E apt-get -y purge azure-cli* docker* ghc* zulu* hhvm* llvm* firefox* google* dotnet* aspnetcore* powershell* openjdk* adoptopenjdk* mysql* php* mongodb* moby* snap* || true
    sudo -E apt-get -qq update
    sudo -E apt-get -qq install libfuse-dev $(curl -fsSL git.io/depends-ubuntu-2204)
    sudo -E apt-get -qq autoremove --purge
    sudo -E apt-get -qq clean
    sudo timedatectl set-timezone "$TZ"
    sudo mkdir -p /workdir
    sudo chown $USER:$GROUPS /workdir
```

## 4. 识别编译是否包含【需要Node环境编译的插件】的方法
1. 下载编译日志压缩包（编译详情右上角齿轮），日志文件压缩包超过`1M`基本可以确定使用了`Node`
   - 在包含`alist`时，日志压缩包`8M+`，解压后`130M+`
   - 去掉`alist`后，日志压缩包`120k`，解压后`500k`
2. 打开日志文件，搜索关键字`node-v` + 版本号，例如目前编译使用的node版本为: `node-v16.19.1`，不知道版本号直接搜`node-v`即可
   - 定位到第一个 `node-v16.19.1` 出现的位置，往上翻第一个插件就是使用`node`编译的

例如下面例子：

[image]

可以看到 `node-v` 一共出现了近 **1.8w** 次，而第一次出现是在`alist`插件的下方，可以确定是`alist`编译用到了`node-v16`，去掉即可。

如果实在不会，安装一个[iStoreOS](https://www.istoreos.com/)，把可以从`iStore`安装的插件全部删除掉

## 5. 最后
以上为本人根据各位大佬的描述，多次在线编译的得出的结论，分享出来供大家参考，并非专业人士，如有错漏，还请勿喷。

lorchr, 2023-11-17:
1. 同样是 7763  在带 `Alist` 和不带 `Alist`，时间从超时变成了3小时，编译 `Alist` 是个很大的坑，完全可以从iStore 里面下载
2. 从[Haiibo](https://github.com/haiibo/OpenWrt)大佬的库里可以找到打印服务器配置的脚本，在编译开始如果发现分配的配置太低，可以关掉重新开始，总能随机到高配的服务器

dzlea, 2023-11-18:
1. 磁盘清理可以使用``"hugoalh/disk-space-optimizer-ghaction@v0.7.0"``这个工具
    
[image]
这是请求前后的空间对比

2. 只需要要在yml文件中
    ```yaml
       - name: Checkout
         uses: actions/checkout@main
    ```
    ``Checkout`` 步骤之前加上下面的代码
    ```yaml
      - name: "Optimize Disk Space"
        uses: "hugoalh/disk-space-optimizer-ghaction@v0.7.0"
        with:
          operate_sudo: "True"
          operate_async: "True"
          general_include: ".+"
          general_exclude: ",^GCC$,^G\\+\\+$"
          docker_include: ".+"
          docker_prune: "True"
          docker_clean: "True"
          apt_prune: "True"
          apt_clean: "True"
          homebrew_prune: "True"
          homebrew_clean: "True"
          npm_prune: "True"
          npm_clean: "True"
          os_swap: "True"
    ```

执行完后就会删除很多编译用不到的软件，从而节省空间

binge8, 2023-11-22:
最近一个礼拜github给分配的都是7763，本来6小时不够用的配置，现在都3小时完成。我每天晚上有4个型号的yml，都是7763

liwenjie119, 2023-11-22:
磁盘不足可以利用/dev/sdb1,挂载点/mnt，把编译链文件夹放这个盘里软链接过去就行，足够用了，但是时间限制还真是个问题
/dev/sda1        63G  4.1G   56G   7% /mnt

Kisoul, 2023-12-14:
> 磁盘不足可以利用/dev/sdb1,挂载点/mnt，把编译链文件夹放这个盘里软链接过去就行，足够用了，但是时间限制还真是个问题 /dev/sda1 63G 4.1G 56G 7% /mnt

求问怎么写这个代码？硬盘实在不够用

Kisoul, 2023-12-14:
[image]
我是这样处理的，如果是借用p3terx的代码就在diy-part1.sh
中加入上面的内容

dzlea, 2024-02-06:
> > 磁盘不足可以利用/dev/sdb1，挂载点/mnt，把编译链文件夹放这个盘里软链接过去就行，足够用了，但是时间限制还真是问题/dev /sda1 63G 4.1G 56G 7% /mnt
> 
> 真心求大佬教教代码如何写？这问题已经困扰我好久了。

[image]
我释放空间后有122G使用空间，你可以参考下

liwenjie119, 2024-02-06:
> ![image](https://private-user-images.githubusercontent.com/54771820/302472889-6e48a641-5eba-481c-975a-18fe88033a46.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

这还真是个好办法，比我清理空间清理得更彻底，而且很方便，

jimgo188, 2024-02-08:
请教大佬，代码怎么写？图像打不开看不到

liwenjie119, 2024-02-08:
> 请教大佬，代码怎么写？图像打不开看不到

我已经参照delea的代码改了一部分了，目前没问题。你可以参照他的，或者我的库，我的库还保留着使用旧的ln挂载文件夹的方法的文件

dzlea, 2024-02-09:
> 请教大佬，代码怎么写？图像打不开看不到

图片在这里
https://github.com/coolsnowwolf/lede/issues/11665#issuecomment-1928607936

具体代码去我代码仓库查看呗

jimgo188, 2024-02-09:
好的，谢谢上面2位大佬，我去抄下

xayane, 2024-02-11:
mark，抄下作业！

ljkk-li, 2024-02-21:
现在默认空间84g掉73了。。。越来越扣了

yuehua-s, 2024-04-16:
真心感谢大佬的帮助，总算解决了硬盘不足的问题。

SummerXxqq, 2024-09-16:
先mark，谢谢，希望用不到

edutfree, 2025-03-26:
> 现在默认空间84g掉73了。。。越来越扣了

好像也没有了 mnt，难道是操作不对？

binge8, 2025-03-26:
> > 现在默认空间84g掉73了。。。越来越扣了
> 
> 好像也没有了 mnt，难道是操作不对？

必须是操作不对

[image]

edutfree, 2025-03-26:
> > > 现在默认空间84g掉73了。。。越来越扣了
> > 
> > 
> > 好像也没有了 mnt，难道是操作不对？
> 
> 必须是操作不对
> 
> [image]

[image]

试过  runs-on: ubuntu-latest 也是这样……

zgc, 2025-04-15:
> > > > 现在默认空间84g掉73了。。。越来越扣了
> > > 
> > > 
> > > 好像也没有了 mnt，难道是操作不对？
> > 
> > 
> > 必须是操作不对
> > [image]
> 
> [image]
> 
> 试过 runs-on: ubuntu-latest 也是这样……

我试了几个账号，早期注册的账号才有mnt，新注册的账号一个也没有

edutfree, 2025-04-16:
> > > > > 现在默认空间84g掉73了。。。越来越扣了
> > > > 
> > > > 
> > > > 好像也没有了 mnt，难道是操作不对？
> > > 
> > > 
> > > 必须是操作不对
> > > [image]
> > 
> > 
> > [image]
> > 试过 runs-on: ubuntu-latest 也是这样……
> 
> 我试了几个账号，早期注册的账号才有mnt，新注册的账号一个也没有

刚拿一个2015年注册的老号试了，也没有mnt

zgc, 2025-04-16:
> > > > > > 现在默认空间84g掉73了。。。越来越扣了
> > > > > 
> > > > > 
> > > > > 好像也没有了 mnt，难道是操作不对？
> > > > 
> > > > 
> > > > 必须是操作不对
> > > > [image]
> > > 
> > > 
> > > [image]
> > > 试过 runs-on: ubuntu-latest 也是这样……
> > 
> > 
> > 我试了几个账号，早期注册的账号才有mnt，新注册的账号一个也没有
> 
> 刚拿一个2015年注册的老号试了，也没有mnt

[image]
抄了一下 @klttr  大佬的作业，现在没有mnt的也能勉强用了。

zgc, 2025-04-19:
> > > > > > 现在默认空间84g掉73了。。。越来越扣了
> > > > > 
> > > > > 
> > > > > 好像也没有了 mnt，难道是操作不对？
> > > > 
> > > > 
> > > > 必须是操作不对
> > > > [image]
> > > 
> > > 
> > > [image]
> > > 试过 runs-on: ubuntu-latest 也是这样……
> > 
> > 
> > 我试了几个账号，早期注册的账号才有mnt，新注册的账号一个也没有
> 
> 刚拿一个2015年注册的老号试了，也没有mnt

[image]
我也不清楚为什么你的账号都没有mnt

binge8, 2025-04-25:
编译前的
[image]
编译后的
[image]
我这编译的大杂烩都绰绰有余，不懂你的啥情况

coolsnowwolf, 2026-09-05:
该 issue 是 GitHub Actions 资源和构建配置的经验性说明，不是 LEDE 源码缺陷；正文给出的缓存 toolchain、减少 Node 插件、清理 runner 空间等方法属于工作流资源管理。仓库无法通过源码提交解决 GitHub runner 的磁盘/超时限制，因此不制作伪修复。

closed 2026-09-05 by coolsnowwolf