coolsnowwolf/lede #11668: Photonicat 在 5.15.138 內核版本下，lsusb 無法識別 Quectel 5g 網卡

shuenpin, 2023-11-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

使用雲編譯，在 5.15.136 的內核版本下面，lsusb 可以正確識別網卡
但是在 5.15.138 版本下面，無法識別網卡

### (1) 具体问题

A：在 11/13 後， clone 出來的源碼編譯成功，卻無法識別 Quectel 5G 網卡。但是之前的版本識別正常

### (2) 路由器型号和固件版本

A：Photonicat / 跟隨著 main tag

### (3) 详细日志

A：可以從 https://github.com/shuenpin/OpenWrt/releases 抓最新的雲編譯版本，進行問題復現。

shuenpin, 2023-11-20:
調查發現 5.15.138 的 usb option.c 並沒有 RM502Q 的 VID/PID

build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.138/drivers/usb/seria/option.c

嘗試 patch 看看

coolsnowwolf, 2023-11-21:
lsusb 看看 id ？

shuenpin, 2023-11-21:
升級到新的內核版本 5.15.138
```
oot@OpenWrt:~# lsusb
Bus 007 Device 001: ID 1d6b:0002 Linux 5.15.138 vhci_hcd USB/IP Virtual Host Controller
Bus 004 Device 001: ID 1d6b:0003 Linux 5.15.138 xhci-hcd xHCI Host Controller
Bus 006 Device 001: ID 1d6b:0003 Linux 5.15.138 xhci-hcd xHCI Host Controller
Bus 008 Device 001: ID 1d6b:0003 Linux 5.15.138 vhci_hcd USB/IP Virtual Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.138 ehci_hcd EHCI Host Controller
Bus 002 Device 002: ID 04ca:3a22
Bus 002 Device 001: ID 1d6b:0001 Linux 5.15.138 ohci_hcd Generic Platform OHCI controller
Bus 003 Device 001: ID 1d6b:0002 Linux 5.15.138 xhci-hcd xHCI Host Controller
Bus 005 Device 001: ID 1d6b:0002 Linux 5.15.138 xhci-hcd xHCI Host Controller
```
```
root@OpenWrt:~# uname -an
Linux OpenWrt 5.15.138 #0 SMP PREEMPT Mon Nov 20 05:25:09 2023 aarch64 GNU/Linux
```
`root@OpenWrt:~# dmesg | grep ttyUSB
`

shuenpin, 2023-11-27:
@coolsnowwolf 

測試結束

我在編譯的時候 checkout 11/12 的 source  (git checkout 59e66205e)，結果一切正常。

這樣是否 5.15.138 的 commit 或是核心有不相容的地方呢？

coolsnowwolf, 2023-12-17:
理论上现在的版本解决了

shuenpin, 2023-12-18:
> 理论上现在的版本解决了

看來還是沒有找到裝置，以下是 lsusb 的資訊

```
root@OpenWrt:~# lsusb
Bus 007 Device 001: ID 1d6b:0002 Linux 5.15.143 vhci_hcd USB/IP Virtual Host Controller
Bus 002 Device 001: ID 1d6b:0001 Linux 5.15.143 ohci_hcd Generic Platform OHCI controller
Bus 004 Device 001: ID 1d6b:0003 Linux 5.15.143 xhci-hcd xHCI Host Controller
Bus 006 Device 001: ID 1d6b:0003 Linux 5.15.143 xhci-hcd xHCI Host Controller
Bus 008 Device 001: ID 1d6b:0003 Linux 5.15.143 vhci_hcd USB/IP Virtual Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.143 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0002 Linux 5.15.143 xhci-hcd xHCI Host Controller
Bus 005 Device 001: ID 1d6b:0002 Linux 5.15.143 xhci-hcd xHCI Host Controller
```

coolsnowwolf, 2026-09-05:
该 issue 仅描述 5.15.136→5.15.138 后现象，没有 `dmesg`、`lsusb -v`、USB VID:PID、供电/复位信息，也未说明使用 QMI、MBIM 还是串口模式。`lsusb` 无法枚举与 qmi_wwan 驱动绑定是不同层次的问题；当前树已包含 Photonicat DTS、USB 主控及 qmi_wwan/MBIM/option 模块，无法据此形成针对性补丁。请在现行版本提供完整 USB 枚举日志和设备 ID 后重新提交，关闭该历史 issue。

closed 2026-09-05 by coolsnowwolf