coolsnowwolf/lede #11671: 咨询下，最新版本如何编译出SSR PLUS

franksz11, 2023-11-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

平台Ubuntu 20.04.6 LTS (GNU/Linux 5.4.0-144-generic x86_64)
1、已修改feeds.conf.default打开了src-git helloworld https://github.com/fw876/helloworld.git
2、make menuconfig 没有看到luci-app-ssr-plus
3、进路由设置echo 0xDEADBEEF > /etc/config/google_fu_mode，也没有看到ssr-plus

### (1) 具体问题

A：如2

### (2) 路由器型号和固件版本

A：x86，编译最新固件

### (3) 详细日志

A：

lyin888, 2023-11-21:
src-git helloworld https://github.com/fw876/helloworld;main

FMJYY23, 2023-11-21:
git clone --depth=1 -b main https://github.com/fw876/helloworld package/helloworld

star-o0O, 2023-11-21:
> 做过了
> 
> > src-git helloworld https://github.com/fw876/helloworld;main

看清楚点，带尾巴的，别瞄一眼就做过了，差着远呢

yanxin152133, 2023-11-26:
https://github.com/kenzok8/openwrt-packages

harrison-guo-chn, 2023-11-26:
src-git helloworld https://github.com/fw876/helloworld;main
分支变了，网上的教程可没更新的哦

yanxin152133, 2023-11-26:
搞不懂你们是需要酸酸乳还是helloworld
[image]
[image]

Maxswang, 2023-11-29:
@franksz11  在feeds.conf.default 文件 添加以下两行内容： 
```shell
src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git small https://github.com/kenzok8/small
```
添加以后执行以下命令：
```shell
 ./scripts/feeds update -a
./scripts/feeds install -a
```

closed 2024-01-19 by coolsnowwolf