coolsnowwolf/lede #11676: 从最新源码编译squashfs固件，请问如何将overlay默认分区类型 f2fs 改为 ext4

Maxswang, 2023-11-28:
由于squashfs 固件 overlay默认分区文件系统类型为 f2fs， 在不增加新分区的情况下，难以完成扩容， 故有想将overlay默认分区类型 f2fs 改为 ext4, 如何实现，希望大佬们给出建议！

Maxswang, 2023-11-28:
@Cathgao  make menuconfig 配置菜单里选中 ext4 输出的是单独 ext4固件， 而我想要一个squashfs固件，但其overlay默认分区为ext4, 而不是 f2fs文件系统类型

Maxswang, 2023-11-28:
我发现openwrt官方发布提供下载的squashfs固件，其overlay默认分区为 ext4类型， 不知道更改overlay默认分区类型是否要涉及到源码修改？

6ang996, 2023-11-28:
overlay分区的类型根据你的CONFIG_TARGET_ROOTFS_PARTSIZE大小减去ROM占用的squashfs大小，得到磁盘剩余空间后自动选择分区类型格式化，剩余空间大于100MB时选f2fs文件系统，否则选ext4
参考https://github.com/openwrt/fstools/blob/master/libfstools/common.c文件中的block_volume_format()函数中的ULOG_INFO("overlay filesystem in %s has not been formatted yet\n", v->blk);和use_f2fs()函数，应该可以通过修改#define F2FS_MINSIZE         (100ULL * 1024ULL * 1024ULL)中的100ULL值来实现

Maxswang, 2023-11-28:
@6ang996  谢谢大佬，通过修改两个源码文件libfstools/common.h 与 libfstools/rootdisk.c中 #define F2FS_MINSIZE (100ULL * 1024ULL * 1024ULL)  成功实现了overlay默认分区为ext4。

closed 2023-12-02 by Maxswang

Mircc, 2023-12-11:
> overlay分区的类型根据你的CONFIG_TARGET_ROOTFS_PARTSIZE大小减去ROM占用的squashfs大小，得到磁盘剩余空间后自动选择分区类型格式化，剩余空间大于100MB时选f2fs文件系统，否则选ext4 参考[https://github.com/openwrt/fstools/blob/master/libfstools/common.c文件中的block_volume_format()函数中的ULOG_INFO("overlay](https://github.com/openwrt/fstools/blob/master/libfstools/common.c%E6%96%87%E4%BB%B6%E4%B8%AD%E7%9A%84block_volume_format()%E5%87%BD%E6%95%B0%E4%B8%AD%E7%9A%84ULOG_INFO(%22overlay) filesystem in %s has not been formatted yet\n", v->blk);和use_f2fs()函数，应该可以通过修改#define F2FS_MINSIZE (100ULL * 1024ULL * 1024ULL)中的100ULL值来实现

大佬也就是说 需要扩容overlay的空间 之需要调整CONFIG_TARGET_ROOTFS_PARTSIZE大小就行对么

Maxswang, 2023-12-12:
@Mircc  编译时op固件会因不同的选择配置产生不同大小的ROM分区，没办法精确得知overlay的大小。 将CONFIG_TARGET_ROOTFS_PARTSIZE大小设置更大，在同样的编译配置下确实可以增大overlay分区空间大小。不过这样最终生成的固件文件也会随之更大。

Mircc, 2024-01-02:
> @Mircc 编译时op固件会因不同的选择配置产生不同大小的ROM分区，没办法精确得知overlay的大小。 将CONFIG_TARGET_ROOTFS_PARTSIZE大小设置更大，在同样的编译配置下确实可以增大overlay分区空间大小。不过这样最终生成的固件文件也会随之更大。
感谢告知。

xiaoyao9184, 2026-09-08:
搜到了，为了让后面的人知道，可以通过`CONFIG_GRUB_BOOTOPTS="fstools_overlay_fstype=ext4"`指定overlay覆盖层格式为ext4