coolsnowwolf/lede #11687: Netgear R7800  ath10k-ct编译错误

du-qg, 2023-12-04:
Applying ./patches/988-ath10k-always-use-mac80211-loss-detection.patch using plaintext: 
patching file ath10k-5.15/mac.c
touch /home/test/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_a5b8a1f5c6fc141bd7bb7a275a6eade9_6664517399ebbbc92a37c5bb081b5c53
make[3]: *** No rule to make target '/home/test/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/home/test/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[3]: Leaving directory '/home/test/lede/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.30#0.31#0.53
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).

如何排错?  (source-code  20231130)

hzjnet, 2024-01-21:
> Applying ./patches/988-ath10k-always-use-mac80211-loss-detection.patch using plaintext: patching file ath10k-5.15/mac.c touch [...]

半年前就遇见了，对比别人成功的配置文件发现是配置问题

SQ528, 2024-07-26:
一样，今年你解决了吗？
make[2]: Circular package/kernel/ath10k-ct/compile <- package/kernel/mac80211/compile dependency dropped.
make[3]: Entering directory '/workdir/openwrt/package/kernel/ath10k-ct'
make[3]: *** No rule to make target '/workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[3]: Leaving directory '/workdir/openwrt/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.16#0.06#0.20
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

commit e59a6368 (2025-01-07): ath10k-ct: select mac80211 debugfs support

Trying to compile ath10k-ct without mac80211 debugfs support will result in:
 8083 | #warning Please enable ATH10K-DEBUGFS kernel option for optimal support for CT firmware.
      |  ^~~~~~~
cc1: all warnings being treated as errors

So, since the driver itself is saying that debugfs is required, then
lets make ath10k-ct select mac80211 debugfs support which is selected
by default anyway.

Fixes: #11687, #12363
closed 2025-01-07 by aiamadeus