coolsnowwolf/lede #11688: GL-MT 2500 WAN 端口問題

Rurikobaka, 2023-12-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

GL-MT 2500 WAN 端口無法連接

Q：是否搜索了issue (使用 "x" 选择)

* [x] 没有类似的issue

## 2. 详细叙述

圖片

[image]

[image]

第二張是手動點擊重連後的情況

[image]

更換官方固件，是沒有問題的，網絡沒有問題的（物理意義上）

### (1) 具体问题

A：WAN 端口無法連接

### (2) 路由器型号和固件版本

A：GL-MT 2500，LEDE 最新源碼編譯（2023.12.5）

### (3) 详细日志

A：

System log

```
Tue Dec  5 11:57:39 2023 daemon.notice procd: /etc/rc.d/S90turboacc: net.ipv4.tcp_congestion_control = cubic
Tue Dec  5 11:57:39 2023 daemon.notice procd: /etc/rc.d/S90turboacc: killall: dnscache: no process killed
Tue Dec  5 11:57:39 2023 kern.info kernel: [   27.289903] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
Tue Dec  5 11:57:39 2023 kern.info kernel: [   27.298785] br-lan: port 1(eth1) entered blocking state
Tue Dec  5 11:57:39 2023 kern.info kernel: [   27.304054] br-lan: port 1(eth1) entered forwarding state
Tue Dec  5 11:57:39 2023 daemon.notice netifd: Network device 'eth1' link is up
Tue Dec  5 11:57:39 2023 daemon.notice netifd: bridge 'br-lan' link is up
Tue Dec  5 11:57:39 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Dec  5 11:57:39 2023 daemon.notice procd: /etc/rc.d/S90turboacc: sh: you need to specify whom to kill
Tue Dec  5 11:57:39 2023 daemon.notice procd: /etc/rc.d/S90turboacc: Stop DNS Caching
Tue Dec  5 11:57:39 2023 daemon.notice procd: /etc/rc.d/S90turboacc: DNSMASQ change
Tue Dec  5 11:57:42 2023 kern.info kernel: [   29.450026] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Tue Dec  5 11:57:42 2023 kern.info kernel: [   29.458612] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Tue Dec  5 11:57:42 2023 daemon.notice netifd: Network device 'eth0' link is up
Tue Dec  5 11:57:42 2023 daemon.notice netifd: Interface 'wan' has link connectivity
Tue Dec  5 11:57:42 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Dec  5 11:57:42 2023 daemon.notice netifd: Interface 'wan6' has link connectivity
Tue Dec  5 11:57:42 2023 daemon.notice netifd: Interface 'wan6' is setting up now
Tue Dec  5 11:57:42 2023 daemon.notice netifd: wan (3196): udhcpc: started, v1.36.0
Tue Dec  5 11:57:42 2023 daemon.err odhcp6c[3201]: Failed to send RS (Address not available)
Tue Dec  5 11:57:42 2023 daemon.notice netifd: wan (3196): udhcpc: broadcasting discover
Tue Dec  5 11:57:43 2023 daemon.err odhcp6c[3201]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Tue Dec  5 11:57:43 2023 daemon.err modprobe: failed to find a module named aufs
Tue Dec  5 11:57:43 2023 daemon.err dockerd[1882]: time="2023-12-05T11:57:43.138870462+08:00" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Tue Dec  5 11:57:43 2023 daemon.err dockerd[1882]: time="2023-12-05T11:57:43.141274385+08:00" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Tue Dec  5 11:57:43 2023 daemon.err dockerd[1882]: time="2023-12-05T11:57:43.212295077+08:00" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay2
Tue Dec  5 11:57:43 2023 daemon.err dockerd[1882]: time="2023-12-05T11:57:43.214370847+08:00" level=error msg="exec: \"fuse-overlayfs\": executable file not found in $PATH" storage-driver=fuse-overlayfs
Tue Dec  5 11:57:43 2023 kern.err kernel: [   30.653117] overlayfs: filesystem on '/opt/check-overlayfs-support2469344384/upper' not supported as upperdir
Tue Dec  5 11:57:43 2023 daemon.err dockerd[1882]: time="2023-12-05T11:57:43.237146077+08:00" level=warning msg="Could not load necessary modules for IPSEC rules: protocol not supported"
Tue Dec  5 11:57:43 2023 daemon.info modprobe: nf_conntrack is already loaded
Tue Dec  5 11:57:43 2023 daemon.info modprobe: nf_conntrack_netlink is already loaded
[... 76 lines cut ...]
Tue Dec  5 12:18:03 2023 daemon.notice netifd: wan (10692): udhcpc: received SIGTERM
Tue Dec  5 12:18:03 2023 daemon.notice netifd: wan (10692): udhcpc: entering released state
Tue Dec  5 12:18:03 2023 daemon.notice netifd: wan (10692): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Tue Dec  5 12:18:03 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Dec  5 12:18:03 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Dec  5 12:18:03 2023 daemon.notice netifd: wan (11133): udhcpc: started, v1.36.0
Tue Dec  5 12:18:03 2023 daemon.notice netifd: wan (11133): udhcpc: broadcasting discover
Tue Dec  5 12:18:07 2023 daemon.notice netifd: wan (11133): udhcpc: broadcasting discover
Tue Dec  5 12:18:08 2023 daemon.notice netifd: Interface 'wan6' is now down
Tue Dec  5 12:18:08 2023 daemon.notice netifd: Interface 'wan6' is setting up now
Tue Dec  5 12:18:08 2023 daemon.notice netifd: wan (11133): udhcpc: received SIGTERM
Tue Dec  5 12:18:08 2023 daemon.notice netifd: wan (11133): udhcpc: entering released state
Tue Dec  5 12:18:08 2023 daemon.notice netifd: wan (11133): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Tue Dec  5 12:18:08 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Dec  5 12:18:08 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Dec  5 12:18:08 2023 daemon.notice netifd: wan (11579): udhcpc: started, v1.36.0
Tue Dec  5 12:18:09 2023 daemon.notice netifd: wan (11579): udhcpc: broadcasting discover
Tue Dec  5 12:18:12 2023 daemon.notice netifd: wan (11579): udhcpc: broadcasting discover
Tue Dec  5 12:18:15 2023 daemon.notice netifd: wan (11579): udhcpc: broadcasting discover
Tue Dec  5 12:18:26 2023 daemon.notice netifd: Interface 'wan6' is now up
Tue Dec  5 12:18:26 2023 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0)
Tue Dec  5 12:18:27 2023 daemon.warn odhcpd[1907]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
```

更換官方固件，是沒有問題的，網絡沒有問題的（物理意義上）

Rurikobaka, 2023-12-05:
PS：有 `Docker` ，沒有 `nftables` ，有 `ipv6help`

aiamadeus, 2023-12-05:
完整的内核日志有吗

Rurikobaka, 2023-12-05:
> 完整的内核日志有吗

Kernel Log

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.138 (ruriko@Ruriko) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6275-d7d01fb74) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP Mon Dec 4 13:36:56 2023
[    0.000000] Machine model: GL.iNet GL-MT2500
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x0000000047c7ffff]
[    0.000000]   node   0: [mem 0x0000000047c80000-0x0000000047ffffff]
[    0.000000]   node   0: [mem 0x0000000048000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 17 pages/cpu s31896 r8192 d29544 u69632
[    0.000000] pcpu-alloc: s31896 r8192 d29544 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line:  root=PARTLABEL=rootfs rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1011964K/1048576K available (8512K kernel code, 934K rwdata, 2232K rodata, 448K init, 305K bss, 36612K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000154] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000164] pid_max: default: 32768 minimum: 301
[    0.000264] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000277] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001409] rcu: Hierarchical SRCU implementation.
[    0.001534] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.001753] smp: Bringing up secondary CPUs ...
[    0.002075] Detected VIPT I-cache on CPU1
[    0.002101] GICv3: CPU1: found redistributor 1 region 0:0x000000000c0a0000
[    0.002130] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.002197] smp: Brought up 1 node, 2 CPUs
[    0.002210] SMP: Total of 2 processors activated.
[    0.002216] CPU features: detected: 32-bit EL0 Support
[    0.002221] CPU features: detected: CRC32 instructions
[    0.002328] CPU: All CPU(s) started at EL2
[    0.002340] alternatives: patching kernel code
[... 148 lines cut ...]
[   22.488388] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   22.496728] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   23.507129] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[   23.621784] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   23.634174] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode
[   23.648894] br-lan: port 1(eth1) entered blocking state
[   23.654166] br-lan: port 1(eth1) entered disabled state
[   23.659563] device eth1 entered promiscuous mode
[   23.665602] br-lan: port 1(eth1) entered blocking state
[   23.670870] br-lan: port 1(eth1) entered forwarding state
[   23.686341] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:05] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
[   23.696611] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/sgmii link mode
[   24.638329] br-lan: port 1(eth1) entered disabled state
[... cut ...]
[   30.532540] overlayfs: filesystem on '/opt/check-overlayfs-support993665072/upper' not supported as upperdir
[   36.169954] Maxlinear Ethernet GPY211C mdio-bus:05: Downshift occurred from negotiated speed 100Mbps to actual speed 10Mbps, check cabling!
[   36.182501] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 10Mbps/Half - flow control off
[   36.190878] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   37.209775] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[   45.530057] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
```

感謝~

aiamadeus, 2024-01-27:
GL 的工程师说可能是硬件问题，已知 mt6000 也有类似的情况出现。

coolsnowwolf, 2026-09-05:
已核对日志与当前代码。该报告使用 2023-12-05 的 5.15.138 自定义固件，日志中 eth0（GPY211C）先协商到 1Gbps，随后明确出现 “Downshift occurred ... actual speed 10Mbps, check cabling!” 并反复 Link is Down/Up；同时存在 dockerd/haproxy/sysntpd crash loop，且官方固件正常。当前仓库没有可由该日志定位的通用 WAN/驱动补丁缺失，现象更符合链路协商/线缆或该设备的旧版自定义配置问题，无法在现行代码中复现。请使用当前版本、最小配置及完整复现步骤重新报告；本条关闭为 not planned。

closed 2026-09-05 by coolsnowwolf