coolsnowwolf/lede #11696: 拉取最新源码编译6.1内核R4S固件报错

bi4tms, 2023-12-09:
拉取最新源码编译6.1内核R4S固件报错，具体代码如下：
AR      drivers/hwspinlock/built-in.a
CC      drivers/devfreq/event/rockchip-dfi.o
drivers/devfreq/event/rockchip-dfi.c: In function 'rockchip_dfi_init':
drivers/devfreq/event/rockchip-dfi.c:565:13: warning: unused variable 'val' [-Wunused-variable]
  565 |         u32 val;
      |             ^~~
drivers/devfreq/event/rockchip-dfi.c: At top level:
drivers/devfreq/event/rockchip-dfi.c:609:19: error: 'rockchip_dfi_probe' undeclared here (not in a function); did you mean 'rockchip_dfi_driver'?
  609 |         .probe  = rockchip_dfi_probe,
      |                   ^~~~~~~~~~~~~~~~~~
      |                   rockchip_dfi_driver
drivers/devfreq/event/rockchip-dfi.c:612:35: error: 'rockchip_dfi_id_match' undeclared here (not in a function); did you mean 'rockchip_dfi_init'?
  612 |                 .of_match_table = rockchip_dfi_id_match,
      |                                   ^~~~~~~~~~~~~~~~~~~~~
      |                                   rockchip_dfi_init
drivers/devfreq/event/rockchip-dfi.c:559:19: warning: 'rockchip_dfi_init' defined but not used [-Wunused-function]
  559 | static __init int rockchip_dfi_init(struct platform_device *pdev,
      |                   ^~~~~~~~~~~~~~~~~
drivers/devfreq/event/rockchip-dfi.c:541:19: warning: 'rk3368_dfi_init' defined but not used [-Wunused-function]
  541 | static __init int rk3368_dfi_init(struct platform_device *pdev,
      |                   ^~~~~~~~~~~~~~~
drivers/devfreq/event/rockchip-dfi.c:504:19: warning: 'rk3288_dfi_init' defined but not used [-Wunused-function]
  504 | static __init int rk3288_dfi_init(struct platform_device *pdev,
      |                   ^~~~~~~~~~~~~~~
drivers/devfreq/event/rockchip-dfi.c:486:19: warning: 'rk3128_dfi_init' defined but not used [-Wunused-function]
  486 | static __init int rk3128_dfi_init(struct platform_device *pdev,
      |                   ^~~~~~~~~~~~~~~
drivers/devfreq/event/rockchip-dfi.c:456:19: warning: 'px30_dfi_init' defined but not used [-Wunused-function]
  456 | static __init int px30_dfi_init(struct platform_device *pdev,
      |                   ^~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:250: drivers/devfreq/event/rockchip-dfi.o] Error 1
make[8]: *** [scripts/Makefile.build:500: drivers/devfreq/event] Error 2
make[7]: *** [scripts/Makefile.build:500: drivers/devfreq] Error 2
make[6]: *** [scripts/Makefile.build:500: drivers] Error 2
make[5]: *** [Makefile:2014: .] Error 2
make[5]: Leaving directory '/home/r4s.6.1/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.66'
make[4]: *** [Makefile:26: /home/r4s.6.1/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.66/.modules] Error 2
make[4]: Leaving directory '/home/r4s.6.1/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/r4s.6.1/lede/target/linux'
time: target/linux/compile#1903.32#151.21#2058.01
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/r4s.6.1/lede'
make[1]: *** [target/Makefile:23: /home/r4s.6.1/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/r4s.6.1/lede'
make: *** [/home/r4s.6.1/lede/include/toplevel.mk:231：world] 错误 2
编译R2S固件同样报错

jh0072110, 2023-12-09:
r5s也一样哈哈哈

coolsnowwolf, 2023-12-09:
fixed

closed 2023-12-10 by bi4tms