coolsnowwolf/lede #11697: 大佬。我想切回5.15.141的kernel内核。。我应该切到哪个版本,

xiaolongzhou123, 2023-12-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

我想切到5.15.151的kernel内核。

WYC-2020, 2023-12-12:
改对应平台的 makefile 稍微看下他们那些切换内核的commit 就知道如何改的 比如改x86 就是 target/linux/x86/Makefile 这 里面的
[image]

xiaolongzhou123, 2023-12-12:
大佬。小版本号呢？我看了一下在include目录 下：LINUX_KERNEL_HASH-5.10.xxx=有个hash值。我在哪里找这个值？

而且 这个包是从哪里下载的？

WYC-2020, 2023-12-12:
> 大佬。小版本号呢？我看了一下在include目录 下：LINUX_KERNEL_HASH-5.10.xxx=有个hash值。我在哪里找这个值？
> 
> 而且 这个包是从哪里下载的？

小版本 自己改 include 里面 的hash 并且有可能path还要更新看你运气，看你能问出这么小白的问题 怀疑你肯定搞不定https://github.com/coolsnowwolf/lede/blob/master/include/kernel-5.15

closed 2023-12-17 by aiamadeus