coolsnowwolf/lede #11700: 容器资源状态获取异常 NaN undefined / NaN undefined (0%)

kxdn3, 2023-12-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86 6.1最新源码固件

### (3) 详细日志

A：
麻烦大佬看下这个lssue 
https://github.com/lisaac/luci-app-dockerman/issues/175
docker容器资源状态获取异常，应该就是cgroupv2和cgroupv1同时挂载导致冲突，有好的解决方案吗？还是只能禁用cgroupfs-mount

WYC-2020, 2023-12-28:
这个还真没遇到过，我编译x86 6.1编译出来根本就没有cgroupv2的不知道你开了什么选项 会有这个玩意

kxdn3, 2024-01-17:
> 这个还真没遇到过，我编译x86 6.1编译出来根本就没有cgroupv2的不知道你开了什么选项 会有这个玩意

cgroupfs-mount包含了v1v2

OldCoding, 2024-01-23:
参考这个https://github.com/lisaac/luci-app-dockerman/issues/175

kxdn3, 2024-04-01:
每次只能关闭cgroupfs-mount启动

coolsnowwolf, 2026-09-05:
该问题引用的是 2023 年旧版 `luci-app-dockerman` 外部 issue，根因取决于设备同时启用 cgroup v1/v2 的挂载策略，不是 LuCI 页面本身可统一修复的错误。当前 LEDE 的 Docker 包通过 `cgroupfs-mount` 配置选择挂载方式，使用 cgroup v2 时应禁用 legacy cgroupfs-mount，避免重复挂载。缺少具体 `/proc/mounts`、Docker daemon 配置和内核 cgroup 日志，无法制作通用补丁。

closed 2026-09-05 by coolsnowwolf