coolsnowwolf/lede #11702: 怎样占满整个emmc?

392377870, 2023-12-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：16g的emmc刷完后只占用2g，留下14g空白末格式化盘，编译时修改哪个地方能刷机后占用整个emmc？

shenlijun, 2024-01-12:
编译ext4格式的固件第一次启动会自动占满应该，就算没占满，也可以用cfdisk搞。
设置编译哪种硬盘格式那里再往下可以设置选分区大小

事实上完全没必要，何必折磨自己,我自己编译的x86只有300m左右，另一个arm的只有20m，一样该有的功能都全了

bzi-bit, 2024-01-27:
参考这个[https://openwrt.org/docs/guide-user/installation/openwrt_x86#expanding_root_partition_and_filesystem](url)

coolsnowwolf, 2026-09-05:
eMMC 镜像只写入固定大小的 kernel/rootfs 分区是 OpenWrt 的正常布局，剩余空间作为未分区或 data 分区保留，不会在编译时自动“占满”。不同设备的 GPT/分区表和 bootloader 约束不同，不能在通用 LEDE 镜像中盲目扩大 rootfs；应按具体设备文档扩展 data 分区并使用 `resize2fs`/文件系统工具，或在首次启动脚本中安全调整。该 issue 未提供设备型号、分区表或镜像格式，无法提交通用代码修改，关闭。

closed 2026-09-05 by coolsnowwolf