coolsnowwolf/lede #11703: X86平台 增加 kmod-video-core 编译失败

xbf321, 2023-12-14:
选择了kmod-video-core就会出错，提示缺少 mc.ko

内核版本：6.1

NOTICE: module '/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.62/drivers/usb/host/ehci-hcd.ko' is built-in.
Packaged contents of /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb-ehci into /openwrt/lede/bin/targets/x86/64/packages/kmod-usb-ehci_6.1.62-1_x86_64.ipk
removed '/openwrt/lede/bin/targets/x86/64/packages/kmod-usb2_6.1.62-1_x86_64.ipk'
rstrip.sh: /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/6.1.62/ehci-platform.ko: relocatable
rstrip.sh: /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/6.1.62/ehci-fsl.ko: relocatable
rstrip.sh: /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2/lib/modules/6.1.62/fsl-mph-dr-of.ko: relocatable
Packaged contents of /openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-usb2 into /openwrt/lede/bin/targets/x86/64/packages/kmod-usb2_6.1.62-1_x86_64.ipk
removed '/openwrt/lede/bin/targets/x86/64/packages/kmod-usb-audio_6.1.62-1_x86_64.ipk'
Package kmod-usb-audio is missing dependencies for the following libraries:
mc.ko
make[3]: *** [modules/usb.mk:562: /openwrt/lede/bin/targets/x86/64/packages/kmod-usb-audio_6.1.62-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/openwrt/lede/package/kernel/linux'
time: package/kernel/linux/compile#29.56#11.15#47.42
ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/openwrt/lede'
make[1]: *** [package/Makefile:110: /openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/openwrt/lede'
make: *** [/openwrt/lede/include/toplevel.mk:231: world] Error 2

aiamadeus, 2023-12-14:
Try `git revert 0f4df4fa226e7d3808ea47c5da77130ec89068f2`

xbf321, 2023-12-15:
好的，我试试

xbf321, 2023-12-15:
我看最近的文件，已经更新，但还是不行
[image]

lxg2001, 2023-12-15:
你的报错里是 kmod-usb-audio 的问题，跟之前已经修复的 kmod-usb-video 都是一样的的问题，都是缺少“mc.ko”，把 kmod-usb-audio 用一样的方法修复，倒是能编译成功，但是在安装软件时还是会失败，所以在同时选择这两个软件时，我做了一个补丁来临时解决问题，把补丁放到 lede/target/linux/x86/patches-6.1
补丁文件（把txt改成patch）：
[1.txt](https://github.com/coolsnowwolf/lede/files/13683611/1.txt)

PR #11713 (2023-12-17): kernel: fixes usb-audio depends
Fixes: #11703
commit 783f460c (2023-12-17): kernel: fixes usb-audio depends (#11713)

Fixes: #11703
closed 2023-12-17 by aiamadeus