coolsnowwolf/lede #11705: Failed to send SOLICIT message to ff02::1:2

lqh280, 2023-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* 其它人的issue，也有提到这个问题，但没看到有解决的。

## 2. 详细叙述

### (1) 具体问题

A：看日志像是IPV6导致断网（有IPV4地址)

### (2) 路由器型号和固件版本

A：X86软件路由，用LEDE源码，自己编译的。OpenWrt R23.10.1 / LuCI Master (git-23.291.46448-4898ed2)

### (3) 详细日志

Fri Dec 15 03:00:31 2023 daemon.info pppd[2493]: Using interface pppoe-wan
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: Connect: pppoe-wan <--> eth1
Fri Dec 15 03:00:31 2023 daemon.info pppd[2493]: syncppp not active
Fri Dec 15 03:00:31 2023 daemon.info pppd[2493]: Remote message: Authentication Successful,Welcome!
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: PAP authentication succeeded
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: peer from calling number 3A:**:69:**:FF:0D authorized
Fri Dec 15 03:00:31 2023 daemon.notice ttyd[10671]: [2023/12/15 03:00:31:8062] N: rops_handle_POLLIN_netlink: DELADDR
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: local  IP address 113.116.133.**
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: remote IP address 113.116.132.1
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: primary   DNS address 202.96.134.133
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: secondary DNS address 202.96.128.166
Fri Dec 15 03:00:31 2023 daemon.info avahi-daemon[2926]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 113.116.133.**.
Fri Dec 15 03:00:31 2023 daemon.info avahi-daemon[2926]: New relevant interface pppoe-wan.IPv4 for mDNS.
Fri Dec 15 03:00:31 2023 daemon.info avahi-daemon[2926]: Registering new address record for 113.116.133.** on pppoe-wan.IPv4.
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: local  LL address fe80::02e0:4c7f:2b6f:505b
Fri Dec 15 03:00:31 2023 daemon.notice pppd[2493]: remote LL address fe80::3826:69ff:fe01:ff0d
Fri Dec 15 03:00:31 2023 daemon.info avahi-daemon[2926]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::2e0:4c7f:2b6f:505b.
Fri Dec 15 03:00:31 2023 daemon.info avahi-daemon[2926]: New relevant interface pppoe-wan.IPv6 for mDNS.
Fri Dec 15 03:00:31 2023 daemon.info avahi-daemon[2926]: Registering new address record for fe80::2e0:4c7f:2b6f:505b on pppoe-wan.*.
Fri Dec 15 03:00:31 2023 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Dec 15 03:00:31 2023 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Fri Dec 15 03:00:31 2023 daemon.notice netifd: Interface 'wan_6' has link connectivity
Fri Dec 15 03:00:31 2023 daemon.notice netifd: Interface 'wan' is now up
Fri Dec 15 03:00:31 2023 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Dec 15 03:00:31 2023 daemon.err odhcp6c[28043]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Fri Dec 15 03:00:31 2023 daemon.notice netifd: Interface 'wan_6' is now down
Fri Dec 15 03:00:31 2023 daemon.notice netifd: Interface 'wan_6' is disabled
Fri Dec 15 03:00:31 2023 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Fri Dec 15 03:00:31 2023 daemon.info avahi-daemon[2926]: Registering new address record for fe80::2e0:4cff:fe6f:505a on br-lan.*.
Fri Dec 15 03:00:32 2023 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Fri Dec 15 03:00:35 2023 daemon.err nmbd[27693]: [2023/12/15 03:00:35.059626,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Fri Dec 15 03:00:35 2023 daemon.err nmbd[27693]:   *****
Fri Dec 15 03:00:35 2023 daemon.err nmbd[27693]:
Fri Dec 15 03:00:35 2023 daemon.err nmbd[27693]:   Samba name server OPENWRT is now a local master browser for workgroup WORKGROUP on subnet 192.168.2.1
Fri Dec 15 03:00:35 2023 daemon.err nmbd[27693]:
Fri Dec 15 03:00:35 2023 daemon.err nmbd[27693]:   *****

yanxin152133, 2024-09-09:
编译的时候选开启`ipv6helper`了吗
[image]

coolsnowwolf, 2026-09-05:
该日志显示 PPPoE 已获取 IPv4 和链路本地 IPv6，但 odhcp6c 向 `ff02::1:2` 发送 SOLICIT 时返回 `Permission denied`，随后接口被 netifd 禁用。现有信息缺少 `/etc/config/network`、防火墙 IPv6 规则、内核 IPv6 状态和 `odhcp6c -v` 诊断，无法证明是 odhcp6c 或 LEDE 源码缺陷；评论也指出需确认是否启用了 ipv6helper。当前 odhcp6c/netifd 实现没有对应待修复回归，建议按配置/防火墙排查，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf