coolsnowwolf/lede #11706: 没编译IPV6的情况下...概览的IPv6 WAN 状态会显示

rrkai, 2023-12-15:
之前是不会的..只要没把ipv6的编译进去 ,概览的IPv6 WAN 状态就会消失.
现在新编译的,就算没把ipv6helper和dnsmasq_full_dhcpv6编译进去.
在概览的IPv6 WAN 状态还是会显示出来.
哪个大佬知道怎么去掉么..

itfish92, 2023-12-17:
先rm -rf .config
然后在 config/Config-build.in 文件里面找到：
    　　config IPV6
    　　　　def_bool y

后面那行改成    　　def_bool n
重新make menuconfig ，里面就自动把ipv6全部disable了

上面的教程我也是本源码里Issues里翻出来的。实测有效

rrkai, 2023-12-17:
> 先rm -rf .config 然后在 config/Config-build.in 文件里面找到： 　　config IPV6 　　　　def_bool y
> 
> 后面那行改成 　　def_bool n 重新make menuconfig ，里面就自动把ipv6全部disable了
> 
> 上面的教程我也是本源码里Issues里翻出来的。实测有效

这样的话 Turbo ACC插件和docker也编译不进去...

itfish92, 2023-12-17:
> > 先rm -rf .config 然后在 config/Config-build.in 文件里面找到： 　　config IPV6 　　　　def_bool y
> > 后面那行改成 　　def_bool n 重新make menuconfig ，里面就自动把ipv6全部disable了
> > 上面的教程我也是本源码里Issues里翻出来的。实测有效
> 
> 这样的话 Turbo ACC插件和docker也编译不进去...

我这边没问题的。。。多学习吧

yanxin152133, 2023-12-18:
周一凌晨刚编译完，没有这种情况

rrkai, 2023-12-18:
> 周一凌晨刚编译完，没有这种情况

我知道是什么原因了...docker依赖ip6防火墙了....

coolsnowwolf, 2026-09-05:
当前 LuCI 状态页仅通过 `network.getWAN6Networks()` 查找实际存在的 IPv6 默认路由（`::/0`），只有找到对应 WAN6 网络才渲染“IPv6 Upstream”卡片；代码位于 `luci-mod-status/.../30_network.js` 与 `luci-base/.../network.js`。因此即使未安装 ipv6helper/dnsmasq DHCPv6，只要系统仍有 `wan6` 配置或 IPv6 默认路由，显示该卡片是预期行为。删除/禁用 `wan6` 配置或移除 IPv6 默认路由即可隐藏。issue 未提供状态接口输出，当前无可复现缺陷，关闭。

closed 2026-09-05 by coolsnowwolf