coolsnowwolf/lede #11714: 拉取最新源码编译5.15内核R4S固件报错

bi4tms, 2023-12-16:
拉取最新源码编译5.15内核R4S固件报错，具体日志如下：
add-more-soc-support.patch using plaintext: 
patching file drivers/devfreq/event/rockchip-dfi.c
Hunk #8 FAILED at 582.
1 out of 8 hunks FAILED -- saving rejects to file drivers/devfreq/event/rockchip-dfi.c.rej
Patch failed!  Please fix /home/r4s.5.15/lede/target/linux/rockchip/patches-5.15/805-PM-devfreq-rockchip-dfi-add-more-soc-support.patch!
make[4]: *** [Makefile:34: /home/r4s.5.15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.142/.prepared_7c21fbed09dba84fc0a4f9436010c7cd] Error 1
make[4]: Leaving directory '/home/r4s.5.15/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/r4s.5.15/lede/target/linux'
time: target/linux/compile#17.08#11.17#27.66
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/r4s.5.15/lede'
make[1]: *** [target/Makefile:23: /home/r4s.5.15/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/r4s.5.15/lede'
make: *** [/home/r4s.5.15/lede/include/toplevel.mk:231：world] 错误 2
编译R2S的5.15内核固件报同样错误
@coolsnowwolf

BenjaminX, 2023-12-17:
同样的编译错误

DanielChen008, 2023-12-17:
+1,删除了文件夹重新 clone 的也是到这里就报错结束。

qx100, 2023-12-17:
使用 Vmware Ubuntu 20.04 单线程编译R5S 固件以及github action工作流均可以复现同样的错误

BenjaminX, 2023-12-17:
[image]

liuyanantj, 2023-12-17:
现在最新的143内核编译没问题了

DanielChen008, 2023-12-17:
> 现在最新的143内核编译没问题了
已核实，大大们周末修改了代码，现在已经可以编译成功。

closed 2023-12-17 by coolsnowwolf

LazyBoBo, 2023-12-19:
我还在报这个错