coolsnowwolf/lede #11724: 玩客云 Thunder OneCloud 解析烧入固件失败

miaoermua, 2023-12-21:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：玩客云老母鸡赚钱宝 3 代，按照 LEDE 默认配置编译 Kernel 6.1 使用 [Amlogic_USB_Burning_Tool_v2.1.6.8](https://androiddatahost.com/wp-content/uploads/Amlogic_USB_Burning_Tool_v2.1.6.8.zip) 解析烧入固件失败，是不是只能使用 USB 读盘引导？不能直写 eMMC？

使用版本: https://github.com/coolsnowwolf/lede/commit/2baf4cdc288c49ca8bb0f5e077517b482276ace6 昨天的

### (2) 路由器型号和固件版本

Target System (Amlogic Meson family)  --->
Subtarget (Amlogic S805 boards (32 bit))  --->
Target Profile (Thunder OneCloud)  --->

### (3) 详细日志

日志正常，没看出明显异常。

```
Signing package index...
make[2]: Leaving directory '/home/cat/lede'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/cat/lede'
WORK_DIR=/home/cat/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/json_info_files /home/cat/lede/scripts/json_overview_image_info.py /home/cat/lede/bin/targets/meson/meson8b/profiles.json 
make[2]: Leaving directory '/home/cat/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/cat/lede'
make[2]: Leaving directory '/home/cat/lede'
make[1]: Leaving directory '/home/cat/lede'
cat@buildcatwrt:~/lede$
```

miaoermua, 2023-12-29:
> lede上的onecloud，不能直接线刷

只能 u 盘写盘引导咯那就是

closed 2023-12-29 by miaoermua

miaoermua, 2024-12-07:
> > > lede上的onecloud，不能直接线刷
> > 
> > 
> > 只能 u 盘写盘引导咯那就是
> 
> 是的，但是在此之前，你需要刷[hzyitc](https://github.com/hzyitc/u-boot-onecloud)的uboot，然后默认 usb -> sd -> emmc 的顺序启动 当然你也可以将[hzyitc](https://github.com/hzyitc/u-boot-onecloud)的uboot放到你的op固件中，用于直接线刷
> 
> [shiyu1314](https://github.com/shiyu1314/openwrt-onecloud) 的固件可以直接线刷，但是他修改了[hzyitc](https://github.com/hzyitc/u-boot-onecloud)的uboot https://github.com/shiyu1314/u-boot-onecloud，只能emmc启动...

今天花了一点时间解决了，把编译好的镜像提取出 0.fat & 1.img，然后用 UBOOT 作者另外一个项目 AmlImg 组装好重新写盘就成功完成线刷了，非常感谢！~~（就是环境弄的有点烦……这个 img2simg 不好弄……）~~

flyaskite, 2025-03-16:
@miaoermua 我也碰到这个问题了。请教一下具体怎么，将编译好的镜像提取文件，并和uboot重新组装呢？谢谢！

miaoermua, 2025-03-17:
> @miaoermua 我也碰到这个问题了。请教一下具体怎么，将编译好的镜像提取文件，并和uboot重新组装呢？谢谢！
> 
> 

我参考了小鸡过河的源码，将 uboot 通过工具组装起来，确实是可以运行，第一次开机需要 3 分钟然后会自动重启，但是 ext4 的包我很少折腾，会出现无法保存配置的问题，应该是要改挂载点。现在机器还在吃灰，没时间弄这个了。

flyaskite, 2025-03-23:
好的，谢谢！