coolsnowwolf/lede #11725: QHora-301W编译失败

schen39, 2023-12-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
没有
## 2. 详细叙述
编译失败，貌似缺依赖
### (1) 具体问题

A：编译失败，貌似缺依赖

### (2) 路由器型号和固件版本

A：QHora-301W/5.10

### (3) 详细日志

A：compilation terminated.
Makefile:18: recipe for target '/home/scott/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8072-ax9000.dtb' failed
make[5]: *** [/home/scott/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8072-ax9000.dtb] Error 1
make[5]: Leaving directory '/home/scott/lede/target/linux/ipq807x/image'
Makefile:26: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/scott/lede/target/linux/ipq807x'
Makefile:11: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/scott/lede/target/linux'
time: target/linux/install#57.91#25.39#83.28
    ERROR: target/linux failed to build.
target/Makefile:28: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 1
make[2]: Leaving directory '/home/scott/lede'
target/Makefile:24: recipe for target '/home/scott/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install' failed
make[1]: *** [/home/scott/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/scott/lede'
/home/scott/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

WYC-2020, 2023-12-22:
算不出来

schen39, 2023-12-22:
> 算不出来

有看到大雕和小宝合作开发istore固件，是否没开源，已经口水了

WYC-2020, 2023-12-25:
> > 算不出来
> 
> 有看到大雕和小宝合作开发istore固件，是否没开源，已经口水了

不清楚，这个开源的x86已经完全够我用了，因为我只玩软路由 硬路由那块不关心 软路由驱动都不用担心 所以够用了

plasmolysis369, 2023-12-31:
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/include -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/include/ -undef -D__DTS__  -o /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8072-301w.dtb.tmp /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8072-301w.dts
/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/scripts/dtc/dtc -O dtb -i/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8072-301w.dtb /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8072-301w.dtb.tmp
/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8074.dtsi:183.31-188.5: Warning (interrupt_provider): /smp2p-wcss/slave-kernel: Missing #address-cells in interrupt provider
/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8074.dtsi:403.25-451.5: Warning (interrupt_provider): /soc/pinctrl@1000000: Missing #address-cells in interrupt provider
  also defined at /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8072-301w.dts:161.7-298.3
/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8074.dtsi:758.38-763.5: Warning (interrupt_provider): /soc/interrupt-controller@b000000: Missing #address-cells in interrupt provider
rm -f /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8072-301w.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/include -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/include/ -undef -D__DTS__  -o /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8071-ax6.dtb.tmp /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8071-ax6.dts
/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/scripts/dtc/dtc -O dtb -i/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8071-ax6.dtb /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8071-ax6.dtb.tmp
/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8074.dtsi:183.31-188.5: Warning (interrupt_provider): /smp2p-wcss/slave-kernel: Missing #address-cells in interrupt provider
/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8074.dtsi:403.25-451.5: Warning (interrupt_provider): /soc/pinctrl@1000000: Missing #address-cells in interrupt provider
  also defined at /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8071-ax3600.dtsi:53.7-69.3
/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8074.dtsi:758.38-763.5: Warning (interrupt_provider): /soc/interrupt-controller@b000000: Missing #address-cells in interrupt provider
rm -f /home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8071-ax6.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/include -I/home/happer/op/arm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/include/ -undef -D__DTS__  -o /home/h [... cut ...]
make[3]: Leaving directory '/home/happer/op/arm/lede/target/linux'
time: target/linux/install#30.45#22.29#49.01
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/happer/op/arm/lede'
make[1]: *** [target/Makefile:24: /home/happer/op/arm/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/happer/op/arm/lede'
make: *** [/home/happer/op/arm/lede/include/toplevel.mk:231: world] Error 2

schen39, 2024-01-05:
已编译成功

Sense1988, 2024-01-16:
> 已编译成功

遇到同样的报错,请问是如何解决的?

schen39, 2024-01-16:
> > 已编译成功
> 
> 遇到同样的错误，请问如何解决？

https://github.com/coolsnowwolf/lede/issues/11744

closed 2024-02-02 by schen39