coolsnowwolf/lede #11726: 求助，编译错误

jxph1123, 2023-12-21:
```
aarch64-openwrt-linux-musl-gcc -L/home/etek/lede-6.1.68/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/home/etek/lede-6.1.68/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro  -shared -o parser.so template_parser.o template_utils.o template_lmo.o template_lualib.o
./mkversion.sh version.lua git-23.346.45702-3e8eb4a "LuCI Master"
/bin/sh: 1: ./mkversion.sh: Permission denied
make[4]: *** [Makefile:14: version.lua] Error 126
make[4]: Leaving directory '/home/etek/lede-6.1.68/build_dir/target-aarch64_generic_musl/luci-base/src'
make[3]: *** [../../luci.mk:242: /home/etek/lede-6.1.68/build_dir/target-aarch64_generic_musl/luci-base/.built] Error 2
make[3]: Leaving directory '/home/etek/lede-6.1.68/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/compile#1.09#1.00#2.08
    ERROR: package/feeds/luci/luci-base failed to build.
make[2]: *** [package/Makefile:116: package/feeds/luci/luci-base/compile] Error 1
make[2]: Leaving directory '/home/etek/lede-6.1.68'
make[1]: *** [package/Makefile:110: /home/etek/lede-6.1.68/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/etek/lede-6.1.68'
make: *** [/home/etek/lede-6.1.68/include/toplevel.mk:231：world] 错误 2

```

WYC-2020, 2023-12-22:
./mkversion.sh: Permission denied 权限没得 看看这个脚本有没执行权限

jxph1123, 2023-12-23:
> ./mkversion.sh: Permission denied 权限没得 看看这个脚本有没执行权限

 很奇怪的问题，我用GitZip将luci-base单独下载进去编译，就会出现一楼的权限问题。重新删luci再./scripts/feeds update -a && ./scripts/feeds install -a，就没问题了，很奇怪的问题。。。
其他的很多用应使用GitZip下载过来没有这个问题。

closed 2023-12-23 by jxph1123