coolsnowwolf/lede #11731: 兆能m2（ipq60xx）编译luci-theme-argon错误（应该是）

Akimio521, 2023-12-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：兆能m2编译luci-theme-argon错误

### (2) 路由器型号和固件版本

A：兆能m2（ipq60xx）、lede（master）

### (3) 详细日志

A：
```shell
Collected errors:
 * resolve_conffiles: Existing conffile /home/akimio/lede/build_dir/target-aarch64_cortex-a53_musl/root-ipq60xx/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/akimio/lede/build_dir/target-aarch64_cortex-a53_musl/root-ipq60xx/etc/config/socat-opkg.
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/akimio/lede/build_dir/target-aarch64_cortex-a53_musl/root-ipq60xx/usr/lib/lua/luci/view/themes/argon/footer.htm
        But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/akimio/lede/build_dir/target-aarch64_cortex-a53_musl/root-ipq60xx/usr/lib/lua/luci/view/themes/argon/header.htm
        But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/akimio/lede/build_dir/target-aarch64_cortex-a53_musl/root-ipq60xx/www/luci-static/argon/favicon.ico
        But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/akimio/lede/build_dir/target-aarch64_cortex-a53_musl/root-ipq60xx/www/luci-static/argon/js/jquery.min.js
        But that file is already provided by package  * luci-theme-argon
 * check_data_file_clashes: Package luci-theme-argon-mod wants to install file /home/akimio/lede/build_dir/target-aarch64_cortex-a53_musl/root-ipq60xx/www/luci-static/argon/js/script.js
        But that file is already provided by package  * luci-theme-argon
 * opkg_install_cmd: Cannot install package luci-theme-argon-mod.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/akimio/lede'
make[1]: *** [package/Makefile:111: /home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/akimio/lede'
make: *** [/home/akimio/lede/include/toplevel.mk:231: world] Error 2
```

Akimio521, 2023-12-22:
在默认状态下编译好像都有问题
```shell
make[5]: Leaving directory '/home/akimio/lede/build_dir/hostpkg/ncurses-6.4/form'
make[4]: Leaving directory '/home/akimio/lede/build_dir/hostpkg/ncurses-6.4'
make -C /home/akimio/lede/build_dir/hostpkg/ncurses-6.4/progs tic
make[4]: Entering directory '/home/akimio/lede/build_dir/hostpkg/ncurses-6.4/progs'
echo "#ifndef __TRANSFORM_H"                                    >transform.h
echo "#define __TRANSFORM_H 1"                                  >>transform.h
echo "#include <progs.priv.h>"                                  >>transform.h
echo "extern bool same_program(const char *, const char *);"    >>transform.h
/usr/bin/env bash -c 'if test -n "" ; then echo "#define SUFFIX_IGNORED \"\"">>transform.h; fi'
echo "#define PROG_CAPTOINFO \"`echo captoinfo|   sed 's/$//'|sed 's,$,,;s,^,,'|sed 's/$//'`\""         >>transform.h
echo "#define PROG_INFOTOCAP \"`echo infotocap|   sed 's/$//'|sed 's,$,,;s,^,,'|sed 's/$//'`\""         >>transform.h
echo "#define PROG_CLEAR     \"`echo clear|       sed 's/$//'|sed 's,$,,;s,^,,'|sed 's/$//'`\""         >>transform.h
echo "#define PROG_RESET     \"`echo reset|       sed 's/$//'|sed 's,$,,;s,^,,'|sed 's/$//'`\""         >>transform.h
echo "#define PROG_INIT      \"`echo init|        sed 's/$//'|sed 's,$,,;s,^,,'|sed 's/$//'`\""         >>transform.h
echo "#endif /* __TRANSFORM_H */"                               >>transform.h
gcc -DHAVE_CONFIG_H -I../progs -I. -I../include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -DPIC -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DNDEBUG -O2 -fPIC --param max-inline-insns-single=1200  -DNCURSES_STATIC -c ../progs/tic.c -o ../objects/tic.o
/usr/bin/env bash ./MKtermsort.sh mawk ./../include/Caps >termsort.h
gcc -DHAVE_CONFIG_H -I../progs -I. -I../include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -DPIC -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DNDEBUG -O2 -fPIC --param max-inline-insns-single=1200  -DNCURSES_STATIC -c ../progs/dump_entry.c -o ../objects/dump_entry.o
gcc -DHAVE_CONFIG_H -I../progs -I. -I../include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -DPIC -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DNDEBUG -O2 -fPIC --param max-inline-insns-single=1200  -DNCURSES_STATIC -c ../progs/tparm_type.c -o ../objects/tparm_type.o
gcc -DHAVE_CONFIG_H -I../progs -I. -I../include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -DPIC -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DNDEBUG -O2 -fPIC --param max-inline-insns-single=1200  -DNCURSES_STATIC -c ../progs/transform.c -o ../objects/transform.o
gcc ../objects/tic.o ../objects/dump_entry.o ../objects/tparm_type.o ../objects/transform.o -L../lib -L/home/akimio/lede/staging_dir/host/lib -L/home/akimio/lede/staging_dir/hostpkg/lib -L/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/lib -DHAVE_CONFIG_H -I../progs -I. -I../include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -DPIC -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DNDEBUG -O2 -fPIC --param max-inline-insns-single=1200  -DNCURSES_STATIC -L../lib  -lncurses -lncurses    -o tic
make[4]: Leaving directory '/home/akimio/lede/build_dir/hostpkg/ncurses-6.4/progs'
touch /home/akimio/lede/build_dir/hostpkg/ncurses-6.4/.built
CFLAGS="-O2 -I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include -DPIC -fPIC" CPPFLAGS="-I/home/akimio/lede/staging_dir/host/include -I/home/akimio/lede/staging_dir/hostpkg/include -I/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/akimio/lede/staging_dir/host/lib -L/home/akimio/lede/staging_dir/hostpkg/lib -L/home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/host/lib" make -j1 -C /home/akimio/lede/build_dir/hostpkg/ncurses-6.4  install
make[4]: Entering directory '/home/akimio/lede/build_dir/hostpkg/ncurses-6.4'
( cd include && make -w -j1 DESTDIR="" RPATH_LIST="/home/akimio/lede/staging_dir/hostpkg/lib" install )
[... cut ...]
touch /home/akimio/lede/build_dir/hostpkg/ncurses-6.4/.built
touch /home/akimio/lede/staging_dir/hostpkg/stamp/.ncurses_installed
make[3]: Leaving directory '/home/akimio/lede/package/libs/ncurses'
time: package/libs/ncurses/host-compile#23.55#3.86#48.39
make[2]: Leaving directory '/home/akimio/lede'
make[1]: *** [package/Makefile:110: /home/akimio/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/akimio/lede'
make: *** [/home/akimio/lede/include/toplevel.mk:231: world] Error 2
```

lxg2001, 2023-12-22:
luci-theme-argon 和 luci-theme-argon-mod 选择其中一个就行了

Akimio521, 2023-12-22:
我直接编译也报错了，没有添加其他任何插件

closed 2023-12-22 by coolsnowwolf