coolsnowwolf/lede #11734: PVE LXC 安装的openwrt下，fullcone 未生效

rheatin, 2023-12-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：打开fullcone 高性能模式和兼容模式都不生效（已关闭windows防火墙，NAT测试是symmetric, 京东云，PCDN都是NAT4，关闭openclash也没有用）

### (2) 路由器型号和固件版本

A：OpenWrt R2023.12.21 / LuCI Master (git-23.346.45702-3e8eb4a) 
在X86_64 的PVE 8.1的系统下，使用LXC安装的openwrt LEDE

### (3) 详细日志

A：

jdnt, 2023-12-25:
LXC好像就是这样，特征取决于宿主系统。我之前为了使用wireguard也是在宿主系统下装了相应的包，LXC下的openwrt才能正常使用。可以试试

rheatin, 2023-12-25:
> LXC好像就是这样，特征取决于宿主系统。我之前为了使用wireguard也是在宿主系统下装了相应的包，LXC下的openwrt才能正常使用。可以试试

谢谢你，正在尝试，但是PVE8.1.3 安装报错，正在尝试解决问题：root@pve:~/netfilter-full-cone-nat# make
make -C /lib/modules/6.5.11-7-pve/build M=/root/netfilter-full-cone-nat modules
make[1]: Entering directory '/usr/src/linux-headers-6.5.11-7-pve'
  CC [M]  /root/netfilter-full-cone-nat/xt_FULLCONENAT.o
/root/netfilter-full-cone-nat/xt_FULLCONENAT.c: In function ‘find_appropriate_port’:
/root/netfilter-full-cone-nat/xt_FULLCONENAT.c:427:24: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’? [-Werror=implicit-function-declaration]
  427 |     start = (uint16_t)(prandom_u32() % (u32)range_size);
      |                        ^~~~~~~~~~~
      |                        get_random_u32
/root/netfilter-full-cone-nat/xt_FULLCONENAT.c: In function ‘fullconenat_tg_check’:
/root/netfilter-full-cone-nat/xt_FULLCONENAT.c:648:22: error: ‘struct nf_ct_event_notifier’ has no member named ‘fcn’
  648 |     ct_event_notifier.fcn = ct_event_cb;
      |                      ^
/root/netfilter-full-cone-nat/xt_FULLCONENAT.c:651:9: error: void value not ignored as it ought to be
  651 |     if (nf_conntrack_register_notifier(par->net, &ct_event_notifier) == 0) {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/netfilter-full-cone-nat/xt_FULLCONENAT.c: In function ‘fullconenat_tg_destroy’:
/root/netfilter-full-cone-nat/xt_FULLCONENAT.c:675:7: error: too many arguments to function ‘nf_conntrack_unregister_notifier’
  675 |       nf_conntrack_unregister_notifier(par->net, &ct_event_notifier);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/netfilter/nf_conntrack_core.h:18,
                 from /root/netfilter-full-cone-nat/xt_FULLCONENAT.c:28:
./include/net/netfilter/nf_conntrack_ecache.h:70:6: note: declared here
   70 | void nf_conntrack_unregister_notifier(struct net *net);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:251: /root/netfilter-full-cone-nat/xt_FULLCONENAT.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.5.11-7-pve/Makefile:2039: /root/netfilter-full-cone-nat] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.5.11-7-pve'
make: *** [Makefile:5: all] Error 2
root@pve:~/netfilter-full-cone-nat#

hahach, 2023-12-27:
玩啥lxc啊，都pve了直接kvm不好么。
lxc这个让我想起十来年前玩vps时候openvz的小鸡很多东西装不了，因为宿主机不支持。
pve下面kvm的教程网上都多烂了。lxc玩的人太少了

rheatin, 2023-12-27:
> 玩啥lxc啊，都pve了直接kvm不好么。 lxc这个让我想起十来年前玩vps时候openvz的小鸡很多东西装不了，因为宿主机不支持。 pve下面kvm的教程网上都多烂了。lxc玩的人太少了

诶，不搞了，主要之前pve7的时候lxc装的op，然后手贱升级了pve8，结果发现好多都不一样了，fullcone在宿主机也装不上了，kvm的op一切问题都迎刃而解了0 0

closed 2023-12-27 by rheatin

Kissycat, 2024-10-27:
分享个[自用的](https://luna.lovegood.cyou/s/alSp)...

AngusLkc, 2025-03-22:
> 谢谢你，正在尝试，但是PVE8.1.3 安装报错，正在尝试解决问题：root@pve:~/netfilter-full-cone-nat# make make -C /lib/modules/6.5.11-7-pve/build M=/root/netfilter-full-cone-nat modules make[1]: Entering directory [...]

编译报错问题请选中内核模块：kmod-ebtables