coolsnowwolf/lede #11740: rtw88_core not loaded，built with rtw88-usb module

beelabcloud, 2023-12-30:
### 详细叙述

rtw88_usb comflicts with kernel rtw88?

[ 16.138567] rtw88_core: exports duplicate symbol check_hw_ready (owned by rtw_core)
[ 16.143431] kmodloader: 10 modules could not be probed
[ 16.163442] kmodloader: dependency not loaded rtw88_core

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

NanoPi R5C

### 详细日志

[ 16.138567] rtw88_core: exports duplicate symbol check_hw_ready (owned by rtw_core)
[ 16.143431] kmodloader: 10 modules could not be probed
[ 16.163442] kmodloader: dependency not loaded rtw88_core

WYC-2020, 2024-01-16:
[image]
你自己安装了什么模块 导致冲突了呗
[ 11.044342] rtw88_core: exports duplicate symbol check_hw_ready (owned by rtw_core) 这句话说的很明白 rtw88_core和rtw_core 都导出了一个相同的方法，你怕是自己手动安装过什么ipk 把，反正我这边编译 出来没你那个什么 rtw88_core

coolsnowwolf, 2026-09-05:
已确认该问题是两套 rtw88 驱动栈同时安装导致的符号冲突：主线 mac80211 的 `rtw88_core` 导出 `check_hw_ready`，而 lwfinger/rtw88 的 `rtw_core` 也导出同名符号。Rock Pi E v3 的默认包已移除 `kmod-rtw88-8821cu`，避免镜像默认引入冲突栈；修复提交：f369628f73a56f7aecd7d7c4facff87a5567e6c8。若仍需使用 USB RTL8821CU，请只选择其中一套驱动（主线 `kmod-rtw88-8821cu` 或 vendor `kmod-rtl8821cu`），不要同时安装。

closed 2026-09-05 by coolsnowwolf