coolsnowwolf/lede #11742: 编译php7和openssl报错

CHNAICORP, 2023-12-31:
### 详细叙述

使用WSL Ubuntu22.04LTS编译
编译命令：PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin make -j1 V=s
好像是报错php-7.4.30和OpenSSL 3.0版本不兼容。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

树莓派3B+

### 详细日志

/home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:5898:49: warning: 'EVP_PKEY_get0_RSA' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
 5898 |                                                 EVP_PKEY_get0_RSA(pkey),
      |                                                 ^~~~~~~~~~~~~~~~~
In file included from /home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:44:
/home/aicat/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/openssl/evp.h:1348:22: note: declared here
 1348 | const struct rsa_st *EVP_PKEY_get0_RSA(const EVP_PKEY *pkey);
      |                      ^~~~~~~~~~~~~~~~~
/home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:5898:49: warning: passing argument 4 of 'RSA_public_encrypt' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 5898 |                                                 EVP_PKEY_get0_RSA(pkey),
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:46:
/home/aicat/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/openssl/rsa.h:283:29: note: expected 'RSA *' {aka 'struct rsa_st *'} but argument is of type 'const struct rsa_st *'
  283 |                        RSA *rsa, int padding);
  283 |                        RSA *rsa, int padding);
      |                        ~~~~~^~~
/home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c: In function 'zif_openssl_public_decrypt':
/home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:5957:25: warning: 'RSA_public_decrypt' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
 5957 |                         cryptedlen = RSA_public_decrypt((int)data_len,
      |                         ^~~~~~~~~~
In file included from /home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:46:
/home/aicat/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/openssl/rsa.h:288:5: note: declared here
  288 | int RSA_public_decrypt(int flen, const unsigned char *from, unsigned char *to,
      |     ^~~~~~~~~~~~~~~~~~
/home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:5960:41: warning: 'EVP_PKEY_get0_RSA' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
 5960 |                                         EVP_PKEY_get0_RSA(pkey),
      |                                         ^~~~~~~~~~~~~~~~~
In file included from /home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:44:
/home/aicat/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/openssl/evp.h:1348:22: note: declared here
 1348 | const struct rsa_st *EVP_PKEY_get0_RSA(const EVP_PKEY *pkey);
      |                      ^~~~~~~~~~~~~~~~~
/home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:5960:41: warning: passing argument 4 of 'RSA_public_decrypt' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 5960 |                                         EVP_PKEY_get0_RSA(pkey),
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/ext/openssl/openssl.c:46:
/home/aicat/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/openssl/rsa.h:289:29: note: expected 'RSA *' {aka 'struct rsa_st *'} but argument is of type 'const struct rsa_st *'
  289 |                        RSA *rsa, int padding);
      |                        ~~~~~^~~
make[4]: *** [Makefile:567: ext/openssl/openssl.lo] Error 1
make[4]: Leaving directory '/home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30'
make[3]: *** [Makefile:621: /home/aicat/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.30/.built] Error 2
make[3]: Leaving directory '/home/aicat/lede/feeds/packages/lang/php7'
time: package/feeds/packages/php7/compile#10.99#3.27#12.78
    ERROR: package/feeds/packages/php7 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/php7/compile] Error 1
make[2]: Leaving directory '/home/aicat/lede'
make[1]: *** [package/Makefile:110: /home/aicat/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2make[1]: Leaving directory '/home/aicat/lede'
make: *** [/home/aicat/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2024-01-01:
也有可能是ubuntu22.04LTS 
[image]

WYC-2020, 2024-01-02:
都php8了 明显就是 php7和最新的openssl 不能适配，我记得不错的话 openssl 有一个版本分割的很多都变化很大了，要么回退openssl 要么升级到php8自己选择

closed 2024-01-26 by coolsnowwolf