coolsnowwolf/lede #11744: 小米AX3600，编译最后阶段失败

chaoyinsu1023, 2024-01-03:
### 详细叙述

/home/daniel/lede-ax3600/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/d
ts/qcom/ipq8074.dtsi:758.38-763.5: Warning (interrupt_provider): /soc/interrupt-controller@b000000: Missing #address-cel
ls in interrupt provider
rm -f /home/daniel/lede-ax3600/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8071-ax3600.dtb.t
mp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/home/daniel/lede-ax3600/build_dir/target-aarch64_cort
ex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts -I/home/daniel/lede-ax3600/build_dir/target-aarch64
_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/include -I/home/daniel/lede-ax3600/build_dir/t
arget-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/include/ -undef -D__DTS__  -o /home/daniel/lede-ax360
0/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image-ipq8072-ax9000.dtb.tmp /home/daniel/lede-ax3600/b
uild_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205/arch/arm64/boot/dts/qcom/ipq8072-ax9000.dts
**cc1: fatal error: /home/daniel/lede-ax3600/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.205
/arch/arm64/boot/dts/qcom/ipq8072-ax9000.dts: No such file or directory**
compilation terminated.
make[5]: *** [Makefile:18: /home/daniel/lede-ax3600/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/image
-ipq8072-ax9000.dtb] Error 1
make[5]: Leaving directory '/home/daniel/lede-ax3600/target/linux/ipq807x/image'
make[4]: *** [Makefile:26: install] Error 2
make[4]: Leaving directory '/home/daniel/lede-ax3600/target/linux/ipq807x'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/daniel/lede-ax3600/target/linux'
time: target/linux/install#6.91#3.20#25.91
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/daniel/lede-ax3600'
make[1]: *** [target/Makefile:24: /home/daniel/lede-ax3600/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_inst
all] Error 2
make[1]: Leaving directory '/home/daniel/lede-ax3600'
make: *** [/home/daniel/lede-ax3600/include/toplevel.mk:231：world] 错误 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

xiaomi ax3600

### 详细日志

make V=s -j1

WYC-2020, 2024-01-03:
字面意思，都被干掉了 还编译个鸡毛 
[image]

WYC-2020, 2024-01-03:
并且你这个编译ax3600 不知道如何搞的会牵扯到ax9000 要么你那里没选对 要么就是删除不干净有尾巴，等会我试试看

chaoyinsu1023, 2024-01-04:
> 并且你这个编译ax3600 不知道如何搞的会牵扯到ax9000 要么你那里没选对 要么就是删除不干净有尾巴，等会我试试看

我是单独git clone到新目录，重头进行配置编译的

CarterCL, 2024-01-04:
301w也是编译失败，一样的错，估计807x的都不行

chaoyinsu1023, 2024-01-04:
> 301w也是编译失败，一样的错，估计807x的都不行

顺着“金水”的视频，在https://fw.koolcenter.com/上找到有现成的，说明自己也应该能编译

WYC-2020, 2024-01-04:
[image]
删除这个ax9000 dts都没得自然在最后打包 查找dts的时候 会出错

WYC-2020, 2024-01-04:
> > 并且你这个编译ax3600 不知道如何搞的会牵扯到ax9000 要么你那里没选对 要么就是删除不干净有尾巴，等会我试试看
> 
> 我是单独git clone到新目录，重头进行配置编译的

我说的不干净 是他们以前删除的不干净 不是说你，这个小米原来是被全部删除了的，后来某人又提交回来了但是9000是没有提交回来的，ax3600和ax6 好像是提交回来了的，
[image]
这里是全删除了小米支持的 ax6 ax3600的
[image]
这里是删除了ax9000的
[image]
这里是吧ax6 ax3600给加回来了，但是ax9000是没有回来的，如果要加回来自己参考
[image]
这个提交吧ax9000加回来就行了，其他他们那些能编译9000的肯定是基于这个仓库自己搞了一份的，用l官方的9000肯定是不行的别听他们乱说

chaoyinsu1023, 2024-01-04:
我的AX3600只能当AP用，好性能浪费了，唉~~

WYC-2020, 2024-01-04:
> 我的AX3600只能当AP用，好性能浪费了，唉~~

正常的很 我主路由从来就不用硬路由 都是软路由当主路由  硬路由拿来做ap，稳如死狗，不会出现各种奇葩的问题

schen39, 2024-01-05:
[image]
[image]
我编译成功了QHora-301W

WYC-2020, 2024-01-08:
> > > > 并且你这个编译ax3600 不知道如何搞的会牵扯到ax9000 要么你那里没选对 要么就是删除不干净有尾巴，等会我试试看
> > > 
> > > 
> > > 我是单独git clone到新目录，重头进行配置编译的
> > 
> > 
> > 我说的不干净 是他们以前删除的不干净 不是说你，这个小米原来是被全部删除了的，后来某人又提交回来了但是9000是没有提交回来的，ax3600和ax6 好像是提交回来了的，
> > ![image](https://private-user-images.githubusercontent.com/34025388/294109127-eb927856-c927-47f6-b788-5de0fc0fe4f7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NzkwMTYsIm5iZiI6MTcwNDY3ODcxNiwicGF0aCI6Ii8zNDAyNTM4OC8yOTQxMDkxMjctZWI5Mjc4NTYtYzkyNy00N2Y2LWI3ODgtNWRlMGZjMGZlNGY3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDAxNTE1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI1MWFkMWYxZTdmNjEyMzc1NjQ3YTlkZDMzNzZjY2YyMTFjODg4YTNjZGQxOGQ5YWFkN2Q2N2RlYmRlM2UyMWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PzbqYaQhDxQVu32Id-hcjyfimUZVQvnRMAG7gTCb4Hs)
> > 这里是全删除了小米支持的 ax6 ax3600的
> > ![image](https://private-user-images.githubusercontent.com/34025388/294109321-156aa4eb-aaa2-4e4f-a6a2-6aefbd705142.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NzkwMTYsIm5iZiI6MTcwNDY3ODcxNiwicGF0aCI6Ii8zNDAyNTM4OC8yOTQxMDkzMjEtMTU2YWE0ZWItYWFhMi00ZTRmLWE2YTItNmFlZmJkNzA1MTQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDAxNTE1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdmYmI4M2E2ZTcwYTI2MWY0ZDAyNDZjZDhkYTgxNDhlMDYzNjViMmI2NzRjNzNmNWZiMDc3ZjE3MDA5MDRlMTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ovskTiMHHrVf7agvH2LWE-TR4dIWt6SnsWSHbIDlZ6A)
> > 这里是删除了ax9000的
> > ![image](https://private-user-images.githubusercontent.com/34025388/294109765-84cc6bba-5029-4411-a46e-1155c76c823e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NzkwMTYsIm5iZiI6MTcwNDY3ODcxNiwicGF0aCI6Ii8zNDAyNTM4OC8yOTQxMDk3NjUtODRjYzZiYmEtNTAyOS00NDExLWE0NmUtMTE1NWM3NmM4MjNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDAxNTE1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVkMTM0NWI5ZmRkZDg3NTE0MWMxOThiMWY1MzM0NDQzZjA0NzE3YmI4OGRhMjMxMWM0YWZmZjI4ZDg3YjhhMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Q8BzvWiB5pOsqfe8Bv-63nvOwBxWIK08gRB4gl5Wdys)
> > 这里是吧ax6 ax3600给加回来了，但是ax9000是没有回来的，如果要加回来自己参考
> > ![image](https://private-user-images.githubusercontent.com/34025388/294109919-d3f040c9-c1e6-463c-9b4e-2294e8fc6848.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NzkwMTYsIm5iZiI6MTcwNDY3ODcxNiwicGF0aCI6Ii8zNDAyNTM4OC8yOTQxMDk5MTktZDNmMDQwYzktYzFlNi00NjNjLTliNGUtMjI5NGU4ZmM2ODQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDAxNTE1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUzYWIxMjFiYTNiMzU2Yjc5MDNlNmJlMWZmNzIxZTJlODc0MDE2YTk5NmI5N2M0YmRmYWJiZWVkNzVkMDk3MmUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-DhExD7zw606WAZ2Sd6RmtlBqLQx3h_MAytRTy-ixuE)
> > 这个提交吧ax9000加回来就行了，其他他们那些能编译9000的肯定是基于这个仓库自己搞了一份的，用l官方的9000肯定是不行的别听他们乱说
> 
> 老哥你这什么软件?看起来很好用！

乌龟啊 ，不知道？tortoisegit

CarterCL, 2024-01-08:
> ![微信截图_20240105152112](https://private-user-images.githubusercontent.com/34901523/294421843-80501c43-7b28-4846-a488-318e1b0dea2d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NzkyOTEsIm5iZiI6MTcwNDY3ODk5MSwicGF0aCI6Ii8zNDkwMTUyMy8yOTQ0MjE4NDMtODA1MDFjNDMtN2IyOC00ODQ2LWE0ODgtMzE4ZTFiMGRlYTJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDAxNTYzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2NTY3ZWVjNzVhNGU0ZjkwMTdiMWVjMDJmNDI0OWM4MWM2MmE3NmVkMmFlMjU1YWU3MjEzNGVlN2NjMWJjMmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.O1bNsOHzB1SnmW1QJSuvXaegonL9rMH4BOILvXtoNM4) ![微信截图_20240105152016](https://private-user-images.githubusercontent.com/34901523/294421865-64683b56-97c0-4734-b2e1-e02b1ddcea66.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NzkyOTEsIm5iZiI6MTcwNDY3ODk5MSwicGF0aCI6Ii8zNDkwMTUyMy8yOTQ0MjE4NjUtNjQ2ODNiNTYtOTdjMC00NzM0LWIyZTEtZTAyYjFkZGNlYTY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDAxNTYzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEzYjE5N2M1OWI3M2NmODQwZmVhMGMwMGRjN2QyMzVkZGU3NWMyNTUyNWM2YjE3NTgyMzk5ZGQxZGI4MWQwOWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hdAvZ2pwFkY-CLaj1lmH0-clYbiJaNOYna4BdleNxJI) 我编译成功了QHora-301W

编译完之后，能通过Uboot刷入吗？

schen39, 2024-01-08:
@CarterCL 可以，不过有注意的点，在root/tmp/sysinfo/board_name中名称问题，需要改动，原始board_name是ap-hk01-c1，改错升级变砖，这边的建议在自己搞不清楚改成什么之前，建议还是TTL刷入保险

schen39, 2024-01-08:
@CarterCL 建议咨询下大雕，以免变砖，虽然有不死uboot，一旦改错强行刷入网络全掉不断重启

coolsnowwolf, 2026-09-05:
该日志来自旧版 `target/linux/ipq807x` 路径，失败原因是构建目录中缺少 `ipq8072-ax9000.dts`。当前 LEDE 已在提交 8fcf7f0915e 完成 ipq807x→qualcommax 目标重命名并整理 DTS，现文件位于 `target/linux/qualcommax/files/arch/arm64/boot/dts/qcom/ipq8072-ax9000.dts`，AX3600/AX9000 设备定义也已正常存在于 `target/linux/qualcommax/image/ipq807x.mk`。这是旧源码构建产物问题，按当前树已解决，关闭 issue。

closed 2026-09-05 by coolsnowwolf