coolsnowwolf/lede #11748: 编译x86/64时选择hostapd出错 ，请求解决

lxl750629, 2024-01-06:
### 详细叙述

编译x86/64时需无线，选择hostapd编译出错  ,请问需怎么解决？谢谢！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86/64

### 详细日志

/home/lxl/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lxl750629/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_request_scan':
acs.c:(.text+0x3a3): undefined reference to `hostapd_driver_scan'
/home/lxl/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lxl750629/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_scan_complete':
acs.c:(.text+0xe55): undefined reference to `hostapd_disable_iface'
/home/lxl/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lxl750629/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_init':
acs.c:(.text+0xed1): undefined reference to `hostapd_drv_do_acs'
/home/lxl/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: acs.c:(.text+0xf1c): undefined reference to `hostapd_set_state'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:2037: wpa_supplicant] Error 1
make[4]: Leaving directory '/home/lxl/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/wpa_supplicant'
make[3]: *** [Makefile:790: /home/lxl/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/.built] Error 2
make[3]: Leaving directory '/home/lxl/lede/package/network/services/hostapd'
time: package/network/services/hostapd/supplicant-full-internal/compile#0.17#0.14#0.28
    ERROR: package/network/services/hostapd failed to build (build variant: supplicant-full-internal).
make[2]: *** [package/Makefile:119: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/lxl/lede'
make[1]: *** [package/Makefile:110: /home/lxl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lxl/lede'
make: *** [/home/lxl/lede/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2024-01-08:
[image]
这个我记得以前修复过 ，好像是这样的  你自己试试 看，或者你直接手动选择 kmod-cfg80211 这个模块 在编译试试看

lxl750629, 2024-01-08:
WYC-2020你好，谢谢你的回复，不过按你的修改方法还是同样出错。

lxl750629, 2024-01-08:
谢谢[WYC-2020的回复，按你提供的修改文件的方法以及直接手动选择kmod-cfg80211 这个模块还是出错。

WYC-2020, 2024-01-09:
> 谢谢[WYC-2020的回复，按你提供的修改文件的方法以及直接手动选择kmod-cfg80211 这个模块还是出错。

config给我 我编译看下

lxl750629, 2024-01-11:
WYC-2020你好，config 已发你邮箱，请问帮我找到原因了吗？谢谢！

WYC-2020, 2024-01-12:
> WYC-2020你好，config 已发你邮箱，请问帮我找到原因了吗？谢谢！

你发那个邮箱了 都没看到，你直接把.config改成config.txt上传到这里不就行了，最近都很忙没时间去看邮箱

WYC-2020, 2024-01-12:
[hostapd.zip](https://github.com/coolsnowwolf/lede/files/13912709/hostapd.zip)
用我这个 替换掉原来的 建议是吧原来的删除 然后用这个试试 @lxl750629

lxl750629, 2024-01-13:
WYC-2020，换了你提供的文件，成功编译，非常感谢！

coolsnowwolf, 2026-09-05:
已定位并修复：`CONFIG_ACS=y` 被无条件传给 supplicant-only 构建，导致 `eapol_test` 链接 `acs.o` 时引用仅由 hostapd 提供的 `hostapd_driver_scan`、`hostapd_disable_iface` 等符号。现仅在 hostapd/wpad 构建启用 ACS，supplicant-only 构建关闭该选项。已用当前 x86_64 工具链执行 `make package/network/services/hostapd/compile -j30 V=s` 成功。修复提交：b84c6b2e0f14b875c0f0889ef2380c2aa2a920b1。

closed 2026-09-05 by coolsnowwolf