coolsnowwolf/lede #11749: 6.1.69版本内核 Cloudflared 编译错误，请求帮助

JIFVGWHvAseSovQ, 2024-01-06:
### 详细叙述

https://github.com/coolsnowwolf/lede/issues/11368

修改/lede/feeds/packages/net/cloudflared/Makefile 版本号和HASH一样报错, 试过2023.7.1, 2023.7.2, 2023.7.3都不行

PKG_NAME:=cloudflared
PKG_VERSION:=2023.5.1
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://codeload.github.com/cloudflare/cloudflared/tar.gz/$(PKG_VERSION)?
PKG_HASH:=ee2c2a4b0c290c39475f79ab74972dfbce817df8e5090813cad0e58f33836194

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Windows11 wsl2 Debian

### 详细日志

mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo "uhttpd" >> /root/lede/staging_dir/target-x86_64_musl/pkginfo/uhttpd.default.install
mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/luci-app-arpbind/.pkgdir/luci-app-arpbind/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/uhttpd-2022-10-31-23977554/.pkgdir/uhttpd-mod-ubus/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-app-arpbind_installed
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.uhttpd-mod-ubus_installed
mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo "uhttpd-mod-ubus" >> /root/lede/staging_dir/target-x86_64_musl/pkginfo/uhttpd.default.install
make[3]: Leaving directory '/root/lede/package/network/services/uhttpd'
time: package/network/services/uhttpd/compile#0.21#0.02#0.26
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/luci-lib-nixio/.pkgdir/luci-lib-nixio/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
echo "luci-app-arpbind" >> /root/lede/staging_dir/target-x86_64_musl/pkginfo/luci-app-arpbind.default.install
make[3]: *** [Makefile:68: /root/lede/build_dir/target-x86_64_musl/cloudflared-2023.5.1/.built] Error 1
make[3]: Leaving directory '/root/lede/feeds/packages/net/cloudflared'
time: package/feeds/packages/cloudflared/compile#8.11#3.79#20.49
mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
    ERROR: package/feeds/packages/cloudflared failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/cloudflared/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/luci-app-arpbind/.pkgdir/luci-i18n-arpbind-zh-cn/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-lib-nixio_installed
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-i18n-arpbind-zh-cn_installed
echo "luci-i18n-arpbind-zh-cn" >> /root/lede/staging_dir/target-x86_64_musl/pkginfo/luci-app-arpbind.default.install
make[3]: Leaving directory '/root/lede/feeds/luci/applications/luci-app-arpbind'
echo "luci-lib-nixio" >> /root/lede/staging_dir/target-x86_64_musl/pkginfo/luci-lib-nixio.default.install
make[3]: Entering directory '/root/lede/feeds/packages/net/bind'
time: package/feeds/luci/luci-app-arpbind/compile#0.17#0.02#0.21
make[3]: Leaving directory '/root/lede/feeds/luci/libs/luci-lib-nixio'
time: package/feeds/luci/luci-lib-nixio/compile#0.41#0.07#0.47
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/root/lede/feeds/packages/net/bind'
time: package/feeds/packages/bind/compile#0.16#0.01#0.17
make[2]: Leaving directory '/root/lede'
make[1]: *** [package/Makefile:110: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'

JIFVGWHvAseSovQ, 2024-01-06:
修改过内核版本
sed -i 's/KERNEL_PATCHVER:=5.15/KERNEL_PATCHVER:=6.1/g' ./target/linux/x86/Makefile

JIFVGWHvAseSovQ, 2024-01-08:
将cloudflared版本改为2023.5.1版本以上，golang Makefile文件中GO_VERSION_MAJOR_MINOR:=1.21改为GO_VERSION_MAJOR_MINOR:=1.20

JIFVGWHvAseSovQ, 2024-01-08:
/lede/build_dir/hostpkg/go-1.20.5此目录新建go.env文件, 并填写以下内容：
# This file contains the initial defaults for go command configuration.
# Values set by 'go env -w' and written to the user's go/env file override these.
# The environment overrides everything else.

# Use the Go module mirror and checksum database by default.
# See https://proxy.golang.org for details.
GOPROXY=https://proxy.golang.org,direct
GOSUMDB=sum.golang.org

# Automatically download newer toolchains as directed by go.mod files.
# See https://go.dev/doc/toolchain for details.
GOTOOLCHAIN=auto

JIFVGWHvAseSovQ, 2024-01-08:
../../vendor/github.com/quic-go/quic-go/internal/qtls/go121.go:5:13: cannot use "The version of quic-go you're using can't be built on Go 1.21 yet. For more details, please see https://github.com/quic-go/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.21 yet. F...) as int value in variable declaration

WYC-2020, 2024-01-08:
这样吧 
[image]
其实已经修复了的 只是没有打版本而已，直接改用git 用最新的源码 就没有问题了，要不然你就自己提取commit搞成path,或者等官方打版本
[image]

WYC-2020, 2024-01-10:
10 我早就试过了，不行的 这个版本 go 1.21 他是没有搞好的，刚又去那边看了下 打了版本 了 用[2024.1.0](https://github.com/cloudflare/cloudflared/commit/159fcb44ce71d02c86e593c7211b2c6aca5d0b08) 这个版本 应该就可以了

ShunbaoLi, 2024-06-17:
/lede/feeds/packages/net/cloudflared/Makefile

用2024.6.0版本可以通过编译

`# SPDX-License-Identifier: GPL-3.0-only
#
# Copyright (C) 2021 ImmortalWrt.org

include $(TOPDIR)/rules.mk

PKG_NAME:=cloudflared
PKG_VERSION:=2024.6.0
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://codeload.github.com/cloudflare/cloudflared/tar.gz/$(PKG_VERSION)?
PKG_HASH:=e75eec7eaf61320f7b5f9f6abc0891285bd3eeebad46b4a5cb53765281a8d88e

PKG_LICENSE:=Apache-2.0
PKG_LICENSE_FILES:=LICENSE
PKG_MAINTAINER:=Tianling Shen <cnsztl@immortalwrt.org>

PKG_BUILD_DEPENDS:=golang/host
PKG_BUILD_PARALLEL:=1
PKG_USE_MIPS16:=0

GO_PKG:=github.com/cloudflare/cloudflared
GO_PKG_LDFLAGS_X:=main.Version=$(PKG_VERSION)

include $(INCLUDE_DIR)/package.mk
include ../../lang/golang/golang-package.mk

define Package/cloudflared
  SECTION:=net
  CATEGORY:=Network
  SUBMENU:=Web Servers/Proxies
  TITLE:=Cloudflare Tunnel client
  URL:=https://developers.cloudflare.com/cloudflare-one/connections/connect-apps/install-and-setup/tunnel-guide
  DEPENDS:=$(GO_ARCH_DEPENDS) +ca-bundle
endef

define Package/cloudflared/description
  Contains the command-line client for Cloudflare Tunnel, a tunneling
  daemon that proxies traffic from the Cloudflare network to your origins.

  This daemon sits between Cloudflare network and your origin (e.g. a
  webserver). Cloudflare attracts client requests and sends them to you
  via this daemon, without requiring you to poke holes on your firewall
  --- your origin can remain as closed as possible.
endef

define Package/cloudflared/conffiles
/etc/config/cloudflared
/etc/cloudflared/
endef

define Package/cloudflared/install
	$(call GoPackage/Package/Install/Bin,$(1))

	$(INSTALL_DIR) $(1)/etc/cloudflared/
	$(INSTALL_CONF) $(CURDIR)/files/sample_config.yml $(1)/etc/cloudflared/config.yml
	$(INSTALL_DIR) $(1)/etc/config/
	$(INSTALL_CONF) $(CURDIR)/files/cloudflared.config $(1)/etc/config/cloudflared
	$(INSTALL_DIR) $(1)/etc/init.d/
	$(INSTALL_BIN) $(CURDIR)/files/cloudflared.init $(1)/etc/init.d/cloudflared
endef

$(eval $(call GoBinPackage,cloudflared))
$(eval $(call BuildPackage,cloudflared))
`

closed 2025-01-09 by aiamadeus