coolsnowwolf/lede #11751: 请教关于 Turbo Acc 可能导致微信消息延迟的问题

55233Lei, 2024-01-07:
### 详细叙述

想问下大家有遇到开启 Flow offload 后，安卓端微信接收消息出现延迟的情况吗？

之前一直没开 Flow offload，最近开了以后发现微信消息经常不能及时收到，会延迟数分钟，两台安卓手机都复现了，家人还错过了一次微信语音（手机放桌上，别人打来语音没响），感觉是微信的长连接挂了，关了 Flow offload 后以上问题就没有再出现了。

看到 CSDN 有类似的情况分析，貌似是 5.4 内核的问题，但我使用最新内核居然也遇到了，按理来说不是修复了么。
https://blog.csdn.net/weixin_43744534/article/details/123418289

另外想请教下，X86 有必要开这个么？考虑到开 Shortcut-FE 可能会导致丢包才编译的 Flow offload，没想到会有这种情况。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

暂无

WYC-2020, 2024-01-08:
x86 都是性能过剩的 没必要开那个玩意 那玩意是给硬路由用的，你这个链接可以 我看看 可以给5.4打一个补丁不

yanxin152133, 2024-01-08:
学习学习：）

WYC-2020, 2024-01-08:
看了下 别人5.10都修复了的
[image]
就5.4没有适配，我大体适配了一下，你可以测测功能，但是不一定你那个直接放就能用，应该可能需要更新后续的patch才行，我的和l的仓库有区别，你试试看直接放进去有没其他patch出错吧
[610-v5.13-55-netfilter-conntrack-Introduce-tcp-offload-timeout-co.patch](https://github.com/coolsnowwolf/lede/files/13857773/610-v5.13-55-netfilter-conntrack-Introduce-tcp-offload-timeout-co.patch)
[610-v5.13-57-netfilter-flowtable-Set-offload-timeouts-according-t.patch](https://github.com/coolsnowwolf/lede/files/13857774/610-v5.13-57-netfilter-flowtable-Set-offload-timeouts-according-t.patch)
[610-v5.13-56-netfilter-conntrack-Introduce-udp-offload-timeout-co.patch](https://github.com/coolsnowwolf/lede/files/13857775/610-v5.13-56-netfilter-conntrack-Introduce-udp-offload-timeout-co.patch)

WYC-2020, 2024-01-08:
[610-v5.13-57-netfilter-flowtable-Set-offload-timeouts-according-t.patch](https://github.com/coolsnowwolf/lede/files/13857961/610-v5.13-57-netfilter-flowtable-Set-offload-timeouts-according-t.patch)
[610-v5.13-56-netfilter-conntrack-Introduce-udp-offload-timeout-co.patch](https://github.com/coolsnowwolf/lede/files/13857962/610-v5.13-56-netfilter-conntrack-Introduce-udp-offload-timeout-co.patch)
[610-v5.13-55-netfilter-conntrack-Introduce-tcp-offload-timeout-co.patch](https://github.com/coolsnowwolf/lede/files/13857963/610-v5.13-55-netfilter-conntrack-Introduce-tcp-offload-timeout-co.patch)
刚patch有一点问题，现在这个亲测编译通过，这些patch来自于
[image]
这个大佬在5.10上面的提交，我看5.4了l有一个 370-netfilter-nf_flow_table-fix-offloaded-connection-tim.patch 这个补丁，不确定是否需要删除，敢死队要上的就上，哈哈