coolsnowwolf/lede #11757: svn不能用了，大佬们都是如何应对的

binge8, 2024-01-10:
### 详细叙述

svn不能用了，大佬们都是如何应对的。怎么添加第三方库里单独的文件夹

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

请教贴

hex-ci, 2024-01-10:
Github 不能用 svn 就改成 git 呗，我写了一个 bash 函数用于拉取一个 git 仓库的某个子目录到一个目标目录

```bash
CURRENT_PATH=$(pwd)

clone_or_update_git_repo() {
  # 参数检查
  if [ "$#" -lt 2 ]; then
    echo "Usage: clone_or_update_git_repo <git_url> <target_directory> [branch] [subdirectory]"
    return 1
  fi

  local git_url="$1"
  local source_target_directory="$2"
  local target_directory="$2"
  local branch="$3"
  local subdirectory="$4"

  if [ -n "$subdirectory" ]; then
    target_directory=$CURRENT_PATH/repos/$(echo "$git_url" | awk -F'/' '{print $(NF-1)"-"$NF}')
  fi

  # 检查目标目录是否存在
  if [ -d "$target_directory" ]; then
    pushd "$target_directory" || return 1
    git pull
    popd
  else
    if [ -n "$branch" ]; then
      git clone --depth=1 -b "$branch" "$git_url" "$target_directory"
    else
      git clone --depth=1 "$git_url" "$target_directory"
    fi
  fi

  if [ -n "$subdirectory" ]; then
    cp -a $target_directory/$subdirectory $source_target_directory
  fi
}

# 用法举例
clone_or_update_git_repo "https://github.com/xxx/yyy" "目标目录" "分支名" "git 子目录"
clone_or_update_git_repo "https://github.com/xxx/yyy" "package/luci-xxx" "openwrt-18.06" "applications/xxx"
clone_or_update_git_repo "https://github.com/xxx/yyy" "package/luci-xxx" "" "applications/xxx"
```

注意会缓存 git 所有内容到当前目录下的 repos 目录下

qqsir-dev, 2024-01-10:
能不能具体点，用了老出错，谢谢

xuexijiaoben, 2024-01-10:
I do it like this:

`git clone https://github.com/kenzok8/small-package.git kenzok8
cp -rf kenzok8/luci-app-adguardhome package/luci-app-adguardhome
rm -rf kenzok8`

not smart but useful

qqsir-dev, 2024-01-10:
> I do it like this:
> 
> `git clone https://github.com/kenzok8/small-package.git kenzok8
> 
> cp -rf kenzok8/luci-app-adguardhome package/luci-app-adguardhome
> 
> rm -rf kenzok8`
> 
> not smart but useful

This works perfect!!!

qqsir-dev, 2024-01-11:
> ```shell
> svn_export() {
> 	# 参数1是分支名, 参数2是子目录, 参数3是目标目录, 参数4仓库地址
> 	trap 'rm -rf "$TMP_DIR"' 0 1 2 3
> 	TMP_DIR="$(mktemp -d)" || exit 1
> 	[ -d "$3" ] || mkdir -p "$3"
> 	TGT_DIR="$(cd "$3"; pwd)"
> 	cd "$TMP_DIR" && \
> 	git init >/dev/null 2>&1 && \
> 	git remote add -f origin "$4" >/dev/null 2>&1 && \
> 	git checkout "remotes/origin/$1" -- "$2" && \
> 	cd "$2" && cp -a . "$TGT_DIR/"
> }
> 
> svn_export "master" "target/linux/x86" "route" "https://github.com/coolsnowwolf/lede"
> ```

我就羡慕又讨厌你们这些会写正则表达式的人 :P

binge8, 2024-01-11:
`function merge_package(){
    repo=`echo $1 | rev | cut -d'/' -f 1 | rev`
    pkg=`echo $2 | rev | cut -d'/' -f 1 | rev`
    # find package/ -follow -name $pkg -not -path "package/custom/*" | xargs -rt rm -rf
    git clone --depth=1 --single-branch $1
    mv $2 package/custom/
    rm -rf $repo
}
merge_package https://github.com/WYC-2020/openwrt-packages openwrt-packages/lua-maxminddb
merge_package https://github.com/kenzok8/small-package small-package/luci-app-socat`
抄来的，很好使。拉取的插件都在 package/custom/ 路径

binge8, 2024-01-11:
```bash
function merge_package(){
    repo=`echo $1 | rev | cut -d'/' -f 1 | rev`
    pkg=`echo $2 | rev | cut -d'/' -f 1 | rev`
    # find package/ -follow -name $pkg -not -path "package/custom/*" | xargs -rt rm -rf
    git clone --depth=1 --single-branch $1
    mv $2 package/custom/
    rm -rf $repo
}
merge_package https://github.com/WYC-2020/openwrt-packages openwrt-packages/lua-maxminddb
merge_package https://github.com/kenzok8/small-package small-package/luci-app-socat
```
抄来的，很好用。

ilxp, 2024-01-11:
> 我就羡慕又讨厌你们这些会写正则表达式的人 :P

这个只能产生一个空目录，里面没有文件。

ilxp, 2024-01-11:
> ```shell
> function merge_package(){
>     repo=`echo $1 | rev | cut -d'/' -f 1 | rev`
>     pkg=`echo $2 | rev | cut -d'/' -f 1 | rev`
>     # find package/ -follow -name $pkg -not -path "package/custom/*" | xargs -rt rm -rf
>     git clone --depth=1 --single-branch $1
>     mv $2 package/custom/
>     rm -rf $repo
> }
> merge_package https://github.com/WYC-2020/openwrt-packages openwrt-packages/lua-maxminddb
> merge_package https://github.com/kenzok8/small-package small-package/luci-app-socat
> ```
> 
> 抄来的，很好用。

这个会全部拉去下来，没有想要的目标文件呢。

binge8, 2024-01-11:
> ```shell
> ```shell
> function merge_package(){
>     repo=`echo $1 | rev | cut -d'/' -f 1 | rev`
>     pkg=`echo $2 | rev | cut -d'/' -f 1 | rev`
>     # find package/ -follow -name $pkg -not -path "package/custom/*" | xargs -rt rm -rf
>     git clone --depth=1 --single-branch $1
>     mv $2 package/custom/
>     rm -rf $repo
> }
> merge_package https://github.com/WYC-2020/openwrt-packages openwrt-packages/lua-maxminddb
> merge_package https://github.com/kenzok8/small-package small-package/luci-app-socat
> ```
> 
> 
>     
>       
>     
> 
>       
>     
> 
>     
>   
> 抄来的，很好用。
> 
> 这个会全部拉去下来，没有想要的目标文件呢。
> ```

你可能不会用

binge8, 2024-01-11:
优化了一下代码，[参考这里吧](https://github.com/binge8/op)

ilxp, 2024-01-12:
> 你可能不会用

我是在windosw上新建个文件夹，把上面函数复制.diy.sh. 然后git bash执行的。

sjkhsl, 2024-01-12:
> ```shell
> svn_export() {
> 	# 参数1是分支名, 参数2是子目录, 参数3是目标目录, 参数4仓库地址
> 	trap 'rm -rf "$TMP_DIR"' 0 1 2 3
> 	TMP_DIR="$(mktemp -d)" || exit 1
> 	[ -d "$3" ] || mkdir -p "$3"
> 	TGT_DIR="$(cd "$3"; pwd)"
> 	cd [...]

好用，感谢~

free-diy, 2024-01-12:
> ```shell
> function merge_package(){
>     repo=`echo $1 | rev | cut -d'/' -f 1 | rev`
>     pkg=`echo $2 | rev | cut -d'/' -f 1 | rev`
>     # find package/ -follow -name $pkg -not -path "package/openwrt-packages/*" | xargs -rt rm -rf
>     git clone --depth=1 --single-branch $1
>     [ -d package/openwrt-packages ] || mkdir -p package/openwrt-packages
>     mv $2 package/openwrt-packages/
>     rm -rf $repo
> }
> merge_package https://github.com/WYC-2020/openwrt-packages openwrt-packages/lua-maxminddb
> merge_package https://github.com/kenzok8/small-package small-package/luci-app-socat
> ```
> 
> 抄来的，很好用。

这个有分支就不好用了啊

sjkhsl, 2024-01-12:
> ```shell
> svn_export() {
> 	# 参数1是分支名, 参数2是子目录, 参数3是目标目录, 参数4仓库地址
> 	trap 'rm -rf "$TMP_DIR"' 0 1 2 3
> 	TMP_DIR="$(mktemp -d)" || exit 1
> 	[ -d "$3" ] || mkdir -p "$3"
> 	TGT_DIR="$(cd "$3"; pwd)"
> 	cd [...]

现在使用突然出现登录，输入github的用户名和密码登录不上啊！
```
Username for 'https://github.com': 
Password for 'https://xxx@github.com':
```

OldCoding, 2024-01-12:
> ```shell
> svn_export() {
> 	# 参数1是分支名, 参数2是子目录, 参数3是目标目录, 参数4仓库地址
> 	trap 'rm -rf "$TMP_DIR"' 0 1 2 3
> 	TMP_DIR="$(mktemp -d)" || exit 1
> 	[ -d "$3" ] || mkdir -p "$3"
> 	TGT_DIR="$(cd "$3"; pwd)"
> 	cd [...]

遇到仓库比较大或提交记录比较多的就会很慢，例如https://github.com/vernesong/OpenClash

binge8, 2024-01-12:
> 遇到仓库比较大或提交记录比较多的就会很慢，例如https://github.com/vernesong/OpenClash

就一个路由器插件库能有多大，顺便了解下--depth=1

fichenx, 2024-01-13:
抄的这位大佬的，在本地试过，不会下载整个库，只会下载指定的目录，并移动到根目录。
https://github.com/kenzok8/small-package

function git_sparse_clone() {
  branch="$1" rurl="$2" localdir="$3" && shift 3
  git clone -b $branch --depth 1 --filter=blob:none --sparse $rurl $localdir
  cd $localdir
  git sparse-checkout init --cone
  git sparse-checkout set $@
  mv -n $@ ../
  cd ..
  rm -rf $localdir
  }

使用方式：
git_sparse_clone openwrt-23.05 "https://github.com/openwrt/openwrt" "openwrt" package/base-files package/network/config/firewall4

fichenx, 2024-01-13:
抄的这位大佬的，在本地试过，不会下载整个库，只会下载指定的目录，并移动到根目录。
https://github.com/kenzok8/small-package

```
function git_sparse_clone() {
branch="$1" rurl="$2" localdir="$3" && shift 3
git clone -b $branch --depth 1 --filter=blob:none --sparse $rurl $localdir
cd $localdir
git sparse-checkout init --cone
git sparse-checkout set $@
mv -n $@ ../
cd ..
rm -rf $localdir
}
```

使用方式：
`git_sparse_clone openwrt-23.05 "https://github.com/openwrt/openwrt" "openwrt" package/base-files package/network/config/firewall4`

binge8, 2024-01-13:
> 抄的这位大佬的，在本地试过，不会下载整个库，只会下载指定的目录，并移动到根目录。 https://github.com/kenzok8/small-package
>
> ```
> function git_sparse_clone() {
> branch="$1" rurl="$2" localdir="$3" && shift 3
> git clone -b $branch --depth 1 [...]

zhe 这明显也是拉的整个库，只不过留下需要的，其余的删了

fichenx, 2024-01-13:
> zhe 这明显也是拉的整个库，只不过留下需要的，其余的删了

我一步一步的试过了，一开始只会下载一个.git目录和根目录的文件，不会下载其他多余的文件。后边的命令会下载指定的目录和文件。

binge8, 2024-01-13:
> 我一步一步的试过了，一开始只会下载一个.git目录和根目录的文件，不会下载其他多余的文件。后边的命令会下载指定的目录和文件。

真是只下载指定文件夹。我在这个基础上，结合几位大佬的优化了下，不错
```bash
function merge_package(){
    # 参数1是分支名,参数2是库地址。所有文件下载到openwrt/package/openwrt-packages路径。
    # 同一个仓库下载多个文件夹直接在后面跟文件名或路径，空格分开。
    branch="$1" curl="$2" && shift 2
    rootdir="$PWD"
    localdir=package/openwrt-packages
    [ -d "$localdir" ] || mkdir -p "$localdir"
    tmpdir="$(mktemp -d)" || exit 1
    git clone -b "$branch" --depth 1 --filter=blob:none --sparse "$curl" "$tmpdir"
    cd "$tmpdir"
    git sparse-checkout init --cone
    git sparse-checkout set "$@"
    mv -f "$@" "$rootdir"/"$localdir"
    cd "$rootdir" && rm -rf "$tmpdir"
}
merge_package master https://github.com/WYC-2020/openwrt-packages luci-app-eqos luci-app-openclash luci-app-ddnsto ddnsto 
merge_package master https://github.com/lisaac/luci-app-dockerman applications/luci-app-dockerman
```

ywt114, 2024-01-14:
https://github.com/stupidloud/nanopi-openwrt/blob/master/scripts/merge_packages.sh
参照stupidloud大佬很早就用这个啦

binge8, 2024-01-15:
> https://github.com/stupidloud/nanopi-openwrt/blob/master/scripts/merge_packages.sh 参照stupidloud大佬很早就用这个啦

这也是先下载整个库吧

ywt114, 2024-01-15:
> > > > 抄的这位大佬的，在本地试过，不会下载整个库，只会下载指定的目录，并移动到根目录。 https://github.com/kenzok8/small-package
> > > > ```
> > > > function git_sparse_clone() {
> > > > branch="$1" rurl="$2" localdir="$3" && shift 3
> > > > git clone -b $branch --depth 1 --filter=blob:none --sparse $rurl $localdir
> > > > cd $localdir
> > > > git sparse-checkout init --cone
> > > > git sparse-checkout set $@
> > > > mv -n $@ ../
> > > > cd ..
> > > > rm -rf $localdir
> > > > }
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > 使用方式： `git_sparse_clone openwrt-23.05 "https://github.com/openwrt/openwrt" "openwrt" package/base-files package/network/config/firewall4`
> > > 
> > > 
> > > zhe 这明显也是拉的整个库，只不过留下需要的，其余的删了
> > 
> > 
> > 我一步一步的试过了，一开始只会下载一个.git目录和根目录的文件，不会下载其他多余的文件。后边的命令会下载指定的目录和文件。
> 
> 真是只下载指定文件夹。我在这个基础上，结合几位大佬的优化了下，不错。[可以参考这里](https://github.com/binge8/op)
> 
> ```shell
> function merge_package(){
>     # 参数1是分支名,参数2是库地址。所有文件下载到openwrt/package/openwrt-packages路径。
>     # 同一个仓库下载多个文件夹直接在后面跟文件名或路径，空格分开。
>     trap 'rm -rf "$tmpdir"' EXIT
>     branch="$1" curl="$2" && shift 2
>     rootdir="$PWD"
>     localdir=package/openwrt-packages
>     [ -d "$localdir" ] || mkdir -p "$localdir"
>     tmpdir="$(mktemp -d)" || exit 1
>     git clone -b "$branch" --depth 1 --filter=blob:none --sparse "$curl" "$tmpdir"
>     cd "$tmpdir"
>     git sparse-checkout init --cone
>     git sparse-checkout set "$@"
>     mv -f "$@" "$rootdir"/"$localdir" && cd "$rootdir"
> }
> merge_package master https://github.com/WYC-2020/openwrt-packages luci-app-eqos luci-app-openclash luci-app-ddnsto ddnsto 
> merge_package master https://github.com/lisaac/luci-app-dockerman applications/luci-app-dockerman
> ```

坦白说，不太希望所有文件都统一下载到同一个目标路径，最好是每条merge都可以指定一个地址$3，所有我调整了一下代码:

#!/bin/bash

function merge_package(){
    # 参数1是分支名,参数2是库地址。所有文件下载到指定路径。
    # 同一个仓库下载多个文件夹直接在后面跟文件名或路径，空格分开。
    trap 'rm -rf "$tmpdir"' EXIT
    branch="$1" curl="$2" target_dir="$3" && shift 3
    rootdir="$PWD"
    localdir="$target_dir"
    [ -d "$localdir" ] || mkdir -p "$localdir"
    tmpdir="$(mktemp -d)" || exit 1
    git clone -b "$branch" --depth 1 --filter=blob:none --sparse "$curl" "$tmpdir"
    cd "$tmpdir"
    git sparse-checkout init --cone
    git sparse-checkout set "$@"
    mv -f "$@" "$rootdir"/"$localdir" && cd "$rootdir"
}

# 添加额外软件到不同目录
merge_package master https://github.com/WYC-2020/openwrt-packages package/lean luci-app-eqos luci-app-openclash luci-app-ddnsto ddnsto 
merge_package master https://github.com/lisaac/luci-app-dockerman package/lean applications/luci-app-dockerman

ywt114, 2024-01-15:
> 真是只下载指定文件夹。我在这个基础上，结合几位大佬的优化了下，不错。[可以参考这里](https://github.com/binge8/op)
>
> ```shell
> function merge_package(){
>     # 参数1是分支名,参数2是库地址。所有文件下载到openwrt/package/openwrt-packages路径。
>     # [...]

还是喜欢每天merge都有一个目标目录，所有调整了下:
> ```shell
>#!/bin/bash

>function merge_package(){
>    # 参数1是分支名,参数2是库地址。所有文件下载到指定路径。
>    # 同一个仓库下载多个文件夹直接在后面跟文件名或路径，空格分开。
>    trap 'rm -rf "$tmpdir"' EXIT
>    branch="$1" curl="$2" target_dir="$3" && shift 3
>    rootdir="$PWD"
>    localdir="$target_dir"
>    [ -d "$localdir" ] || mkdir -p "$localdir"
>    tmpdir="$(mktemp -d)" || exit 1
>    git clone -b "$branch" --depth 1 --filter=blob:none --sparse "$curl" "$tmpdir"
>    cd "$tmpdir"
>    git sparse-checkout init --cone
>    git sparse-checkout set "$@"
>    # 使用循环逐个移动文件夹
>    for folder in "$@"; do
>        mv -f "$folder" "$rootdir/$localdir"
>    done
>    cd "$rootdir"
>}
> merge_package master https://github.com/WYC-2020/openwrt-packages package/openwrt-packages luci-app-eqos luci-app-openclash luci-app-ddnsto ddnsto 
> merge_package master https://github.com/lisaac/luci-app-dockerman package/lean applications/luci-app-dockerman
> ```

ZqinKing, 2024-03-01:
> 我就羡慕又讨厌你们这些会写正则表达式的人 :P

这里有正则？

smdx, 2024-09-22:
各位大佬，如果是要拉取一个旧的推送应该怎么用啊？谢谢

coolsnowwolf, 2026-09-05:
该 issue 没有 LEDE 源码故障或可修复日志，而是询问如何只拉取第三方仓库的子目录。当前推荐使用 Git sparse-checkout/partial clone（例如 `git clone --filter=blob:none --sparse` 后 `git sparse-checkout set <path>`）；评论中已经给出了可用脚本和示例。LEDE 本身也不依赖 SVN 获取这些包，因此无需仓库改动，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf