coolsnowwolf/lede #11760: 编译出错，大佬看看是哪里出错了。

caixin1218, 2024-01-11:
### 详细叙述

git pull 更新后编译就出错了，11号10点前还是没有问题的。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

  INSTALL /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/user_headers/include
make[5]: Leaving directory '/ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32'
grep '=[ym]' /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set | LC_ALL=C sort | /ubuntu/family/staging_dir/host/bin/mkhash md5 > /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.vermagic
touch /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.configured
rm -f /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/vmlinux /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/System.map
make -C /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32 KCFLAGS="-fmacro-prefix-map=/ubuntu/family/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/ubuntu/family/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jun  8 19:08:07 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/ubuntu/family/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.32 CC="x86_64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32'
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/conf
Segmentation fault
scripts/Kconfig.include:44: Sorry, this C compiler is not supported.
make[6]: *** [scripts/kconfig/Makefile:77: olddefconfig] Error 1
make[5]: *** [Makefile:704: olddefconfig] Error 2
make[5]: Leaving directory '/ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32'
make[4]: *** [Makefile:28: /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.modules] Error 2
make[4]: Leaving directory '/ubuntu/family/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/ubuntu/family/target/linux'
time: target/linux/compile#36.82#13.76#42.44
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/ubuntu/family'
make[1]: *** [target/Makefile:23: /ubuntu/family/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/ubuntu/family'
make: *** [/ubuntu/family/include/toplevel.mk:231: world] Error 2
caixin@ubuntu:/ubuntu/family$ make V=s -j1
make[2]: Entering directory '/ubuntu/family/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/ubuntu/family/scripts/config'
make[1]: Entering directory '/ubuntu/family'
make[2]: Entering directory '/ubuntu/family'
make[3]: Entering directory '/ubuntu/family/target/linux'
make[4]: Entering directory '/ubuntu/family/target/linux/x86'
rm -f /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/localversion
/ubuntu/family/scripts/kconfig.pl  + + /ubuntu/family/target/linux/generic/config-6.1 /ubuntu/family/target/linux/x86/config-6.1 /ubuntu/family/target/linux/x86/64/config-6.1 > /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /ubuntu/family/.config >> /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.target
/ubuntu/family/scripts/package-metadata.pl kconfig /ubuntu/family/tmp/.packageinfo /ubuntu/family/.config 6.1 > /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.override
/ubuntu/family/scripts/kconfig.pl 'm+' '+' /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.target /dev/null /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.override > /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set
mv /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.old
grep -v INITRAMFS /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.old > /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set
rm -rf /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.prev || { cp /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config; cp /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.prev; }
export MAKEFLAGS= ; [ -d /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/user_headers ] || make -C /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32 KCFLAGS="-fmacro-prefix-map=/ubuntu/family/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/ubuntu/family/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jun  8 19:08:07 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/ubuntu/family/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.32 CC="x86_64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/user_headers headers_install
grep '=[ym]' /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.config.set | LC_ALL=C sort | /ubuntu/family/staging_dir/host/bin/mkhash md5 > /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.vermagic
touch /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/.configured
rm -f /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/vmlinux /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32/System.map
make -C /ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32 KCFLAGS="-fmacro-prefix-map=/ubuntu/family/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/ubuntu/family/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jun  8 19:08:07 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/ubuntu/family/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.32 CC="x86_64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32'
Segmentation fault
scripts/Kconfig.include:44: Sorry, this C compiler is not supported.
make[6]: *** [scripts/kconfig/Makefile:77: olddefconfig] Error 1
make[5]: *** [Makefile:704: olddefconfig] Error 2
make[5]: Leaving directory '/ubuntu/family/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.32'
[... cut ...]
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/ubuntu/family/target/linux'
time: target/linux/compile#0.73#0.22#0.96
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/ubuntu/family'
make[1]: *** [target/Makefile:23: /ubuntu/family/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/ubuntu/family'
make: *** [/ubuntu/family/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2024-01-11:
make clean

caixin1218, 2024-01-11:
make clean 也不行，但20:30后已经可以正常编译了。