coolsnowwolf/lede #11764: 6.6内核支持代码和工具相关的几个提交后编译有问题

GreatMichaelLee, 2024-01-12:
### 详细叙述

如题，编译提示好像找不到python路径

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

touch stamp-system-gdbinit
touch stamp-syscalls
make[8]: Leaving directory '/home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/gdb/data-directory'
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/gdb'
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/gdb'
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1'
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1'
touch /home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/.built
mkdir -p /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin
install -m0755 /home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/gdb/gdb /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gdb
ln -fs x86_64-openwrt-linux-musl-gdb /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-gdb
strip /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gdb
mkdir -p /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/share/gdb
cp -R /home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/gdb/data-directory/python /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/share/gdb/
cp: cannot stat '/home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/gdb/data-directory/python': No such file or directory
make[3]: [Makefile:82: /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.gdb_installed] Error 1 (ignored)
cp -R /home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/gdb/data-directory/syscalls /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/share/gdb/
cp -R /home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/gdb/data-directory/system-gdbinit /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/share/gdb/
mkdir -p /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp
touch /home/builder/lede_x86/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gdb-14.1/.built
touch /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.gdb_installed
make[3]: Leaving directory '/home/builder/lede_x86/toolchain/gdb'
time: toolchain/gdb/compile#594.36#118.95#234.67
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [toolchain/Makefile:93: /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231: world] Error 2
builder@Supernode-TX:~/lede_x86$

kxdn3, 2024-01-12:
我就问问可以编译6.6内核了吗

lqhorochi, 2024-01-12:
一样的问题...

coolsnowwolf, 2024-01-13:
重新生成 .config

coolsnowwolf, 2024-01-13:
rm -rf staging_dir

GreatMichaelLee, 2024-01-13:
> rm -rf staging_dir

没用，依旧报错

mkdir -p /home/builder/lede_x86/staging_dir/host/stamp
touch /home/builder/lede_x86/build_dir/host/zstd-1.5.4/.built
touch /home/builder/lede_x86/staging_dir/host/stamp/.zstd_installed
make[3]: Leaving directory '/home/builder/lede_x86/tools/zstd'
time: tools/zstd/compile#16.56#1.03#17.59
make[3]: Entering directory '/home/builder/lede_x86/tools/cmake'
MAKEFLAGS="" /home/builder/lede_x86/staging_dir/host/bin/ninja -v -j1 -C /home/builder/lede_x86/build_dir/host/cmake-3.25.2 install
ninja: Entering directory `/home/builder/lede_x86/build_dir/host/cmake-3.25.2'
ninja: error: '/home/builder/lede_x86/staging_dir/host/lib/libzstd.so', needed by 'bin/cmake', missing and no known rule to make it
make[3]: *** [Makefile:59: /home/builder/lede_x86/staging_dir/host/stamp/.cmake_installed] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/tools/cmake'
time: tools/cmake/compile#0.13#0.17#0.22
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:221: tools/cmake/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [tools/Makefile:217: /home/builder/lede_x86/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231: world] Error 2
builder@Supernode-TX:~/lede_x86$

GreatMichaelLee, 2024-01-13:
另一台编译机上报错不一样：

make[8]: Leaving directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma/api'
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma/api'
make[7]: Entering directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma'
make[8]: Entering directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma'
 /usr/bin/mkdir -p '/home/builder/lede_x86/staging_dir/host/lib'
 /usr/bin/env bash ../../libtool   --mode=install /home/builder/lede_x86/staging_dir/host/bin/install -c   liblzma.la '/home/builder/lede_x86/staging_dir/host/lib'
 /usr/bin/mkdir -p '/home/builder/lede_x86/staging_dir/host/lib/pkgconfig'
 /home/builder/lede_x86/staging_dir/host/bin/install -c -m 644 liblzma.pc '/home/builder/lede_x86/staging_dir/host/lib/pkgconfig'
../../libtool: line 960: /home/builder/lede_x86/staging_dir/host/bin/sed: No such file or directory
../../libtool: line 966: /home/builder/lede_x86/staging_dir/host/bin/sed: No such file or directory
../../libtool: line 1661: /home/builder/lede_x86/staging_dir/host/bin/sed: No such file or directory
../../libtool: line 1661: /home/builder/lede_x86/staging_dir/host/bin/sed: No such file or directory
../../libtool: line 1661: /home/builder/lede_x86/staging_dir/host/bin/sed: No such file or directory
../../libtool: line 1661: /home/builder/lede_x86/staging_dir/host/bin/sed: No such file or directory
/home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma/: /home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma/: Is a directory
make[8]: *** [Makefile:924: install-libLTLIBRARIES] Error 126
make[8]: Leaving directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma'
make[7]: *** [Makefile:1905: install-am] Error 2
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma'
make[6]: *** [Makefile:1742: install-recursive] Error 1
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1/src/liblzma'
make[5]: *** [Makefile:426: install-recursive] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1/src'
make[4]: *** [Makefile:623: install-recursive] Error 1
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/host/xz-5.4.1'
make[3]: *** [Makefile:40: /home/builder/lede_x86/staging_dir/host/stamp/.xz_installed] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/tools/xz'
time: tools/xz/compile#12.73#5.55#2.83
    ERROR: tools/xz failed to build.
make[2]: *** [tools/Makefile:221: tools/xz/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [tools/Makefile:217: /home/builder/lede_x86/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

GreatMichaelLee, 2024-01-13:
以上全部都是删除staging_dir 和.config重新make menuconfig之后报的错

coolsnowwolf, 2024-01-14:
tmp 没删除

closed 2024-01-14 by coolsnowwolf