coolsnowwolf/lede #11766: 编译x86出错了

free-diy, 2024-01-12:
### 详细叙述

x86编译报错了

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

make[4]: *** [Makefile:1226: tools/chacl.o] Error 1
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/acl-2.3.1'
make[3]: *** [Makefile:87: /workdir/openwrt/build_dir/target-x86_64_musl/acl-2.3.1/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/acl'
time: package/feeds/packages/acl/compile#0.15#0.06#0.19
    ERROR: package/feeds/packages/acl failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/acl/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

bi4tms, 2024-01-12:
github action在线编译出现同样错误

lifetyper, 2024-01-13:
看来不止我一个

caixin1218, 2024-01-13:
> ### 详细叙述
> x86编译报错了
> 
> ### 重复 issue
> * [x] 没有类似的 issue
> 
> ### 具体型号
> x86的
> 
> ### 详细日志
> make[4]： *** [Makefile：1226： tools/chacl.o] 错误 1 make[4]： 离开目录 '/workdir/openwrt/build_dir/target-x86_64_musl/acl-2.3.1' make[3]： *** [Makefile：87： /workdir/openwrt/build_dir/target-x86_64_musl/acl-2.3.1/.built] 错误 2 make[3]： 离开目录 '/workdir/openwrt/feeds/packages/utils/acl' 时间： package/feeds/packages/acl/compile#0.15#0.06#0.19 错误：package/feeds/packages/acl 构建失败。 make[2]： *** [package/Makefile：116： package/feeds/packages/acl/compile] 错误 1 make[2]： 离开目录 '/workdir/openwrt' make[1]： *** [package/Makefile：110： /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] 错误 2 make[1]：离开目录 '/workdir/openwrt' make： *** [/workdir/openwrt/include/toplevel.mk：231： world] 错误 2 错误：进程已完成，退出代码为 2。

太少，看不出问题，这是我11：40编译完成的，可能是没有 acl 就不会失败。
Signing package index...
make[2]: Leaving directory '/ubuntu/peaceful'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/ubuntu/peaceful'
WORK_DIR=/ubuntu/peaceful/build_dir/target-x86_64_musl/json_info_files /ubuntu/peaceful/scripts/json_overview_image_info.py /ubuntu/peaceful/bin/targets/x86/64/profiles.json 
make[2]: Leaving directory '/ubuntu/peaceful'
export MAKEFLAGS= ;make -w -r checksum

coolsnowwolf, 2024-01-13:
没有 j1 日志

lifetyper, 2024-01-13:
``
 make[3] -C package/kernel/linux compile
    ERROR: package/libs/libselinux failed to build.
```
```
make[4]: *** [Makefile:1226: tools/chacl.o] Error 1
make[4]: Leaving directory '/home/lifetyper/Router/openwrt/x86/build_dir/target-x86_64_musl/acl-2.3.1'
make[3]: *** [Makefile:87: /home/lifetyper/Router/openwrt/x86/build_dir/target-x86_64_musl/acl-2.3.1/.built] Error 2
make[3]: Leaving directory '/home/lifetyper/Router/openwrt/x86/feeds/packages/utils/acl'
time: package/feeds/packages/acl/compile#7.34#1.10#11.21
    ERROR: package/feeds/packages/acl failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/acl/compile] Error 1
make[2]: Leaving directory '/home/lifetyper/Router/openwrt/x86'
make[1]: *** [package/Makefile:110: /home/lifetyper/Router/openwrt/x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lifetyper/Router/openwrt/x86'
make: *** [/home/lifetyper/Router/openwrt/x86/include/toplevel.mk:231: world] Error 2
```

binge8, 2024-01-13:
两个报错，日志来了
[报错2.txt](https://github.com/coolsnowwolf/lede/files/13926748/2.txt)
[报错.txt](https://github.com/coolsnowwolf/lede/files/13926749/default.txt)

poweryb9802, 2024-01-13:
Makefile:1226: recipe for target 'tools/chacl.o' failed
make[4]: *** [tools/chacl.o] Error 1
make[4]: Leaving directory '/home/lede/lede/build_dir/target-x86_64_musl/acl-2.3.1'
Makefile:84: recipe for target '/home/lede/lede/build_dir/target-x86_64_musl/acl-2.3.1/.built' failed
make[3]: *** [/home/lede/lede/build_dir/target-x86_64_musl/acl-2.3.1/.built] Error 2
make[3]: Leaving directory '/home/lede/lede/feeds/packages/utils/acl'
time: package/feeds/packages/acl/compile#0.14#0.03#0.16
    ERROR: package/feeds/packages/acl failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/acl/compile' failed
make[2]: *** [package/feeds/packages/acl/compile] Error 1
make[2]: Leaving directory '/home/lede/lede'
package/Makefile:110: recipe for target '/home/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lede/lede'
/home/lede/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

closed 2024-01-13 by coolsnowwolf