coolsnowwolf/lede #11767: 编译错误

gswangmd-gssi, 2024-01-13:
### 详细叙述

更新代码后编译错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

r5s

### 详细日志

make[3]: Leaving directory '/home/gswangmd/lede/package/utils/e2fsprogs'
time: package/utils/e2fsprogs/compile#21.94#3.02#25.29
    ERROR: package/utils/e2fsprogs failed to build.
make[2]: *** [package/Makefile:116: package/utils/e2fsprogs/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [package/Makefile:110: /home/gswangmd/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:231: world] Error 2

shanghaiqw, 2024-01-13:
刚才acl错误解决了  又出现和你一样的错误

coolsnowwolf, 2024-01-13:
rm -rf staging_dir

closed 2024-01-13 by coolsnowwolf