coolsnowwolf/lede #11770: 依赖报错

caixin1218, 2024-01-13:
### 详细叙述

$ ./scripts/feeds install -a
提示依赖不存在

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

$ ./scripts/feeds install -a
WARNING: Makefile 'package/feeds/packages/clamav/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/clamav/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/exim/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/exim/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/exim/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/exim/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/fish/Makefile' has a dependency on 'libpcre2-32', which does not exist
WARNING: Makefile 'package/feeds/packages/glib2/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/glib2/Makefile' has a build dependency on 'pcre2/host', which does not exist
WARNING: Makefile 'package/feeds/packages/grep/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/libyang/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/lighttpd/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/lighttpd/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/lighttpd/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/php7/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/php8/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/php8/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/qtbase/Makefile' has a dependency on 'libpcre2-16', which does not exist
WARNING: Makefile 'package/feeds/packages/ripgrep/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/telephony/rtpengine/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/telephony/rtpengine/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/telephony/sipgrep/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/telephony/sngrep/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/sslh/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/feeds/packages/tvheadend/Makefile' has a dependency on 'libpcre2', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed helloworld.
Installing all packages from feed Alist.

WYC-2020, 2024-01-13:
这玩意就在 package这个仓库里面 你怕不是没有./scripts/feeds update -a  吧

caixin1218, 2024-01-14:
> 这玩意就在 package这个仓库里面 你怕不是没有./scripts/feeds update -a 吧

重新拉取的源码，正常编译，但只要是git tag 20230609这个分支就会报依赖错误。./scripts/feeds update -a也不行。

closed 2024-01-14 by coolsnowwolf