coolsnowwolf/lede #11771: 编译错误

gswangmd-gssi, 2024-01-13:
### 详细叙述

R5S依然存在编译错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

r5s

### 详细日志

mkfs/rootdir.c: In function 'btrfs_mkfs_shrink_fs':
mkfs/rootdir.c:917:23: error: storage size of 'file_stat' isn't known
  917 |         struct stat64 file_stat;
      |                       ^~~~~~~~~
make[4]: *** [Makefile:414: mkfs/rootdir.o] Error 1
make[4]: Leaving directory '/home/gswangmd/lede/build_dir/target-aarch64_generic_musl/btrfs-progs-v5.16.1'
make[3]: *** [Makefile:100: /home/gswangmd/lede/build_dir/target-aarch64_generic_musl/btrfs-progs-v5.16.1/.built] Error 2
make[3]: Leaving directory '/home/gswangmd/lede/feeds/packages/utils/btrfs-progs'
time: package/feeds/packages/btrfs-progs/compile#11.99#0.94#14.04
    ERROR: package/feeds/packages/btrfs-progs failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/btrfs-progs/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [package/Makefile:110: /home/gswangmd/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:231: world] Error 2
gswangmd@wmdhonor:~/lede$

WYC-2020, 2024-01-13:
这升级musl导致的而已，看这个应该和文件的操作那部分有问题，等会有时间的话 修下

WYC-2020, 2024-01-13:
增加红框里面的 很早我就修复过了
[image]
ifeq ($(CONFIG_USE_MUSL),y)
TARGET_CFLAGS += -D_LARGEFILE64_SOURCE
endif

qiyuey, 2024-01-13:
> 增加红框里面的 很早我就修复过了 [...]

感谢，我手动修改之后已经可以正常编译了，不过这个提交似乎并没有进入 https://github.com/coolsnowwolf/packages 中

jiscsafe, 2024-01-14:
我是个小白，我是使用Action编译的，麻烦帮忙指导下如何修改这个问题呢？ @WYC-2020 @qiyuey

closed 2024-01-14 by coolsnowwolf