coolsnowwolf/lede #11772: 编译报错，求大佬们帮忙看看

xianfanjie, 2024-01-13:
### 详细叙述

编译过程报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ipq60xx_generic_DEVICE_360_v6

### 详细日志

aarch64-openwrt-linux-musl-gcc -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xianfanjie/lede/build_dir/target-aarch64_cortex-a53_musl/libselinux-3.3=libselinux-3.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include  -I../include -D_GNU_SOURCE -DNO_ANDROID_BACKEND -I/home/xianfanjie/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include   -c -o selinux_check_securetty_context.o selinux_check_securetty_context.c
aarch64-openwrt-linux-musl-gcc -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xianfanjie/lede/build_dir/target-aarch64_cortex-a53_musl/libselinux-3.3=libselinux-3.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include  -I../include -D_GNU_SOURCE -DNO_ANDROID_BACKEND -I/home/xianfanjie/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include   -c -o selinux_config.o selinux_config.c
aarch64-openwrt-linux-musl-gcc -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xianfanjie/lede/build_dir/target-aarch64_cortex-a53_musl/libselinux-3.3=libselinux-3.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify -I/home/xianfanjie/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include  -I../include -D_GNU_SOURCE -DNO_ANDROID_BACKEND -I/home/xianfanjie/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include   -c -o selinux_restorecon.o selinux_restorecon.c
selinux_restorecon.c: In function 'filespec_add':
selinux_restorecon.c:425:23: error: storage size of 'sb' isn't known
  425 |         struct stat64 sb;
      |                       ^~
selinux_restorecon.c:438:31: warning: implicit declaration of function 'lstat64'; did you mean 'lstat'? [-Wimplicit-function-declaration]
  438 |                         ret = lstat64(fl->file, &sb);
      |                               ^~~~~~~
      |                               lstat
make[5]: *** [Makefile:167: selinux_restorecon.o] Error 1
make[5]: Leaving directory '/home/xianfanjie/lede/build_dir/target-aarch64_cortex-a53_musl/libselinux-3.3/src'
make[4]: *** [Makefile:48: all] Error 1
make[4]: Leaving directory '/home/xianfanjie/lede/build_dir/target-aarch64_cortex-a53_musl/libselinux-3.3'
make[3]: *** [Makefile:161: /home/xianfanjie/lede/build_dir/target-aarch64_cortex-a53_musl/libselinux-3.3/.built] Error 2
make[3]: Leaving directory '/home/xianfanjie/lede/package/libs/libselinux'
time: package/libs/libselinux/compile#5.19#1.07#17.90
    ERROR: package/libs/libselinux failed to build.
make[2]: *** [package/Makefile:116: package/libs/libselinux/compile] Error 1
make[2]: Leaving directory '/home/xianfanjie/lede'
make[1]: *** [package/Makefile:110: /home/xianfanjie/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xianfanjie/lede'
make: *** [/home/xianfanjie/lede/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2024-01-13:
[image]
如此修复，没修复host的情况l这边

xianfanjie, 2024-01-13:
> ![image](https://private-user-images.githubusercontent.com/34025388/296496455-780c5e50-41f0-4ce1-9ecc-ecc439467d74.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

谢谢大佬，成功编译了。

closed 2024-01-13 by xianfanjie