coolsnowwolf/lede #11773: 最新源码编译出错...

lqhorochi, 2024-01-13:
### 详细叙述

更新了最新源码直接报错....

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

checking whether basename is declared... yes
checking whether ffs is declared... yes
checking whether stpcpy is declared... yes
checking whether asprintf is declared... yes
checking whether vasprintf is declared... yes
checking whether strnlen is declared... yes
checking whether ___lc_codepage_func is declared... no
checking pkg-config is at least version 0.9.0... yes
checking for libzstd >= 1.4.0... no
configure: error: --with-zstd was given, but pkgconfig/libzstd.pc is not found
make[5]: *** [Makefile:3018: configure-bfd] Error 1
make[5]: Leaving directory '/home/kenneth/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40'
make[4]: *** [Makefile:1005: all] Error 2
make[4]: Leaving directory '/home/kenneth/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40'
make[3]: *** [Makefile:112: /home/kenneth/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built] Error 2
make[3]: Leaving directory '/home/kenneth/lede/toolchain/binutils'
time: toolchain/binutils/compile#1.53#0.34#6.44
    ERROR: toolchain/binutils failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/binutils/compile] Error 1
make[2]: Leaving directory '/home/kenneth/lede'
make[1]: *** [toolchain/Makefile:93: /home/kenneth/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/kenneth/lede'
make: *** [/home/kenneth/lede/include/toplevel.mk:231：world] 错误 2

flyhong666, 2024-01-13:
同样的错误

WYC-2020, 2024-01-13:
问题出在这里 
[image]
 配置要使用libzstd但是没找到这个玩意，这个东西 我这边 编译没任何问题 不过我是自己gcc修改成用的13 应该不是这个问题导致的 建议你 用万能大法重新试下

bi4tms, 2024-01-13:
make distclean然后重新编译，我在线重新编译的没有问题

closed 2024-01-14 by coolsnowwolf