coolsnowwolf/lede #11774: 最新源码编译错误

qq1316237299, 2024-01-13:
### 详细叙述

最新源码出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

/usr/bin/ld: warning: libgcc_s.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010001
/usr/bin/ld: warning: libgcc_s.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010002
/usr/bin/ld: warning: libgcc_s.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010001
/usr/bin/ld: warning: libgcc_s.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010002
/usr/bin/ld: warning: libc.so, needed by libgcc_s.so.1, not found (try using -rpath or -rpath-link)
libgcc_s.so.1: undefined reference to `pthread_getspecific'
libgcc_s.so.1: undefined reference to `pthread_key_create'
libgcc_s.so.1: undefined reference to `pthread_once'
libgcc_s.so.1: undefined reference to `pthread_setspecific'
collect2: error: ld returned 1 exit status
/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/c/Make-lang.in:86: recipe for target 'cc1' failed
make[6]: *** [cc1] Error 1
make[6]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/gcc'
Makefile:4532: recipe for target 'install-gcc' failed
make[5]: *** [install-gcc] Error 2
make[5]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
Makefile:2444: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
Makefile:89: recipe for target '/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.gcc_final_installed' failed
make[3]: *** [/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.gcc_final_installed] Error 2
make[3]: Leaving directory '/home/qq1316237299/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#1.60#9.07#12.44
    ERROR: toolchain/gcc/final failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/gcc/final/compile' failed
make[2]: *** [toolchain/gcc/final/compile] Error 1
make[2]: Leaving directory '/home/qq1316237299/lede'
toolchain/Makefile:93: recipe for target '/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/qq1316237299/lede'
/home/qq1316237299/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
[image]

qq1316237299, 2024-01-14:
有没有大佬给个解决方案呢

closed 2024-01-14 by qq1316237299

nary24, 2024-08-01:
请问下，最终怎么处理的呢

qq1316237299, 2024-08-01:
> 请问下，最终怎么处理的呢

删总源码 重新编译了