coolsnowwolf/lede #11779: 6.1.72最新源码出错请大佬们帮我看下，给个解决方案

qq1316237299, 2024-01-14:
### 详细叙述

6.1.72最新源码出错请大佬们帮我看下，给个解决方案 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

cnumber/libdecnumber.a ../libbacktrace/.libs/libbacktrace.a \
                     checksum-options > cc1-checksum.c.tmp && 		 \
  /home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/../move-if-change cc1-checksum.c.tmp cc1-checksum.c; \
fi
g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -o cc1 c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-fold.o c/gimple-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o i386-c.o glibc-c.o \
  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a   -L/home/qq1316237299/lede/staging_dir/host/lib -L/home/qq1316237299/lede/staging_dir/host/lib -L/home/qq1316237299/lede/staging_dir/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz 
/usr/bin/ld: warning: libgcc_s.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010001
/usr/bin/ld: warning: libgcc_s.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010002
/usr/bin/ld: warning: libgcc_s.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010001
/usr/bin/ld: warning: libgcc_s.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0010002
/usr/bin/ld: warning: libc.so, needed by libgcc_s.so.1, not found (try using -rpath or -rpath-link)
libgcc_s.so.1: undefined reference to `pthread_getspecific'
libgcc_s.so.1: undefined reference to `pthread_key_create'
libgcc_s.so.1: undefined reference to `pthread_once'
libgcc_s.so.1: undefined reference to `pthread_setspecific'
collect2: error: ld returned 1 exit status
/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/c/Make-lang.in:86: recipe for target 'cc1' failed
make[6]: *** [cc1] Error 1
make[6]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/gcc'
Makefile:4532: recipe for target 'install-gcc' failed
make[5]: *** [install-gcc] Error 2
make[5]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
Makefile:2444: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
Makefile:89: recipe for target '/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.gcc_final_installed' failed
make[3]: *** [/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.gcc_final_installed] Error 2
make[3]: Leaving directory '/home/qq1316237299/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#1.17#8.94#11.90
    ERROR: toolchain/gcc/final failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/gcc/final/compile' failed
make[2]: *** [toolchain/gcc/final/compile] Error 1
make[2]: Leaving directory '/home/qq1316237299/lede'
toolchain/Makefile:93: recipe for target '/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/qq1316237299/lede'
/home/qq1316237299/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
[image]

qq1316237299, 2024-01-14:
> 最近的更新就是有很多bug的，你这是gcc的问题，我还遇到了别的软件包的问题。

gcc问题怎么修复

qq1316237299, 2024-01-14:
> 如果不着急的话，建议再等等，如果着急的话，先去换官方openwrt的源。

我不急  就是要等办法

qq1316237299, 2024-01-14:
> 你这个问题看上去像是编译环境没有装cmake这个包。 不过这个时候我也不建议你再继续尝试编译了，因为目前发现lede的源中超过20个软件包存在代码bug了，即便是通过安装相应的软件包解决了这个问题你还是会遇到其他软件包错误。解决bug的工程量很大，指望社区迅速解决bug也不现实。

那只能等L大修复了呗

qq1316237299, 2024-01-14:
> > > 你这个问题看上去像是编译环境没有装cmake这个包。 不过这个时候我也不建议你再继续尝试编译了，因为目前发现lede的源中超过20个软件包存在代码bug了，即便是通过安装相应的软件包解决了这个问题你还是会遇到其他软件包错误。解决bug的工程量很大，指望社区迅速解决bug也不现实。
> > 
> > 
> > 那只能等L大修复了呗
> 
> 是的，所以如果着急的话，现在换openwrt官方源是最正确的选择。

官方源重来没用过，一直用的L大的 更新的快  听说官方的还不能用跨境的吗？

qq1316237299, 2024-01-14:
checking for fls... no
checking for getgid... yes
checking for getpagesize... yes
checking for getrlimit... yes
checking for getuid... yes
checking for sysconf... yes
checking whether basename is declared... yes
checking whether ffs is declared... yes
checking whether stpcpy is declared... yes
checking whether asprintf is declared... yes
checking whether vasprintf is declared... yes
checking whether strnlen is declared... yes
checking whether ___lc_codepage_func is declared... no
checking pkg-config is at least version 0.9.0... yes
checking for libzstd >= 1.4.0... no
configure: error: --with-zstd was given, but pkgconfig/libzstd.pc is not found
Makefile:3018: recipe for target 'configure-bfd' failed
make[5]: *** [configure-bfd] Error 1
make[5]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40'
Makefile:1004: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40'
Makefile:111: recipe for target '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built' failed
make[3]: *** [/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built] Error 2
make[3]: Leaving directory '/home/qq1316237299/lede/toolchain/binutils'
time: toolchain/binutils/compile#1.56#1.09#2.51
    ERROR: toolchain/binutils failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/binutils/compile' failed
make[2]: *** [toolchain/binutils/compile] Error 1
make[2]: Leaving directory '/home/qq1316237299/lede'
toolchain/Makefile:93: recipe for target '/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/qq1316237299/lede'
/home/qq1316237299/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
[image]
这是最新的 我全部卸载掉搞过了

qq1316237299, 2024-01-14:
错误还挺多的

qq1316237299, 2024-01-14:
> > checking for fls... no checking for getgid... yes checking for getpagesize... yes checking for getrlimit... yes checking for getuid... yes checking for sysconf... yes checking whether basename is declared... yes checking whether ffs is declared... yes checking whether stpcpy is declared... yes checking whether asprintf is declared... yes checking whether vasprintf is declared... yes checking whether strnlen is declared... yes checking whether ___lc_codepage_func is declared... no checking pkg-config is at least version 0.9.0... yes checking for libzstd >= 1.4.0... no configure: error: --with-zstd was given, but pkgconfig/libzstd.pc is not found Makefile:3018: recipe for target 'configure-bfd' failed make[5]: *** [configure-bfd] Error 1 make[5]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40' Makefile:1004: recipe for target 'all' failed make[4]: *** [all] Error 2 make[4]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40' Makefile:111: recipe for target '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built' failed make[3]: *** [/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built] Error 2 make[3]: Leaving directory '/home/qq1316237299/lede/toolchain/binutils' time: toolchain/binutils/compile#1.56#1.09#2.51 ERROR: toolchain/binutils failed to build. toolchain/Makefile:95: recipe for target 'toolchain/binutils/compile' failed make[2]: *** [toolchain/binutils/compile] Error 1 make[2]: Leaving directory '/home/qq1316237299/lede' toolchain/Makefile:93: recipe for target '/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile' failed make[1]: *** [/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2 make[1]: Leaving directory '/home/qq1316237299/lede' /home/qq1316237299/lede/include/toplevel.mk:229: recipe for target 'world' failed make: *** [world] Error 2 ![QQ截图20240114192748](https://private-user-images.githubusercontent.com/42686656/296546601-40f842ab-7d22-4550-9bd7-d5717a7b8cec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyMzE5NjksIm5iZiI6MTcwNTIzMTY2OSwicGF0aCI6Ii80MjY4NjY1Ni8yOTY1NDY2MDEtNDBmODQyYWItN2QyMi00NTUwLTliZDctZDU3MTdhN2I4Y2VjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE0VDExMjc0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzYzJjNGYyZmJjNmRkNTA3YzUzNjUyMjlhMTE4ODAxMTU5ZjIzNmZkYTUyYWJmYzU5NzdmY2I0NzBmM2M5N2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IWTht2aOrc6l5C-tt3nWO281msScRiY05lmew7kunfE) 这是最新的 我全部卸载掉搞过了
> 
> 你看吧，不止有一个问题。

是的，错误还挺多的

qq1316237299, 2024-01-14:
> > > 检查 FLS...没有检查getgid...是的，正在检查 getPageSize...是的，正在检查 getrlimit...是的，正在检查 getuid...是正在检查 sysconf...是的，检查是否声明了 basename...是的，检查是否声明了 FFS...是的，检查是否声明了 stpcpy...是的，检查是否声明了 asprintf...是的，检查是否声明了 vasprintf...是的，检查是否声明了 strnlen...是的，检查是否声明___lc_codepage_func...没有检查 pkg-config 至少是 0.9.0 版本...是的，检查 libzstd >= 1.4.0...没有配置：错误：给出了--with-zstd，但找不到pkgconfig/libzstd.pc Makefile：3018：目标“configure-bfd”的配方失败 make[5]：*** [configure-bfd] 错误 1 make[5]：离开目录“/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40” Makefile：1004：目标“all”的配方失败 make[4]：*** [all] 错误 2 make[4]：离开目录“/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40” Makefile：111： 目标“/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built”的配方失败 make[3]：*** [/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built] 错误 2 make[3]：离开目录“/home/qq1316237299/lede/toolchain/binutils”时间：toolchain/binutils/compile#1.56#1.09#2.51 错误：toolchain/binutils 无法构建。toolchain/Makefile：95：目标“toolchain/binutils/compile”的配方失败 make[2]：*** [toolchain/binutils/compile] 错误 1 make[2]：离开目录“/home/qq1316237299/lede”工具链/Makefile：93：目标“/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile”的配方失败 make[1]：*** [/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] 错误 2 make[1]： 离开目录“/home/qq1316237299/lede” /home/qq1316237299/lede/include/toplevel.mk：229：目标“world”的配方失败 make： *** [world] Error 2 ![QQ截图20240114192748](https://private-user-images.githubusercontent.com/42686656/296546601-40f842ab-7d22-4550-9bd7-d5717a7b8cec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyMzE5NjksIm5iZiI6MTcwNTIzMTY2OSwicGF0aCI6Ii80MjY4NjY1Ni8yOTY1NDY2MDEtNDBmODQyYWItN2QyMi00NTUwLTliZDctZDU3MTdhN2I4Y2VjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE0VDExMjc0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzYzJjNGYyZmJjNmRkNTA3YzUzNjUyMjlhMTE4ODAxMTU5ZjIzNmZkYTUyYWJmYzU5NzdmY2I0NzBmM2M5N2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IWTht2aOrc6l5C-tt3nWO281msScRiY05lmew7kunfE) 这是最新的 我全部卸载掉搞过了
> > 
> > 
> > 你看吧，不止有一个问题。
> 
> 是的，错误还挺多的

希望L大以及社区的大佬们能尽快修复问题

qq1316237299, 2024-01-14:
我记得十号的时候编译还是可以正常编译的

qq1316237299, 2024-01-14:
> 所以我的建议是：暂时不要继续尝试编译了，直到没有人再继续反馈编译的相关issues为止。

checking for fcntl... yes
checking for fdopen... yes
checking for fileno... yes
checking for fls... no
checking for getgid... yes
checking for getpagesize... yes
checking for getrlimit... yes
checking for getuid... yes
checking for sysconf... yes
checking whether basename is declared... yes
checking whether ffs is declared... yes
checking whether stpcpy is declared... yes
checking whether asprintf is declared... yes
checking whether vasprintf is declared... yes
checking whether strnlen is declared... yes
checking whether ___lc_codepage_func is declared... no
checking pkg-config is at least version 0.9.0... yes
checking for libzstd >= 1.4.0... no
configure: error: --with-zstd was given, but pkgconfig/libzstd.pc is not found
Makefile:3018: recipe for target 'configure-bfd' failed
make[5]: *** [configure-bfd] Error 1
make[5]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40'
Makefile:1004: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40'
Makefile:111: recipe for target '/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built' failed
make[3]: *** [/home/qq1316237299/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/binutils-2.40/.built] Error 2
make[3]: Leaving directory '/home/qq1316237299/lede/toolchain/binutils'
time: toolchain/binutils/compile#1.56#1.09#2.51
    ERROR: toolchain/binutils failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/binutils/compile' failed
make[2]: *** [toolchain/binutils/compile] Error 1
make[2]: Leaving directory '/home/qq1316237299/lede'
toolchain/Makefile:93: recipe for target '/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/qq1316237299/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/qq1316237299/lede'
/home/qq1316237299/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
我怀疑是这个之后开始的
[image]

1105384898, 2024-01-14:
make[8]: *** [scripts/Makefile.build:250: net/netfilter/xt_FLOWOFFLOAD.o] Error 1
make[7]: *** [scripts/Makefile.build:500: net/netfilter] Error 2
make[6]: *** [scripts/Makefile.build:500: net] Error 2
make[5]: *** [Makefile:2014: .] Error 2
make[5]: Leaving directory '/home/yangyang/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72'
make[4]: *** [Makefile:29: /home/yangyang/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.modules] Error 2
make[4]: Leaving directory '/home/yangyang/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/yangyang/lede/target/linux'
time: target/linux/compile#1.24#0.61#1.79
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/yangyang/lede'
make[1]: *** [target/Makefile:23: /home/yangyang/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/yangyang/lede'
make: *** [/home/yangyang/lede/include/toplevel.mk:231：world] 错误 2

zhtcl, 2024-01-14:
ake[3] -C toolchain/nasm compile
 make[3] -C toolchain/gcc/initial compile
 make[3] -C toolchain/kernel-headers compile
 make[3] -C toolchain/musl compile
 make[3] -C toolchain/gcc/final compile
 make[2] target/compile
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/Ledemaster/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/Ledemaster/openwrt/scripts/config'
make[1]: Entering directory '/Ledemaster/openwrt'
make[2]: Entering directory '/Ledemaster/openwrt'
make[3]: Entering directory '/Ledemaster/openwrt/target/linux'
make[4]: Entering directory '/Ledemaster/openwrt/target/linux/x86'
rm -f /Ledemaster/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/localversion
/Ledemaster/openwrt/scripts/kconfig.pl  + + /Ledemaster/openwrt/target/linux/generic/config-6.1 /Ledemaster/openwrt/target/linux/x86/config-6.1 /Ledemaster/openwrt/target/linux/x86/64/config-6.1 > /Ledemaster/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /Ledemaster/openwrt/.config >> /Ledemaster/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /Ledemaster/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /Ledemaster/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /Ledemaster/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
/Ledemaster/openwrt/scripts/package-metadata.pl kconfig /Ledemaster/openwrt/tmp/.packageinfo /Ledemaster/openwrt/.config 6.1 > /Ledemaster/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.override
/Ledemaster/openwrt/scripts/kconfig.pl 'm+' '+' /Ledemaster/openwrt/build_dir/targ

coolsnowwolf, 2024-01-14:
你们还挺能憋着不肯 pr 提交的

Aes64X, 2024-01-14:
[image]
🤣

qq1316237299, 2024-01-14:
> ![image](https://private-user-images.githubusercontent.com/47932152/296552364-e3650079-873a-4fd1-9d4b-0638b546a4dc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyMzkwNTIsIm5iZiI6MTcwNTIzODc1MiwicGF0aCI6Ii80NzkzMjE1Mi8yOTY1NTIzNjQtZTM2NTAwNzktODczYS00ZmQxLTlkNGItMDYzOGI1NDZhNGRjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE0VDEzMjU1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThhYzhmNDY5OWUxNTZkODQzNTZkYWEwNTA3OGM2ZDRhNjQwMTNkZGJkYzAzMjIxMjJlMjg0ZTk2NzlhMThmMDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xe7KLMp4FW1tafXJnJpJCOLd4WOgKpuXCs9qe0kEyHI) 🤣

哈哈

coolsnowwolf, 2024-01-14:
rm -rf staging_dir/
rm -rf bin/
rm -rf build_dir/
rm -rf tmp/
rm .config 

之后 make menuconfig 重新生成 .config 即可

qq1316237299, 2024-01-14:
> rm -rf staging_dir/ rm -rf bin/ rm -rf build_dir/ rm -rf tmp/ rm .config
> 
> 之后 make menuconfig 重新生成 .config 即可

大佬你终于来了 谢谢 我试试看

81268, 2024-01-14:
冲呀！
大家可以使用.config.old备份文件

qq1316237299, 2024-01-14:
> 冲呀！ 大家可以使用.config.old备份文件

这个用什么命令 make .config.old吗

qq1316237299, 2024-01-14:
> rm -rf staging_dir/ rm -rf bin/ rm -rf build_dir/ rm -rf tmp/ rm .config
> 
> 之后 make menuconfig 重新生成 .config 即可

make[3]: Leaving directory '/home/qq1316237299/lede/feeds/passwall_packages/dns2socks'
time: package/feeds/passwall_packages/dns2socks/compile#0.07#0.04#0.11
make[3]: Entering directory '/home/qq1316237299/lede/feeds/passwall_packages/dns2tcp'
mkdir -p /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/qq1316237299/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qq1316237299/lede/build_dir/target-x86_64_musl/dns2tcp-1.1.0/.pkgdir/dns2tcp/. /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.dns2tcp_installed
echo "dns2tcp" >> /home/qq1316237299/lede/staging_dir/target-x86_64_musl/pkginfo/dns2tcp.default.install
make[3]: Leaving directory '/home/qq1316237299/lede/feeds/passwall_packages/dns2tcp'
time: package/feeds/passwall_packages/dns2tcp/compile#0.10#0.02#0.12
make[3]: Entering directory '/home/qq1316237299/lede/feeds/passwall_packages/microsocks'
mkdir -p /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/qq1316237299/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qq1316237299/lede/build_dir/target-x86_64_musl/microsocks-1.0.3/.pkgdir/microsocks/. /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.microsocks_installed
echo "microsocks" >> /home/qq1316237299/lede/staging_dir/target-x86_64_musl/pkginfo/microsocks.default.install
make[3]: Leaving directory '/home/qq1316237299/lede/feeds/passwall_packages/microsocks'
time: package/feeds/passwall_packages/microsocks/compile#0.09#0.03#0.11
make[3]: Entering directory '/home/qq1316237299/lede/feeds/passwall_packages/tcping'
mkdir -p /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/qq1316237299/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qq1316237299/lede/build_dir/target-x86_64_musl/tcping//tcping-0.3/.pkgdir/tcping/. /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qq1316237299/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.tcping_installed
echo "tcping" >> /home/qq1316237299/lede/staging_dir/target-x86_64_musl/pkginfo/tcping.default.install
make[3]: Leaving directory '/home/qq1316237299/lede/feeds/passwall_packages/tcping'
time: package/feeds/passwall_packages/tcping/compile#0.08#0.04#0.11
make[3]: Entering directory '/home/qq1316237299/lede/feeds/passwall_packages/brook'
touch /home/qq1316237299/lede/build_dir/target-x86_64_musl/brook-20240214/.prepared_3c6106f9752e335233f64cbebbf59946_6664517399ebbbc92a37c5bb081b5c53_check
. /home/qq1316237299/lede/include/shell.sh; /home/qq1316237299/lede/staging_dir/host/bin/libdeflate-gzip -dc /home/qq1316237299/lede/dl/brook-20240214.tar.gz | tar -C /home/qq1316237299/lede/build_dir/target-x86_64_musl/brook-20240214/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/qq1316237299/lede/build_dir/target-x86_64_musl/brook-20240214

Applying ./patches/0001-update-iploc-to-fix-build-for-riscv64.patch using plaintext: 
patching file go.mod
Hunk #1 FAILED at 8.
1 out of 1 hunk FAILED -- saving rejects to file go.mod.rej
patching file go.sum
Hunk #1 FAILED at 180.
1 out of 1 hunk FAILED -- saving rejects to file go.sum.rej
Patch failed!  Please fix ./patches/0001-update-iploc-to-fix-build-for-riscv64.patch!
Makefile:45: recipe for target '/home/qq1316237299/lede/build_dir/target-x86_64_musl/brook-20240214/.prepared_3c6106f9752e335233f64cbebbf59946_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/qq1316237299/lede/build_dir/target-x86_64_musl/brook-20240214/.prepared_3c6106f9752e335233f64cbebbf59946_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/qq1316237299/lede/feeds/passwall_packages/brook'
time: package/feeds/passwall_packages/brook/compile#0.08#0.04#0.10
    ERROR: package/feeds/passwall_packages/brook failed to build.
package/Makefile:114: recipe for target 'package/feeds/passwall_packages/brook/compile' failed
make[2]: *** [package/feeds/passwall_packages/brook/compile] Error 1
make[2]: Leaving directory '/home/qq1316237299/lede'
package/Makefile:110: recipe for target '/home/qq1316237299/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/qq1316237299/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qq1316237299/lede'
/home/qq1316237299/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
L大还是出错了
[image]

coolsnowwolf, 2024-01-14:
lede 并不包含 brook

qq1316237299, 2024-01-14:
> lede 并不包含 brook

什么叫brook 是pswd吗

bi4tms, 2024-01-14:
等待passwall那边修复

qq1316237299, 2024-01-14:
> 等待passwall那边修复

我把passwall取消了编译试试，又重新用了rm -rf staging_dir/
rm -rf bin/
rm -rf build_dir/
rm -rf tmp/
rm .config命令

1105384898, 2024-01-14:
这个是啥错误
yangyang/lede/bin/targets/x86/64/packages/kmod-sched-core_6.1.69-1_x86_64.ipk
removed '/home/yangyang/lede/bin/targets/x86/64/packages/kmod-sched_6.1.69-1_x86_64.ipk'
ERROR: module '/home/yangyang/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.69/net/sched/act_skbmod.ko' is missing.
make[3]: *** [modules/netsupport.mk:1037: /home/yangyang/lede/bin/targets/x86/64/packages/kmod-sched_6.1.69-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/yangyang/lede/package/kernel/linux'
time: package/kernel/linux/compile#10.32#3.35#11.82
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/yangyang/lede'
make[1]: *** [package/Makefile:110: /home/yangyang/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yangyang/lede'
make: *** [/home/yangyang/lede/include/toplevel.mk:231：world] 错误 2

swxk521, 2024-01-14:
> 这个是啥错误 yangyang/lede/bin/targets/x86/64/packages/kmod-sched-core_6.1.69-1_x86_64.ipk removed '/home/yangyang/lede/bin/targets/x86/64/packages/kmod-sched_6.1.69-1_x86_64.ipk' ERROR: module [...]

一样的错误

WYC-2020, 2024-01-15:
这车翻的 哈哈

GreatMichaelLee, 2024-01-15:
> 这车翻的 哈哈

......是不是你干的

GreatMichaelLee, 2024-01-15:
按一系列删除后，4ad82ac之后依旧报错

mv -f .deps/mount_clnt.Tpo .deps/mount_clnt.Po
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../support/include  -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include/tirpc  -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include  -pipe  -Wall  -Wextra  -Wstrict-prototypes  -Werror=missing-prototypes  -Werror=missing-declarations  -Werror=format=2  -Werror=undef  -Werror=missing-include-dirs  -Werror=strict-aliasing=2  -Werror=init-self  -Werror=implicit-function-declaration  -Werror=return-type  -Werror=switch  -Werror=overflow  -Werror=parentheses  -Werror=aggregate-return  -Werror=unused-result  -fno-strict-aliasing  -Werror=format-overflow=2 -Werror=int-conversion -Werror=incompatible-pointer-types -Werror=misleading-indentation -Wno-cast-function-type -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2=nfs-utils-2.6.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error=implicit-function-declaration -Wno-error=strict-prototypes -Wno-error=incompatible-pointer-types -Wno-error=format-security -Wno-error=undef -Wno-error=missing-include-dirs  -MT mount_xdr.o -MD -MP -MF .deps/mount_xdr.Tpo -c -o mount_xdr.o mount_xdr.c
mv -f .deps/mount_xdr.Tpo .deps/mount_xdr.Po
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../support/include  -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include/tirpc  -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include  -pipe  -Wall  -Wextra  -Wstrict-prototypes  -Werror=missing-prototypes  -Werror=missing-declarations  -Werror=format=2  -Werror=undef  -Werror=missing-include-dirs  -Werror=strict-aliasing=2  -Werror=init-self  -Werror=implicit-function-declaration  -Werror=return-type  -Werror=switch  -Werror=overflow  -Werror=parentheses  -Werror=aggregate-return  -Werror=unused-result  -fno-strict-aliasing  -Werror=format-overflow=2 -Werror=int-conversion -Werror=incompatible-pointer-types -Werror=misleading-indentation -Wno-cast-function-type -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2=nfs-utils-2.6.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error=implicit-function-declaration -Wno-error=strict-prototypes -Wno-error=incompatible-pointer-types -Wno-error=format-security -Wno-error=undef -Wno-error=missing-include-dirs  -MT cache.o -MD -MP -MF .deps/cache.Tpo -c -o cache.o cache.c
cache.c: In function 'uuid_by_path':
cache.c:363:25: error: storage size of 'st' isn't known
  363 |         struct statfs64 st;
      |                         ^~
cache.c:363:25: warning: unused variable 'st' [-Wunused-variable]
make[7]: *** [Makefile:483: cache.o] Error 1
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/support/export'
make[6]: *** [Makefile:416: all] Error 2
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/support/export'
make[5]: *** [Makefile:442: all-recursive] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/support'
make[4]: *** [Makefile:482: all-recursive] Error 1
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2'
make[3]: *** [Makefile:214: /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/nfs-kernel-server'
time: package/feeds/packages/nfs-kernel-server/compile#6.32#5.19#10.52
    ERROR: package/feeds/packages/nfs-kernel-server failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/nfs-kernel-server/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

WYC-2020, 2024-01-15:
> > 这车翻的 哈哈
> 
> ......是不是你干的

肯定不是我干的撒，等会我试试 看能否一把过

GreatMichaelLee, 2024-01-15:
> 按大佬的一系列目录和文件删除操作后，用最新的[4ad82ac](https://github.com/coolsnowwolf/lede/commit/4ad82acf0eace6bbef550cdf7435e3bd043e4444) 单线程编译依旧报错
> 
> mv -f .deps/mount_clnt.Tpo .deps/mount_clnt.Po x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../support/include -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include/tirpc -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -pipe -Wall -Wextra -Wstrict-prototypes -Werror=missing-prototypes -Werror=missing-declarations -Werror=format=2 -Werror=undef -Werror=missing-include-dirs -Werror=strict-aliasing=2 -Werror=init-self -Werror=implicit-function-declaration -Werror=return-type -Werror=switch -Werror=overflow -Werror=parentheses -Werror=aggregate-return -Werror=unused-result -fno-strict-aliasing -Werror=format-overflow=2 -Werror=int-conversion -Werror=incompatible-pointer-types -Werror=misleading-indentation -Wno-cast-function-type -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2=nfs-utils-2.6.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error=implicit-function-declaration -Wno-error=strict-prototypes -Wno-error=incompatible-pointer-types -Wno-error=format-security -Wno-error=undef -Wno-error=missing-include-dirs -MT mount_xdr.o -MD -MP -MF .deps/mount_xdr.Tpo -c -o mount_xdr.o mount_xdr.c mv -f .deps/mount_xdr.Tpo .deps/mount_xdr.Po x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../support/include -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include/tirpc -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -pipe -Wall -Wextra -Wstrict-prototypes -Werror=missing-prototypes -Werror=missing-declarations -Werror=format=2 -Werror=undef -Werror=missing-include-dirs -Werror=strict-aliasing=2 -Werror=init-self -Werror=implicit-function-declaration -Werror=return-type -Werror=switch -Werror=overflow -Werror=parentheses -Werror=aggregate-return -Werror=unused-result -fno-strict-aliasing -Werror=format-overflow=2 -Werror=int-conversion -Werror=incompatible-pointer-types -Werror=misleading-indentation -Wno-cast-function-type -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2=nfs-utils-2.6.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error=implicit-function-declaration -Wno-error=strict-prototypes -Wno-error=incompatible-pointer-types -Wno-error=format-security -Wno-error=undef -Wno-error=missing-include-dirs -MT cache.o -MD -MP -MF .deps/cache.Tpo -c -o cache.o cache.c cache.c: In function 'uuid_by_path': cache.c:363:25: error: storage size of 'st' isn't known 363 | struct statfs64 st; | ^~ cache.c:363:25: warning: unused variable 'st' [-Wunused-variable] make[7]: *** [Makefile:483: cache.o] Error 1 make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/support/export' make[6]: *** [Makefile:416: all] Error 2 make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/support/export' make[5]: *** [Makefile:442: all-recursive] Error 1 make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/support' make[4]: *** [Makefile:482: all-recursive] Error 1 make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2' make[3]: *** [Makefile:214: /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/.built] Error 2 make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/nfs-kernel-server' time: package/feeds/packages/nfs-kernel-server/compile#6.32#5.19#10.52 ERROR: package/feeds/packages/nfs-kernel-server failed to build. make[2]: *** [package/Makefile:116: package/feeds/packages/nfs-kernel-server/compile] Error 1 make[2]: Leaving directory '/home/builder/lede_x86' make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/builder/lede_x86' make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2 builder@Build-Server:/home/builder/lede_x86$

我两个编译机上都报这个错误。

WYC-2020, 2024-01-15:
> 我两个编译机上都报这个错误。

明显一把过，这个错误根本就不是内核引起的，我直接把L回退内核那个提交撤回来编译的 6.1.72没问题，最起码在我这边选择的这些库上面，你这个明显就是nfs-utils 这个垃圾玩意有问题，看了下 如此修复：
[image]
-D_LARGEFILE64_SOURCE

GreatMichaelLee, 2024-01-15:
> -D_LARGEFILE64_SOURCE

大佬威武，爱大佬~

WYC-2020, 2024-01-15:
> 这个是啥错误 yangyang/lede/bin/targets/x86/64/packages/kmod-sched-core_6.1.69-1_x86_64.ipk removed '/home/yangyang/lede/bin/targets/x86/64/packages/kmod-sched_6.1.69-1_x86_64.ipk' ERROR: module [...]

[image]
如此修复 
CONFIG_NET_ACT_SKBMOD

WYC-2020, 2024-01-15:
我去玩玩6.6看能编译过不，这玩意肯定问题很多哈哈

Ucopyrightable, 2024-01-15:
> > 等待passwall那边修复
> 
> 我把passwall取消了编译试试，又重新用了rm -rf staging_dir/ rm -rf bin/ rm -rf build_dir/ rm -rf tmp/ rm .config命令

取消passwall编译成功没？

WYC-2020, 2024-01-15:
passwall 并没有任何问题，我book20230606版本就能过，刚我过去看了 passwall那边已经 删除了这些path了
[image]
你们自己更新一下就可以的，我还用的老版本 ，还没升级

GreatMichaelLee, 2024-01-15:
> 我去玩玩6.6看能编译过不，这玩意肯定问题很多哈哈

大佬quick hand,最好一把过，哈哈

WYC-2020, 2024-01-15:
[image]
这就是Action自动更新所带来的bug，哈哈

qq1316237299, 2024-01-15:
> > > 等待passwall那边修复
> > 
> > 
> > 我把passwall取消了编译试试，又重新用了rm -rf staging_dir/ rm -rf bin/ rm -rf build_dir/ rm -rf tmp/ rm .config命令
> 
> 取消passwall编译成功没？

passwall也修复了  取消可以编译成功了 现在应该passwall添加也可以了

qq1316237299, 2024-01-15:
> > > > 等待passwall那边修复
> > > 
> > > 
> > > 我把passwall取消了编译试试，又重新用了rm -rf staging_dir/ rm -rf bin/ rm -rf build_dir/ rm -rf tmp/ rm .config命令
> > 
> > 
> > 取消passwall编译成功没？
> 
> passwall也修复了 取消可以编译成功了 现在应该passwall添加也可以了

passwll也好了 ，编译全部OK。。感谢各位大佬

swxk521, 2024-01-15:
> ![image](https://private-user-images.githubusercontent.com/34025388/296647480-3ddf36d8-16a4-4bcc-a502-6f9d083fbcff.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

大佬，这什么意思？是去掉CONFIG_NET_ACT_SKBMOD这个吗，那么会影响限速等流控插件吗？

qq1316237299, 2024-01-15:
> 意肯定问题很多哈

大佬 能过吗6.6的 L大回退了6.1.69了啊

qq1316237299, 2024-01-15:
> > 我去玩玩6.6看能编译过不，这玩意肯定问题很多哈哈
> 
> 大佬quick hand,最好一把过，哈哈
大佬 能过吗6.6的 L大回退了6.1.69了啊

GreatMichaelLee, 2024-01-15:
> 大佬，这什么意思？是去掉CONFIG_NET_ACT_SKBMOD这个吗，那么会影响限速等流控插件吗？

人家那图里不是明明白白吗。。。。

swxk521, 2024-01-15:
> 人家那图里不是明明白白吗。。。。

没看见 前面加号，现在明白了。

Ucopyrightable, 2024-01-15:
> 我去玩玩6.6看能编译过不，这玩意肯定问题很多哈哈

6.6编译过了没？

WYC-2020, 2024-01-15:
> > 我去玩玩6.6看能编译过不，这玩意肯定问题很多哈哈
> 
> 6.6编译过了没？

[image]
[image]

需要修复一些才能过，已经上车

WYC-2020, 2024-01-15:
> 又出错了：
> 
> CC [M] net/mptcp/mptcp_diag.o CC [M] net/netfilter/xt_FLOWOFFLOAD.o net/netfilter/xt_FLOWOFFLOAD.c: In function 'nf_flow_rule_route_inet': net/netfilter/xt_FLOWOFFLOAD.c:627:52: warning: passing argument 2 of 'nf_flow_rule_route_ipv4' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers] 627 | err = nf_flow_rule_route_ipv4(net, flow, dir, flow_rule); | ^~~~ In file included from net/netfilter/xt_FLOWOFFLOAD.c:17: ./include/net/netfilter/nf_flow_table.h:[333](https://github.com/mdaylight/actions-openwrt-x86/actions/runs/7520749801/job/20470748994#step:12:334):67: note: expected 'struct flow_offload *' but argument is of type 'const struct flow_offload *' 333 | int nf_flow_rule_route_ipv4(struct net *net, struct flow_offload *flow, | ~~~~~~~~~~~~~~~~~~~~~^~~~ net/netfilter/xt_FLOWOFFLOAD.c:630:52: warning: passing argument 2 of 'nf_flow_rule_route_ipv6' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers] 630 | err = nf_flow_rule_route_ipv6(net, flow, dir, flow_rule); | ^~~~ In file included from net/netfilter/xt_FLOWOFFLOAD.c:17: ./include/net/netfilter/nf_flow_table.h:[336](https://github.com/mdaylight/actions-openwrt-x86/actions/runs/7520749801/job/20470748994#step:12:337):67: note: expected 'struct flow_offload *' but argument is of type 'const struct flow_offload *' 336 | int nf_flow_rule_route_ipv6(struct net *net, struct flow_offload _flow, | ~~~~~~~~~~~~~~~~~~~~~^~~~ net/netfilter/xt_FLOWOFFLOAD.c: At top level: net/netfilter/xt_FLOWOFFLOAD.c:644:27: error: initialization of 'int (_)(struct net *, struct flow_offload *, enum flow_offload_tuple_dir, struct nf_flow_rule _)' from incompatible pointer type 'int (_)(struct net *, const struct flow_offload *, enum flow_offload_tuple_dir, struct nf_flow_rule *)' [-Werror=incompatible-pointer-types] 644 | .action = nf_flow_rule_route_inet, | ^~~~~~~~~~~~~~~~~~~~~~~ net/netfilter/xt_FLOWOFFLOAD.c:644:27: note: (near initialization for 'flowtable_inet.action') cc1: some warnings being treated as errors make[8]: *** [scripts/Makefile.build:250: net/netfilter/xt_FLOWOFFLOAD.o] Error 1 make[7]: *** [scripts/Makefile.build:500: net/netfilter] Error 2 make[6]: *** [scripts/Makefile.build:500: net] Error 2 make[5]: *** [Makefile:2014: .] Error 2 make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72' make[4]: *** [Makefile:29: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.modules] Error 2 make[4]: Leaving directory '/workdir/openwrt/target/linux/x86' make[3]: *** [Makefile:11: compile] Error 2 make[3]: Leaving directory '/workdir/openwrt/target/linux' time: target/linux/compile#115.41#14.58#129.67 ERROR: target/linux failed to build. make[2]: *** [target/Makefile:30: target/linux/compile] Error 1 make[2]: Leaving directory '/workdir/openwrt' make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2 make[1]: Leaving directory '/workdir/openwrt' make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2 Error: Process completed with exit code 2.

类型不匹配而已 const 转换成非const了 我是没遇到，应该是l那个补丁上面有问题而已，我这边应该是不知道在哪个提交里面修复了的，所以没遇到你这个问题，这个没法看 因为我的和l这边 还是有区别，只能你自己排查了

closed 2024-01-15 by coolsnowwolf

81268, 2024-01-15:
[image]
[image]
今天凌晨已经编译成功使用中🙂

81268, 2024-01-15:
[image]

[image]

81268, 2024-01-15:
[Uploading QQ截图20240115211522.png…]

81268, 2024-01-15:
[image]

[image]

81268, 2024-01-15:
[image]
[image]

81268, 2024-01-15:
[image]

Ucopyrightable, 2024-01-16:
> ![image](https://private-user-images.githubusercontent.com/34025388/296681557-69d6bd55-5a85-426e-b206-e428115cbde1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

请教哪些地方需要修复？