coolsnowwolf/lede #11783: armv8架构acpid报错

binge8, 2024-01-15:
### 详细叙述

armv8架构acpid报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

armv8

### 详细日志

[image]

WYC-2020, 2024-01-15:
稍微看下最近一两天 编译错误的bug就知道如何修改的 照葫芦画瓢还不会？
[image]

binge8, 2024-01-15:
> 稍微看下最近一两天 编译错误的bug就知道如何修改的 照葫芦画瓢还不会？ [...]

好的，怎么没人提交修复啊

iKenzo, 2024-01-15:
x86也一样

WYC-2020, 2024-01-15:
遇到问题多看看最近编译错误的，都一堆相同问题 照葫芦画瓢就是了

closed 2024-01-15 by coolsnowwolf